fairecasoimeme/Zlinky_TIC #1: More infos needed for the reported Attributes

pipiche38, 2021-09-23:
It would be nice to have for each attribute reported:

- Data Type
- Min and Max
- Unit
Could you clarify if a bind is required to get it reported ?
I understood that there is no Configure Reporting possible, do you confirmed ? We just have to use the Cluster 0xff66 and Attribute 0x0100 to change the reporting cycle. It would be important to know if we have to use Manufacturer Specific flag and code, and if yes, what is the Manufacturer Code.

fairecasoimeme, 2021-09-23:
* Yes I will clarify data type, min max and unit for all attributes
* No bind is mandatory for "forced report"
* Bind is required for configure report
* You can do configure report with attribut marked RP
* No 0xFF66 is only for "forced report"
* No manufacturer specific flag

fairecasoimeme, 2021-09-24:
The array is updated.

closed 2021-09-24 by fairecasoimeme