fairecasoimeme/Zlinky_TIC #13: Cluster 0x0004 and 0x0005 reported

pipiche38, 2021-09-28:
Is there any particular reason why Group and Scene clusters are available ? ( At least they are reported via a EndPoint descriptor request )

fairecasoimeme, 2021-09-28:
No reason, I will filter and cancel this cluster

pipiche38, 2021-10-31:
Still reported on lateste firmware

fairecasoimeme, 2021-11-02:
Not on the new firmware in release section

Type: 0x8043 (Simple Descriptor Response)
  SQN: 0x88
  Status: 0x00
  Short Address: 0x0F7E
  Length: 22
  EndPoint: 0x01
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0053 (Unknown)
  Input Cluster Count: 6
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0003 (General: Identify)
    Cluster 2:  Cluster ID: 0x0702 (Smart Energy: Metering)
    Cluster 3:  Cluster ID: 0x0B04 (Unknown)
    Cluster 4:  Cluster ID: 0x0B01 (Unknown)
    Cluster 5:  Cluster ID: 0xFF66 (Unknown)
  Output Cluster Count: 1
    Cluster 0:  Cluster ID: 0x0019 (General: OTA)

closed 2021-11-02 by pipiche38