fairecasoimeme/Zlinky_TIC #6: Seting Configure Reporting on Cluster 0x0702 fail

pipiche38, 2021-09-28:
When doing a Configure Reporting on Cluster 0x0702

We get 0x8c error code on Attribute 0x0000
We get 0x86 error  code on Attribute 0x0400

pipiche38, 2021-09-28:
Looks like 0x0400 is indeed not a reported attribute, so the error make sense, but not for 0x0000 attribute

fairecasoimeme, 2021-09-28:
0x8c error seems to be this kind of error :  ZPS_APL_ZDP_E_TABLE_FULL          			= 0x8c
For the moment there is only 5 configure report available. I should increase this value

pipiche38, 2021-10-31:
still valid on latest firmware

fairecasoimeme, 2021-11-02:
Hi,
It's a mistake. the right error is : E_ZCL_CMDS_UNREPORTABLE_ATTRIBUTE : 0x8c
This error is given by coordinator because it's table on this attribute is not reportable but the ZLinky ignore this message and add to its reportable table.

I'll modify reportable on ZiGate new version but this kind of error could occur to the others coordinators

pipiche38, 2021-11-02:
Is that not vice-versa. This UNREPORTABLE ATTRIBUTE is provided by the ZLinky when setting a Configure Reporting !

fairecasoimeme, 2021-11-02:
I'm going to recheck to be sure.

fairecasoimeme, 2021-11-03:
finally, you're right, there is a bug and it's fixed for next firmware

closed 2022-01-04 by fairecasoimeme