fairecasoimeme/Zlinky_TIC #9: 0x0B04 / 0x050A seems to report 2 differents values

pipiche38, 2021-09-28:
[image]

pipiche38, 2021-09-28:
Same for that one 

[image]

pipiche38, 2021-09-28:
Same here
[image]

fairecasoimeme, 2021-09-28:
It depend on EDF subscription and mono/triphasé

IMAX = mono
IMAX1,2,3 = triphasé

IMAX and IMAX1 can't be in the same time

the same for IINST and IINST1

HCHP is a different subscription of EJPHN and BBRHCJB.
This can't be in the same time

closed 2021-09-28 by fairecasoimeme