fairecasoimeme/Zlinky_TIC #14: ZLinky is not issuing a Device Annoucement after a Power Off/On

pipiche38, 2021-11-01:
As stated in the Title when power Off/On of the ZLinky there is no device annoucement !

fairecasoimeme, 2021-11-02:
This is normal on new firmware mode router because this function is only for device with zigbee 3.0.

[JN-UG-3114.pdf](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjD_LKLvvnzAhWGzYUKHR-iDvIQFnoECAUQAQ&url=https%3A%2F%2Fwww.nxp.com%2Fdocs%2Fen%2Fuser-guide%2FJN-UG-3114.pdf&usg=AOvVaw0eeoeKnZw7S4AQ8j_b28wR)

Page 20

[image]

Normally ZLinky can't do a power off/on because it is alimented by general counter. Unlike devices, ZLinky should not be moved. If counter is down coordinator is down too. For me, rejoin on power off/on is not useful.

If really, there is a need, we have to implement on button push but I don't know if it's possible yet

pipiche38, 2021-11-02:
this deviate from the ZB3.0 devices I have seen so far like Legrand-Netatmo

closed 2021-11-02 by pipiche38

fairecasoimeme, 2021-11-02:
To my knowledge they are the only ones who do. But why not. I wonder what is the use

pipiche38, 2021-11-02:
Up to now, ZLinky is the first router not issuing a Device Annoucement after a Power Off/On

fairecasoimeme, 2021-11-02:
however for tuya routers like the QS-Zigbee-C0, this is the case.

pipiche38, 2021-11-02:
I would not use Tuya as a reference. but as it is documented I have close the issue

fairecasoimeme, 2021-11-02:
unfortunately, we cannot say either that Legrand is uncompromising on the standard. the real question is to know what is the usefulness of this function.

to force a DEVICE ANNOUNCE with a REJOIN, we can use the leave function of the coordinator (E_SL_MSG_LEAVE_REQUEST)

pipiche38, 2021-11-02:
For me the feature is very simple, it will allow to know that the router has been out of the network a while, since we get a Device Annoucement when powered on.
As you stated it is expected that the ZLinky is always on. However during the test, if I'm recall correctly we have seen the ZLinky having Device Annoucement (when it was configured as a End Device )