fairecasoimeme/Zlinky_TIC #15: Requesting Binding Table respond with "Reserved" status

pipiche38, 2021-11-01:
When requesting the Binding Table, we get a status code of 0xc1 "Reserved" and we don't get the Binding table !

fairecasoimeme, 2021-11-02:
How you request Binding table please ? with which command ? it's possible that is not implemented

pipiche38, 2021-11-02:
Using Cluster 0x0033 with Raw APS command ( 0x0530)

fairecasoimeme, 2021-11-02:
OK, I'll see

fairecasoimeme, 2021-11-03:
What is the goal to get Binding table for ZLinky ? Could you explain a usecase for binding Zlinky to an other device ?

pipiche38, 2021-11-03:
The purpose is simply to get the binding table of any Zigbee devices, by queering them.
For debugging purposes for example, it is interesting to see what is in the binding table of a device, to see if the binds are correct or not.

closed 2022-02-08 by pipiche38