fairecasoimeme/Zlinky_TIC #16: intégration à HA

chpego, 2021-11-04:
Bonjour,

Qu'avez-vous besoin pour effectuer l'intégration dans HA ?
Je possède la solution, avec une clé ConBee II , je vois bien le ZLinky_TIC dans la carte réseau des appareils zigbe, mais deconz n'arrive pas à l'appairer.

Je pense qu'il faudrait ajouter l'appareil dans la liste "supportés" https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support , non ?

max5962, 2021-11-05:
Avec la Zigate, j'arrive à l'appareiller avec ZHA ( sur home assistant )  et j'ai ça comme sensor : 

[image]

Pour être franc, je sais pas trop quoi en faire. 
J'ai vu que ça se comportait comme un routeur d'après ZHA, donc même si pour le moment c'est pas fonctionnel ça fiabilise mon réseau :)

chpego, 2021-11-06:
Ouverture d'une demande de support ici https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459

capi92, 2021-11-06:
L’intégration au réseau zigbee ne fonctionne pas non plus sous zigbee2mqtt..

fairecasoimeme, 2021-11-06:
There is a guy who work on this: http://twitter.com/ralmn45.
@ralmn

ralmn, 2021-11-06:
Pour Zigbee2MQTT j'ai fait une 1ere version : https://gist.github.com/ralmn/02de46e193bf111062f15be07d3e9bd2 J'attends un peu avant de la proposer pour quelle soit intégré officiellement. Le temps d'y rajouter plus d'info si besoin. 

I do a first version of zigbee2mqtt integration  : https://gist.github.com/ralmn/02de46e193bf111062f15be07d3e9bd2 
I'm waiting before proposing it to be officially integrated. Time to add more info if needed.

Rodka08, 2021-11-21:
> Ouverture d'une demande de support ici [dresden-elektronik/deconz-rest-plugin#5459](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459)

Bonsoir @chpego, je suis votre demande de support et je suis novice, je possède la même solution que vous et j’ai vu que vous avez réussi à modifier le general.xml mais je ne parvient pas à l’envoyer sur la clé Conbee 2 via le vnc deconz sur ha… ? Avez-vous une solution à me proposer ? 

Je constate également que le firmware de la clé ne nous permet pas d’intégrer le zlinky dans ha si j’ai bien compris? 😅

En vous remerciant d’avance et en vous souhaitant une bonne soirée.

chpego, 2021-11-21:
> > Ouverture d'une demande de support ici [dresden-elektronik/deconz-rest-plugin#5459](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459)
> 
> Bonsoir @chpego, je suis votre demande de support et je suis novice, je possède la même solution que vous et j’ai vu que vous avez réussi à modifier le general.xml mais je ne parvient pas à l’envoyer sur la clé Conbee 2 via le vnc deconz sur ha… ? Avez-vous une solution à me proposer ?
> 
> Je constate également que le firmware de la clé ne nous permet pas d’intégrer le zlinky dans ha si j’ai bien compris? sweat_smile
> 
> En vous remerciant d’avance et en vous souhaitant une bonne soirée.

Oui, j'ai pu intégrer le fichier general.xml. 
En fait, j'ai créé le fichier et l'ai déposer dans le partage samba de HA. Ensuite, il suffit de se connecter en tant que root sous SSH , et de faire une copie du fichier entre le répertoire du partage samba et le répertoire où tourne le docker deconz (**/mnt/data/supervisor/addons/data/core_deconz**).

En l'état non, on ne peut pas l'intégrer dans la version de deconz, mais ça devrait venir, sinon j'invite @AdrienBigot à nous paratger les dernières informations ;)

AdrienBigot, 2021-11-21:
Salut,

J'ai reussi à intégrer dans HomeAssistant une certaine partie grace à l'utilisation des DDF. Mais pour cela il faut être dans une des dernières version de deCONZ.
Pour le moment on ne peut pas tout faire via les DDF et cette feature est en version beta donc elle se désactive à chaque redémarrage de deCONZ.
Je n'ai pour le moment par exemple pas pu intégrer le Cluster spécifique LiXee qui contient certains attributs comme : l'option tarifaire, les horaires HP/HC, l'avertissement de dépassement de puissance souscrite etc ....

Aussi, le binding fait via les DDF n'est pas généric car il y a plusieurs combinaisons possibles d'abonnements et type d'installation. Par exemple je suis sur un compteur monophasé avec un abonnement base (sans HP/HC) et une puissance de 6kVA.
Chez moi la récupération de la valeur de l'index compteur se fait via le cluster 0x0702 sur l'attribut 0x0000 mais pour une personne avec un abonnement Heure pleines /  heures creuses il faudra interroger les attributs 0x0100 et 0x0102.

Ce n'est qu'un exemple et à ce titre @Smanar m'a averti qu'il faudrait au final toucher au code plutôt que de passer par les DDF : 

> And we will have a problem too, the device is easy to make it working, it s zigbee compliant. But according to the contract or the mode (historical/standard), bind and report will not be the same.
IDK if tis will be possible without touching the c++ code.

@ suivre ici : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459  

Voici une capture d'un test sur le dashboard Home Assistant.

[image]

chpego, 2021-12-22:
On progresse, je viens d'ajouter, via le travail d'@AdrienBigot , une partie des informations du compteur ... voir [ici](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-999224084)

AdrienBigot, 2021-12-22:
Par contre, il faut préciser que rien n'est magique pour le moment. Il faut modifier le fichier general.xml à la mano dans le container et ajouter le fichier DDF également soit en le deposant soit via l'interface GUI. 
Pour le fichier general.xml il n'est question que de quelques jours le temps que la PR soit validée mais pour le fichier DDF il faudrait que les mainteneurs de l'intégration offrent la possibilité de déposer le fichier directement via l'interface de configuration de l'intégration deCONZ.
Etant donné qu'ils viennent de mettre à jour l'intégration avec la dernière version de deCONZ, je pense que c'est dans les cartons. Je vais voir si on peut ouvrir une feature request à ce sujet .

AdrienBigot, 2021-12-22:
Ha et bien voilà : https://github.com/home-assistant/addons/issues/2272

Il faudrait rajouter un petit :+1:  pour reveiller cette feature request s'il vous plaît :)

chpego, 2021-12-22:
C'est fait 👍

AdrienBigot, 2021-12-23:
BOn et bien ticket fermé car ce n'est pas une issue.
La feature request a été ouverte ici et nécessite un petit pouce :+1:    svp  :-D   : https://community.home-assistant.io/t/deconz-support-management-of-ddf-files-through-ha-config/370331

Merci

WatermelonFox, 2022-04-01:
Bonjour à tous,
Je suis carrément néophyte avec HA et deconz (conbee2), vous utilisez deconz dans un container docker (https://github.com/deconz-community/deconz-docker) ?
J'ai du mal à comprendre où faire les ajustements mano pour réussir à appairer ma ZLinky_TIC à deconz pour ensuite l'intégrer dans HA 😨
Una âme généreuse voudrait-elle bien donner de son temps pour expliquer comment jouer avec tout ça?

Smanar, 2022-04-01:
Ici tu as quelqu'un qui cherchait le repertoire ou installer le DDF avec HA https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-1019496135

Par contre, pour le moment il y a plusieurs DDF, pour le mode standard/historique, normal ou HP/HC, production ou pas, triphasé ou pas

Un autre source d'infos https://forum.phoscon.de/t/lixee-zlinky-tic-ddf-file/1164

WatermelonFox, 2022-04-06:
J'ai installé le DDF dans mon dossier devices/generic et il semble bien être utilisé/reconnu par deCONZ car il est visible dans deCONZ en novnc cependant il n'est pas reconnu dans l'interface "phoscon app", j'ai tenté de le ré-appairer plusieurs fois et de redémarrer la clef, le conteneur, etc mais rien ne semble y faire, est-ce qu'il me manque quelque chose?
[image]

WatermelonFox, 2022-04-06:
Merci pour la réponse rapide encore une fois!
C'était visiblement déjà au bon endroit, j'ai enregistré depuis le noVNC et la date de modification de mon fichier a bien été rafraichi.
Je l'ai passé au status "Gold" comme indiqué puis ai effectué un "Hot reload" et là ça a bien changé mon node à un "Power 16"!
Merci pour ton assistance ❤️

Smanar, 2022-04-06:
De rien, et il y a un post pour cet appareil sur le forum de phoscon si tu t'ennuis ^^.
On est en retard sur les autres projets zigbee, la on va rajouter le compteur pour la production et on fignole le HP/HC.

mverove, 2022-04-20:
Bonjour,

Est-ce qu'il est possible d'uploader le fichier .ddf depuis l'interface VNC ? Sur Raspberry, il faut brancher un clavier et un écran pour accéder à la console... pas simple.

Autre question: ma version de deconz est la 6.12.0 dans mon instance HomeAssistant.
Est-ce que vous savez s'il prévu d'intégrer la reconnaissance "automatique" du zlinky dans une prochaine version de deconz ?

D'avance, merci !

Smanar, 2022-04-20:
Tu veux dire SSH ? De mon coté je n'utilise ni clavier, ni écran, que ce soit pour le SSH, le VNC ou le mode desktop.
Alors oui en theorie c'est possible, il y en a qui l'ont deja fait https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-1019501674
Mais bizarrement les repertoires changent d'un utilisateur a l'autre et la je n'y connais rien en docker pour aider. Si tu sais comment creer/editer un fichier, pas de soucis pour toi.

>Est-ce que vous savez s'il prévu d'intégrer la reconnaissance "automatique" du zlinky dans une prochaine version de deconz ?

Mince j'ai encore oublié, je les met en PR de suite, mais par contre, il y en a plusieurs, du coup il sont desactivés et ce sera a toi d'activer celui dont tu as besoin, via le VNC (editeur) ou le SSH (editeur de texte), en fonction du contrat, du mode, ect ...

mverove, 2022-04-20:
J'ai accès en ssh à mon instance HA mais dans la version que j'ai (la seule à l'époque documentée pour rasp), on a accès directement au container, pas au système "hôte".
Je pense que c'est pour cette raison que plusieurs personnes ont des répertoires différents (s'ils accèdent à l'hôte, dans le volume monté par le container docker suivant où il est monté depuis l'hôte, etc...). 

Il faudrait que je puisse, depuis le container, me connecter au système mais quand je fais un ssh vers l'IP de l'hôte depuis le container:
```
[core-ssh share]$ ssh homeassistant@172.30.32.1
homeassistant@172.30.32.1: Permission denied (publickey,keyboard-interactive).
[core-ssh share]$ 
```
> Mince j'ai encore oublié, je les met en PR de suite, mais par contre, il y en a plusieurs, du coup il sont desactivés et ce sera a toi d'activer celui dont tu as besoin, via le VNC (editeur) ou le SSH (editeur de texte), en fonction du contrat, du mode, ect ...

Pas de souci, je choisirai celui qui correspond à mon contrat. Au passage, je veux bien le lien vers ta PR. Merci !

Smanar, 2022-04-20:
Le PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5969
Mais seul les 2 "base" sont fonctionnels nativement, pour le HP/HC il fonctionne aussi, mais la personne a utilisé une astuce qui va changer a la prochaine stable, l'avantage c'est que du coup pour le moment le support est natif sur HA.
Pour le triphasé/production on attend que HA incorpore la future stable de deconz pour les tests.

Désolé du hors sujet, mais si tu réussis a t'en sortir avec HA et le SSH, j'ai lancé une bouteille a la mer ici https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-ha-using-text-editor/1839

mverove, 2022-05-06:
Bonjour à tou.te.s,

Merci @Smanar , grâce à ton post et la réponse apportée, j'ai réussi à pousser le fichier DDF [zlinky_tic_historique_mono_base.json](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5969/files#diff-c61cf49fbaf58fadb274b5f06d5fd2c3626945ef824a5faaf7e2d63027bafec5) sur le host de mon instance HA (Raspberry).

Je peux voir mon Zlinky dans Deconz:
[image]
J'ai fait un clic droit pour faire un _edit DDF_:
[image]
Puis ouvert le fichier .json que je venais de pousser:
[image]
Je n'ai pas trouvé comment on enregistrer pour prise en compte.

Ensuite, j'ai redémarré mon mondule Deconz au travers de mon HA GUI mais je me retrouve toujours avec:
[image]

Est-que j'ai oublié quelque chose ?

Smanar, 2022-05-06:
Oui ^^. Les fichiers DDFs pour cet appareils sont desactivés pour le moment. Le status est en "Draft".
Il te faut passer le fichier DDF en "Gold", soit via l'editeur de DDF soit avec un simple editeur de texte 

`"status": "Draft",` a passer en `"status": "Gold",`

En le passant en gold, quelque soit tes réglages il se sera obligatoirement choisit, donc deconz va la reconnaitre directement au prochain demarrage ou apres le "hot relaod".

Le nom du node va passer de 0xXXXX a quelque chose de plus concret, et si tu fais "edit DDF" tu vas revoir ton fichier.

Dans la barre de titre tu vas retrouver le chemin et le nom du fichier.

mverove, 2022-05-06:
Pas de souci !!
Merci pour ton retour ! En suivant ta procédure, ça fonctionne !!!

Nickel !! Merci beaucoup.

Pour info, après avoir redémarré le module:
[image]
Reconnu via Phoscon:
[image]
Mis en dashboard (faut que je l'améliore ;-)):
[image]
Encore merci !!!

StephanGR, 2022-05-22:
> Pas de souci !! Merci pour ton retour ! En suivant ta procédure, ça fonctionne !!!
> 
> Nickel !! Merci beaucoup.
> 
> Pour info, après avoir redémarré le module: [image] Reconnu via Phoscon: [image] Mis en dashboard (faut que je l'améliore ;-)): [image] Encore merci !!!

Aurais-tu la possibilité de me partager ton fichier de conf ?

Merci

AdrienBigot, 2022-06-10:
Salut,

Juste pour être sûr, quand tu dis que tu te connectes en SSH, c'est à quoi que tu te connectes ? A ton raspbery qui possèdes le dongle raspbee II ? A ton HomeAssitant ?

Une question peut être bête mais on sait jamais , le dossier .local est un dossier caché. Un simple 'ls' ne suffit pas tu fais bien un 'ls -a' ?

Ton problème est peut être aussi lié au fait que que tu ne cherches pas au bon endroit. Le dossier .local se trouve dans le home directory de l'utilisateur avec lequel tourne deconz.
Chez moi deconz tourne avec l'utilisateur pi, et non pas en root. Donc le chemin est :
```shell
pi@raspi-garage:~/.local/share/dresden-elektronik/deCONZ $ pwd
/home/pi/.local/share/dresden-elektronik/deCONZ`

Smanar, 2022-06-10:
Si tu es sous HA, ca peut t'aider https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

tatawi, 2022-06-10:
Je me connecte via le terminal web du module complémentaire ssh & terminal.
Je faisais bien un ls -a .
Par contre effectivement , je ne me connecte pas pour le coup avec le bon compte.

Je vais regarder ça.
Pour le moment en tout cas, zha a reconnu le zlinky après un redémarrage donc j'ai bien les éléments qui remontent.
Mes capteurs avec deconz et zlinky avec zha. Ça va sûrement devenir ma solution en attendant que ce soit intégré et reconnu par phoscon.

Merci pour l'aide

Téléchargez Outlook pour iOS<https://aka.ms/o0ukef>

tatawi, 2022-06-11:
Bonjour,

Je me connecte à Home Assistant (avec root). A la connexion j'ai un message : "Welcome to the Home Assistant command line".

Comment peux-tu savoir avec quel utilisateur tourne deConz ? j'ai essayer de me connecter en ssh avec l'utilisateur pi également mais je ne connais pas le mot de passe pour me connecter.

AdrienBigot, 2022-06-11:
Ce que je t'expliquais pour le user deconz, c'était dans le cas où tu te connectais à ton raspberry en SSH. 
Avec HA, deconz est dans un container docker la procédure pour y accéder se fait via le lien que ta donné @Smanar un peu plus haut.

tatawi, 2022-06-12:
Bonjour,

J'ai bien avancé, merci pour votre aide ! En faite je ne me connectais au bon élément, en me connectant sur le port 22222, je vois bien les bons éléments et j'ai pu créer mon fichier DDF (je n'ai pas trouvé d'éditeur de texte mais j'ai pu créer le fichier avec un printf et en collant le contenu du fichier).
J'ai chargé le fichié via deCONZ, fait un hot releaod et le nom a bien changé. Je vois également le fichier dans phoscon.

Mon capteur "alarm 39" n'est par contre pas remonté dans HA, je vais attendre un peu pour voir.

Est-ce que vous savez si on peut mettre à jour le firmware du zlinky via l'integration deConz ?

Smanar, 2022-06-12:
>mais j'ai pu créer le fichier avec un printf et en collant le contenu du fichier

Alors la, ca m’intéresse ta manip ^^, pour la rajouter sur le tuto, en general les gens font avec nano (pas forcement present) ou cat

>Mon capteur "alarm 39" n'est par contre pas remonté dans HA, je vais attendre un peu pour voir.

Oui si il est present dans l'API, il va remonter, surement un soucis de synchro

>Est-ce que vous savez si on peut mettre à jour le firmware du zlinky via l'integration deConz ?

Bonne question, d'apres la page d’accueil ca depend des versions, regardes si tu as le cluster OTA pour ton appareil dans deconz, (le deuxieme rond en haut a droite dans le node, ca va deplier la liste)
Si oui tu vas trouver ton bonheur dans Panel / OTAU update.

tatawi, 2022-06-13:
Merci pour les infos.
Pour creer le fichier DDF, j'ai juste créé un fichier vide
>touch zlinky.json
puis : 
>printf 'CONTENU DU FICHIER'

Bianf76, 2023-04-21:
Hello à tous, je ressors de terre ce thread concernant le module. Je pense avoir effectué toutes les manips et maj du DDF (j'ai d'ailleurs testé plusieurs pour essayer de regler mon souci). Coté Deonz je pense être bon, le modul remonte bien : 
[image]

Mais Coté HA, j'ai cette erreure dans le log:
[image]

Et pour tout vous dire, je seche. @mverove tu n'as pas rencontré ce souci ?

Smanar, 2023-04-21:
Ca se serait plutot un probleme coté HA.

Pour info, je sais pas quelle methode tu as utilisé mais il vaut mieux creer un nouveau DDF dans le dossier `/data/.local/share/dresden-elektronik/deCONZ/devices/` plutot que de bidouiller l install deconz.

Je vais demander au dev HA si il voit ce que c'est ou si tu peux direct faire une demande sur leur github

Bianf76, 2023-04-21:
Concernant la méthode, j'ai récupéré la config que tu as publié sur le forum deconz, copié dans un fichier renommer en .json. Copier ce fichier dans le container comme vu ci-dessous : 
[image]
J'ai testé 2 DDF pour verifier.
PS : je suis néophyte, je découvre ;o). Pour la demande sur leur github, preneur des infos.

Bianf76, 2023-04-21:
Concernant la méthode, j'ai récupéré la config que tu as publié sur le forum deconz, copieé dans un fichier renommer en .json. Copier ce fichier dans le container comme vu ci-dessous : 
[image]
J'ai testé 2 DDF pour verifier.
PS : je suis néophyte, je découvre ;o)

Smanar, 2023-04-22:
Alors le probleme viendrait que tu aurais un ZHAalarm mais sans le champs state/alarm.
Je pense qu'il y a un loupé sur le DDF.
HA fonctionne a partir de docker, et perso je ne peux pas t'aider la dessus non plus. Si tu n'utilises pas le docker de deconz mais le docker du plugin deconz venant de HA, tu as la longue methode ici https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
Et pour faire court le DDF est a mettre dans `/data/.local/share/dresden-elektronik/deCONZ/devices/` tu fais une copie de celui que tu veux dans ce dossier, et tu le passes en "Gold".

Bianf76, 2023-04-24:
Hello Smanar,
Merci pour ta réponse. Au final j'ai tout refait et réinstallé une version HA différente qui intègre ZHA. Au final c'est beaucoup plus simple.

AnnoyingTechnology, 2023-11-29:
Bonjour,

Merci pour le travail fait sur ce capteur. 
J'ai ajouté le DDF à la main dans /opt/deconz/devices/lixee/, HA a trouvé 3 entities : 
* Puissance instantanée en VA
* Consummation cumulée en kWh
* Une alerte que j'imagine être le dépasse de puissance souscrite d'après une autre issue. 

**Est-il possible d'obtenir la puissance instantanée en Watt ?** 
J'ai pas mal de puissance réactive, donc les VA sont faussés et je serais très intéressé d'avoir des W à la place ou en plus des VA.

Merci pour votre aide.

Smanar, 2023-11-29:
Je pense pas, je viens de vérifier https://github.com/fairecasoimeme/Zlinky_TIC et je ne vois que de la puissance apparente. Mais normallement par defaut le linky ne fait que la puissance apparente (contrairement aux anciens compteurs).

Par contre la différence ne devrait pas etre énorme, je dirais max 0.8, Je dis peut etre une connerie mais meme si tu vas payer la puissance active, on s'en fout un peu car ce qui va compter ici c'est le compteur d'energie, pas de puissance.

AnnoyingTechnology, 2023-11-29:
Merci du retour.
On est d'accord, économiquement je m'en fou. 
Par contre pour regarder ce que je consomme, avoir une valeur quelque part entre x1.2 et x1.5 ne permet pas de se faire une idée. 

J'essaie d'écrire un sensor virtuel qui utilise les kWh cumulé reporté pour faire une moyenne de puissance instantanée à la minute. Si j'y arrive ça fera le job modulo un petit délais d'une minute.

AnnoyingTechnology, 2023-11-30:
[image]

Regarder le VA qui oscille à 450W alors que la maison consomme 100W réel est vraiment trop loin du compte. J'ai fait un sensor qui utilise les kWh et leur évolution sur les 2 dernières minutes, ça fait le job.

jgirard78540, 2024-01-13:
> Pas de souci !! Merci pour ton retour ! En suivant ta procédure, ça fonctionne !!!
> 
> Nickel !! Merci beaucoup.
> 
> Pour info, après avoir redémarré le module: [image] Reconnu via Phoscon: [image] Mis en dashboard (faut que je l'améliore ;-)): [image] Encore merci !!!

Bonjour à tous,

ça fonctionne parfaitement pour moi mais en revanche à chaque redémarrage du module (coupure électrique ou mise à jour de home assistant) je dois refaire la manip.
Avez vous ce problème ?

Smanar, 2024-01-13:
Ha le soucis avec les DDF ? ou il te faut changer un status ?
C'est le fonctionnement de HA, c'est le principe des dockers. Il te faut utiliser un repertoire persistent https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Avec ce systeme tu peux creer un DDF qui restera persistent, mais la manip est loin d'etre simple.

Et je suis même pas sur que deconz ne va pas prendre un DDF en "draft"  a la place de celui que tu vas mettre en "gold".

Smanar, 2024-01-13:
Pas besoin de le supprimer. Surtout si il n'est pas reconnu, sinon il sera juste mit a jour.
Il te faut mettre le DDF dont tu as besoin dans `/data/.local/share/dresden-elektronik/deCONZ/devices/` le passer en gold et redemarrer deconz.
Et re-inclure l'appareil si pas deja reconnu.