fairecasoimeme/Zlinky_TIC #17: remontées des infos

antfleu, 2021-11-07:
Bonjour,

L’actualisation des données est envoyées toutes les 5 min, il me semblait avoir lu sur le readme il y a un petit moment la possibilité de modifier ce délai?
Je fais du délestage chez moi et 5 min me paraisse beaucoup trop (toutes les 15s avant en wifinfo). 

merci

fairecasoimeme, 2021-11-07:
Bonjour,
Sur la dernière version toutes les données du Linky sont mis à jour sur le module toutes les 7 secondes.
Ensuite c'est la box ou le plugin que vous utilisez qui s'occupe de récupérer les données. 
Je ne sais pas quel plugin / box vous utilisez mais c'est vers eux qu'il faut se rapprocher.

Pour le délestage, il me semble qu'il y a une information qui peut être utile 

ADPS | 0xFF66 | 0x0005 | RP |   | Uint16 | 3 car | A | Avertissement de Dépassement De Puissance Souscrite | 0
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

Pour récupérer les informations, il existe 2 méthodes :
* soit utiliser le config report (après config c'est le ZLinky qui enverra les infos)
* soit venir poller les informations (le coordinateur fait une demande et le ZLinky répond)

Attention dans tous les cas, si vous faites remontées toutes les informations  toutes les 15 secondes, cela risque de saturer votre réseau zigbee.

antfleu, 2021-11-07:
Super merci pour le retour rapide et complet :). 
Je suis sûr jeedom/Zigbee (plugin officiel), j’avais pas vu le cron ttes les 5min. 
Je vais regarder ADPS apparement en cas de dépassement on a 250s avant coupure. Actuellement en valeur j’ai 45A (correspondant à mon abonnement) à voir comment ça fonctionne. 
merci encore et bon Dimanche
Antoine

closed 2021-11-07 by antfleu