fairecasoimeme/Zlinky_TIC #19: Comment distinguer le mode standard du mode historique ?

pdecat, 2021-11-09:
Bonjour,

j'ai un doute sur le mode actuel de mon compteur Linky.

Existe t-il un cluster qui permet de déterminer cela à coup sûr ?

Le seul attribut que je vois se rafraîchir toutes les 10 à 15 secondes est `rms_current` (0x0508) sur le cluster `ElectricalMeasurement` (0x0b04):

Il semble correspondre directement à l'entité `electrical_measurement rms_current`:

[image]

L'attribut `current_summ_delivered` (id: 0x0000) du cluster `Metering` (0x0702) semble correspondre à l'entité `smartenergy_metering summation_delivered` qui ne bouge que lorsque je consulte la valeur explicitement via l'interface permettant d'interroger les clusters :

[image]

[image]

Les attributs correspondant à `electrical_measurement` et `electrical_measurement rms_voltage` ne bougent jamais : 

[image]

[image]

Note: il n'y a que quatre entités importantes du point de vue de HA : 

[image]

SilentT19, 2021-11-11:
Le seul moyen c'est d'appuyer sur le bouton du compteur jusqu'à arriver sur "MODE TIC"

SilentT19, 2021-11-11:
Après quand on lis : https://www.enedis.fr/media/2035/download page : 24-25/38

6.2.3.14.
Registre de statuts

État de la sortie télé-information
0 = mode historique
1 = mode standard
bits 17

fairecasoimeme, 2021-11-11:
Bonjour,
quel abonnement as-tu ? Base ou HCHP ou autres ?
le compteur renvoi un index en Wh. Pour avoir une consommation, il faut que le graphe soit en mode variation, c'est à dire faire une soustraction entre l'index recu et le dernier reçu
Fred

pdecat, 2021-11-11:
> quel abonnement as-tu ? Base ou HCHP ou autres ?

J'ai un abonnement de base.

fairecasoimeme, 2021-11-11:
Si tu interroges le cluster 0xFF66 attribut 0x0000, il te sort quoi ?

pdecat, 2021-11-11:
ZHA / Home Assistant ne me permet pas d'interroger ce cluster :

```
Nov 11 18:08:07 poulpy hass[1489]: 2021-11-11 18:08:07 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [65382] cluster_type: [in] endpoint_id: [1] attribute: [0] manufacturer: [4151] response: [None] failure: [{0: <Status.UNSUPPORTED_ATTRIBUTE: 134>}],
```

Je vais tenter de développer le quirk.

pdecat, 2021-11-11:
J'ai ouvert une issue déjà: https://github.com/zigpy/zha-device-handlers/issues/1146

pdecat, 2021-11-20:
J'ai ouvert sur une PR sur le quirk implémenté par @N0ciple pour pouvoir interroger l'attribut 0x0000 du cluster 0xFF66 avec ZHA: https://github.com/zigpy/zha-device-handlers/pull/1165

Par contre, quand je l'interroge, j'ai une erreur:
```
Nov 20 13:07:22 myhost hass[1551]: 2021-11-20 13:07:22 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xE837), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=65382, TSN=249, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x04\x37\x10\xF9\x00\x00\x00')
Nov 20 13:07:22 myhost hass[1551]: 2021-11-20 13:07:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1024, SrcAddr=0xF785, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=66, SecurityUse=<Bool.false: 0>, TimeStamp=3167755, TSN=0, Data=b'\x18\xB1\x0A\x00\x00\x21\x32\x59', MacSrcAddr=0x25B3, MsgResultRadius=28)
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy.zcl] [0xf785:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=177 command_id=Command.Report_Attributes>
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy.zcl] [0xf785:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=22834>)]]
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy.zcl] [0xf785:1:0x0400] Attribute report received: measured_value=22834
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=249)
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=65382, SrcAddr=0xE837, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=90, SecurityUse=<Bool.false: 0>, TimeStamp=3169625, TSN=0, Data=b'\x1C\x37\x10\xF9\x01\x00\x00\x86', MacSrcAddr=0x2FD0, MsgResultRadius=28)
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [zigpy.zcl] [0xe837:1:0xff66] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4151 tsn=249 command_id=Command.Read_Attributes_rsp>
Nov 20 13:07:24 myhost hass[1551]: 2021-11-20 13:07:24 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [65382] cluster_type: [in] endpoint_id: [1] attribute: [0] manufacturer: [4151] response: [None] failure: [{0: <Status.UNSUPPORTED_ATTRIBUTE: 134>}],
```

Je creuse...

pdecat, 2021-11-20:
> Le seul moyen c'est d'appuyer sur le bouton du compteur jusqu'à arriver sur "MODE TIC"

Je suis enfin allé vérifier, je suis en mode "historique". J'ai fait ma demande pour passer en mode "standard".

N0ciple, 2021-11-21:
Le quirk permet de requêter l'info via "gérer les clusters" mais malheureusement pas de faire un sensor, c'est bien ça ?

pdecat, 2021-12-31:
Je suis enfin passé en mode TIC Standard :tada: 

[image]

Cela m'a nécessité plusieurs allez-retours entre le service client de mon fournisseur (EDF en l'occurrence) et celui d'Enedis à qui appartient le compteur Linky, alors voici la procédure à suivre pour y parvenir : 

1. Contactez votre fournisseur d'énergie (EDF dans mon cas)
2. Demandez leur un changement de type `F185` _Modification du dispositif de comptage sans impact sur la formule tarifaire d'acheminement_ (c'est le service technique Enedis qui m'a fourni l'information)
3. Si votre fournisseur vous dit qu'il faut demander à Enedis, insistez. Enedis ne veut pas faire le changement si la demande ne vient pas du fournisseur. Il est possible que votre fournisseur vous dise que cela a un coût (~31€ dans mon cas mais je ne l'ai pas encore vu passé)
4. Une fois la demande faite par votre fournisseur à Enedis, vous pouvez contacter Enedis pour en avoir la confirmation.
5. Le changement a été effectif au bout d'une dizaine de jours je dirais (mon ZLinky_TIC perdant systématiquement la connexion au bout de quelques heures après insertion dans le port TIC, je ne suis pas allé vérifié tous les jours physiquement...)

ElieDeloumeau, 2022-01-03:
Chez Mint, voici la procédure qui a fonctionné pour moi :

1. Contactez Mint sur cette page https://www.mint-energie.com/Pages/Contact/contact.aspx en demandant de modifier le mode TIC de votre compteur en "Standard"
2. Attendez qu'ils ne répondent pas (2/3 jours) 🤦‍♂️
3. Tweetez "@mint_eco Bonjour, pouvez-vous traiter ma demande n°XXXXX SVP ?" et attendez qu'ils répondent (ils vont répondre, c'est mieux pour leur image)
4. Donnez votre référence client en DM à @mint_eco
5. Le changement sera effectif sous 24 heures

fairecasoimeme, 2022-02-04:
Fixed with v4 

<html><body>
<h3 dir="auto">Linky Mode ( From V4)</h3>
<h3 dir="auto"><a id="user-content-values" class="anchor" aria-hidden="true" href="https://github.com/fairecasoimeme/Zlinky_TIC#values"><svg class="octicon octicon-link" viewBox="0 0 16 16" version="1.1" width="16" height="16" aria-hidden="true"></svg></a></h3><h3 dir="auto">values</h3>

Cluster | Attribut | Value
-- | -- | --
0xFF66 | 0x0300 | 0 - Mode historique monophasé 1 - Mode standard monophasé  2 - Mode historique triphasé 3 - Mode standard triphasé 5 - Mode standard monophasé producteur 7 - Mode standard triphasé producteur

<h3 dir="auto"><a id="user-content-bit-values" class="anchor" aria-hidden="true" href="https://github.com/fairecasoimeme/Zlinky_TIC#bit-values"><svg class="octicon octicon-link" viewBox="0 0 16 16" version="1.1" width="16" height="16" aria-hidden="true"></svg></a></h3><h3 dir="auto">bit values</h3>
<p dir="auto">bit 0 - Mode historique / standard<br>
bit 1 - Triphasé / Monophasé<br>
bit 2 - Production / non production</p>
</body>
</html>[](https://github.com/fairecasoimeme/Zlinky_TIC#values)[](https://github.com/fairecasoimeme/Zlinky_TIC#bit-values)Linky Mode ( From V4)
values
Cluster 	Attribut 	Value
0xFF66 	0x0300 	0 - Mode historique monophasé
1 - Mode standard monophasé
2 - Mode historique triphasé
3 - Mode standard triphasé
5 - Mode standard monophasé producteur
7 - Mode standard triphasé producteur
bit values

bit 0 - Mode historique / standard
bit 1 - Triphasé / Monophasé
bit 2 - Production / non production

closed 2022-02-04 by fairecasoimeme