fairecasoimeme/Zlinky_TIC #23: PTEC ne remonte pas

Titom59, 2021-11-13:
Bonjour ayant reçu le module je me suis attelé à l'inclusion de celui sur zigbee2mqtt grâce à [https://gist.github.com/ralmn/02de46e193bf111062f15be07d3e9bd2](https://gist.github.com/ralmn/02de46e193bf111062f15be07d3e9bd2) 

Je suis en mode historique avec les HC et HP mais le module ne mère monte pas l'info.

d3v2a, 2021-11-13:
C'est normal, car cet élément n'a d'attribut dans le fichier, il faut remplacer la ligne 49 par
```js
{"name":"PTEC","unit":"-","clusterId":1794,"attributId":32,"comment":"","description":"Periode tarifaire en cours","cluster":"seMetering","attribut": 32,"triphase":false,"production":false,"standard":false},
```

ralmn, 2021-11-15:
L'attribut en 32 (`0x20`) n'est pas connu dans la lib de Z2M. 

Par contre @fairecasoimeme  je le ne vois pas non plus dans la spec https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf (page 715)...

fairecasoimeme, 2021-11-16:
Salut,
si dans la doc, il y a bien l'attribut 0x20 - ActiveRegisterTier Delivered (page 716)
[image]

fairecasoimeme, 2021-11-16:
Pour le PTEC, l'info existe bien côté ZLinky, il remonte bien un CString (data type 0x42) avec 4 caractères : HP.. ou HC..

ralmn, 2021-11-16:
Effectivement j'avais mal vu dans la spec

newbeetle, 2021-11-20:
Sous Jeedom la requête "clusterId":1794,"attributId":32 ne renvoie aucune information

newbeetle, 2021-11-22:
Finalement au bout d'un certain temps, la remontée se fait

closed 2021-12-16 by fairecasoimeme