fairecasoimeme/Zlinky_TIC #24: Timeout sur la configuration du reporting sur le cluster 0x0702 et l attribut 0x0000

d3v2a, 2021-11-13:
L'attribut est marqué comme reportable mais le module ne répond pas à la demande de configuration 

|Commande TIC|CLUSTER|Attribut|Droit|Comment|data type|size max|unit|designation|valeur par defaut|
|------------|-------|--------|-----|--------------|---------|-------|----|-----------|------------|	
|[image] BASE|0x0702|0x0000|RP||Uint32|9 car|Wh| Index Base|0|

Je ne sais pas si c est possible, ce serait intéressant d'avoir aussi les autres index (0x0100, 0x0102, etc.) en reportable

fairecasoimeme, 2021-11-16:
Oui sur la version actuelle quand on fait un config report sur ces valeurs on a en retour le code erreur 0x8c mais le ZLinky prend quand même la demande.

Sur la prochaine version, le retour du config report sera en "success"

ElieLiabeuf, 2021-12-31:
Hello @fairecasoimeme avez-vous une date pour cette release ? Ca semble être le point bloquant pour avoir une rémontée dans home assitante (voir : https://github.com/fairecasoimeme/Zlinky_TIC/issues/18)

fairecasoimeme, 2022-02-04:
Sur la dernière version 4. Tous les éléments demandés sont reportables

closed 2022-02-04 by fairecasoimeme