fairecasoimeme/Zlinky_TIC #27: Possibilité de désactiver la LED bleue

WoodyTheWoo, 2021-11-16:
Bonjour,

Est-ce qu'il serait possible de proposer la désactivation de la LED bleue en façade du module svp ?

Le choix de l'état logique (inversé ou non) de cette LED au travers d'un interrupteur virtuel Zigbee serait parfait.

D'avance, merci.

fairecasoimeme, 2021-11-17:
Bonjour,
L'objectif pour cet appareil est de passer la labelisation enedis.
Et pour avoir ce label, la gestion de la LED est très importante. Voici un extrait de l'annexe du document Enedis-NOI-CPT_02E.pdf 
[image]

WoodyTheWoo, 2021-11-17:
Bonjour,
Merci pour votre réponse rapide, en effet je comprends, pas de soucis.
Je tâcherai de regarder dans le code et de compiler une version qui correspond à mon besoin alors :)

closed 2021-11-18 by fairecasoimeme