fairecasoimeme/Zlinky_TIC #30: Too aggressive report

vk496, 2021-11-27:
Hello!

I've just started using zlinky with Zigbee2mqtt (currently in development by @ralmn) and I noticed that zlinky report data massively (20 times per second?), which completely blocks the whole IO of the storage and the system (and increase the DB some GB every hour)

More precisely, the `linkquality` is the attribute which constantly changes. The rest of attributes retain the same for much more time.

My current workaround is disabling most of the entities , but it would be much more gentile if zlinky would report only when a attribute change.

Can someone confirm that this behaviour is not only in my side?

Br,
Valentin

djsyl, 2021-11-28:
Hello,
in Z2M you have in settings (spécific) parameter measurement_poll_interval
you have to restart z2m after change this
[image]

have fun

vk496, 2021-11-28:
Hello,

Thank you for your response @djsyl 

Currently I'm limited to the previous version of Z2M (arm64), so I manually force a poll interval of 120 in the code.

However, this affect only non reportable attributes. The reportable attributes are sent over Zigbee network multiples times per second (firmware behaviour), which are forwarded to the MQTT network.

Moreover, I think Zlinky send ALL data each time, so I suspecting that **all** attributes are transmitted over Zigbee constantly + the 60s poll interval (which is irrelevant if I'm right).

I couldn't check it directly within the Zigbee network, but my home assistant DB was 200 MB after months of usage. After setting up the zlinky with only 2 attributes enabled (energy and PTEC), it increased 400 MB this night (with all attributes, consumes 1G in 2h)

BR,
Valentin

djsyl, 2021-11-28:
Je sais pas ce que vous avez fait 
mais j'ai aussi modifier mon fichier.js 
et mis des paramètres toutes les 15 secondes sur le pool
sur une version 1.22 de zigbee2mqtt
constatez dans le fichier gif ci-joint que l'on n'est loin de la saturation.
[image]

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const { repInterval } = require('zigbee-herdsman-converters/lib/constants');

const { precisionRound, postfixWithEndpointName } = require('zigbee-herdsman-converters/lib/utils');

const e = exposes.presets;
const ea = exposes.access;

// Options 

let option_standard = () => exposes.binary(`standard`, ea.SET, true, false).withDescription(`Compteur avec la TIC en mode standard`)
let option_triphase = () => exposes.binary(`triphase`, ea.SET, true, false).withDescription(`Compteur triphase`)
let option_production = () => exposes.binary(`production`, ea.SET, true, false).withDescription(`Compteur avec de la production`)

// All attributes avaialables
let data = [
    { "name": " BASE", "unit": "Wh", "clusterId": 1794, "attributId": 0, "comment": "", "description": "Index Base", "cluster": "seMetering", "attribut": "currentSummDelivered", "triphase": false, "production": false, "standard": false },
    { "name": "EAIT", "unit": "Wh", "clusterId": 1794, "attributId": 1, "comment": "Production", "description": "Energie active injectée totale (Triphasé)", "cluster": "seMetering", "attribut": "currentSummReceived", "triphase": true, "production": f$
    { "name": "PMAX", "unit": "W", "clusterId": 2820, "attributId": 1293, "comment": "Triphasé", "description": "Puissance maximale triphasée atteinte (Triphasé)", "cluster": "haElectricalMeasurement", "attribut": "activePowerMax", "triphase": true$
    { "name": "PAPP", "unit": "VA", "clusterId": 2820, "attributId": 1295, "comment": "Triphasé", "description": "Puissance apparente (Triphasé)", "cluster": "haElectricalMeasurement", "attribut": "apparentPower", "triphase": true, "production": fals$
    { "name": " HCHC", "unit": "Wh", "clusterId": 1794, "attributId": 256, "comment": "", "description": "Index HCHC", "cluster": "seMetering", "attribut": "currentTier1SummDelivered", "triphase": false, "production": false, "standard": false },
    { "name": " HCHP", "unit": "Wh", "clusterId": 1794, "attributId": 258, "comment": "", "description": "Index HCHP", "cluster": "seMetering", "attribut": "currentTier2SummDelivered", "triphase": false, "production": false, "standard": false },
    {
        "name": "EASF02", "unit": "Wh", "clusterId": 1794, "attributId": 258, "comment": "", "description": "Energie active soutirée Fournisseur, index 02", "cluster": "seMetering", "attribut": "currentTier2SummDelivered", "triphase": false, "producti$
    {
            "name": "CCASN", "unit": "W", "clusterId": 2820, "attributId": 1291, "comment": "", "description": "Point n de la courbe de charge active soutirée", "cluster": "haElectricalMeasurement", "attribut": "activePower", "triphase": false, "productio$
    {
                "name": "CCASN-1", "unit": "W", "clusterId": 2820, "attributId": 2315, "comment": "", "description": "Point n-1 de la courbe de charge active soutirée", "cluster": "haElectricalMeasurement", "attribut": "activePowerPhB", "triphase": false, "pr$
    {
                    "name": "UMOY1", "unit": "V", "clusterId": 2820, "attributId": 1297, "comment": "mono / triphasé", "description": "Tension moy. ph. 1", "cluster": "haElectricalMeasurement", "attribut": "averageRmsVoltageMeasPeriod", "triphase": false, "produc$
    { "name": "IRMS1", "unit": "A", "clusterId": 2820, "attributId": 1288, "comment": "mono / triphasé", "description": "Courant efficace, phase 1", "cluster": "haElectricalMeasurement", "attribut": "rmsCurrent", "triphase": false, "production": fals$
];

function createChunk(array, chunkSize) {
    let R = [];
    for (let i = 0; i < array.length; i += chunkSize) {
        R.push(array.slice(i, i + chunkSize));
    }
    return R;
}

function onlyUnique(value, index, self) {
    return self.indexOf(value) === index;
}

function extractAttribute(cluster, standard, triphase, production) {
    let attributes = data.filter(item => item.cluster === cluster);
    if (standard == false) {
        attributes = attributes.filter(a => a.standard == false);
    }
    if (triphase == false) {
        attributes = attributes.filter(a => a.triphase == false);
    }
    if (production == false) {
        attributes = attributes.filter(a => a.production == false);
    }
    return attributes.map(i => i.attribut).filter(a => a != null).filter(onlyUnique);
[... 38 lines cut ...]
    type: ['attributeReport', 'readResponse', 'write'],
    convert: (model, msg, publish, options, meta) => {
        const standard = options && options.standard ? options.standard : false;
        const triphase = options && options.triphase ? options.triphase : false;
        const production = options && options.production ? options.production : false;
        return parsingResponse(model, msg, extractAttribute('haElectricalMeasurement', standard, triphase, production))
    }
};

const seMeteringFZ = {
[... cut ...]
        const production = options && options.production ? options.production : false;
        let payload = parsingResponse(model, msg, extractAttribute('seMetering', standard, triphase, production))
        //fusion HC/HP dans l'index commun
        if (payload['currentSummDelivered'] == 0 && (payload['currentTier1SummDelivered'] > 0 || payload['currentTier2SummDelivered'] > 0)) {
            payload['currentSummDelivered'] = payload['currentTier1SummDelivered'] + payload['currentTier2SummDelivered'];
        }
        return payload;
    }
};
```

vk496, 2021-11-29:
Hi,

Thanks @djsyl !

I checked again and the problem was the version of the Z2M. When I updated to 1.22 all was working as expected (seems that edge versions don't update automatically, so switched to stable).

I suppose that in 1.21 the poll interval was completely ignored and used as real time.

closed 2021-11-29 by vk496

djsyl, 2021-11-29:
@vk496 vincent
super si tout fonctionne 
bonne journée

vk496, 2021-11-30:
Hello,

Even if my issue is solved, I still thinking that it would be better if the firmware report when something changes (and each N minutes, configurable) instead of poll from Z2M or any other Zigbee Controller. So I think is better to keep this issue opened for now

fairecasoimeme, 2021-12-16:
this feature is already available for the reportables attributes. With zigbee, you can use the config report command to manage a report change with a value. But warning, all of this parameters can flood the zigbee network and provoque a power issue

pipiche38, 2022-08-30:
> Hello,
> 
> Even if my issue is solved, I still thinking that it would be better if the firmware report when something changes (and each N minutes, configurable) instead of poll from Z2M or any other Zigbee Controller. So I think is better to keep this issue opened for now

fully with you as (1) the polling is network consuming and you can poll an attribute while this has no update, so you have consume energy for nothing
(2) the configure reporting table of the ZLinky seems to be limited to 20, which means that if you have a Standard Mode you still have to do polling 

The firmware when receiving the Linky frame and when populating the various attributes can detect a change , and in such case could send an automatic report.