fairecasoimeme/Zlinky_TIC #31: Zlinky clignote 7 fois

nlhomme, 2021-12-11:
Bonjour,

J'ai reçu hier mon Zlinky.
A peine branché, il ne met à clignoter non pas 2 ou 4 fois comme indiqué dans la doc mais... 7/8 fois:
https://user-images.githubusercontent.com/9362475/145674006-50615484-9a8b-4505-8056-e2d147fb5871.mp4

Cela n'a pas empêché sa liaison avec Home Assistant (en version 2021.11.5 + ZiGate 3.1e):
[image]

Ceci dit je vois qu'aucune donnée n'a été relevée depuis la nuit dernière et que quand je suis allé voir ce matin mon compteur Linky, j'ai constaté que la LED était complètement éteinte.

Du coup je ne comprends pas ce que je rate...
Il faut savoir que:
* Actuellement le relevé à la seconde de mon compteur est désactivé (seul le relevé horaire l'est).
* Sur le site d'Enedis, on ne peut qu'activer le relevé horaire, pas à la seconde, est-ce que je dois contacter mon fournisseur pour ça ?

nlhomme, 2021-12-11:
Pour compléter mon message précédent, j'ai cru comprendre au gré des issues que j'ai pu parcourir ces derniers jour que la mise à jour de décembre de Home Assistant allait améliorer significativement la prise en charge du Zlinky, est-ce bien cela?

N0ciple, 2021-12-12:
Bonjour @nlhomme 

Avec home assistant 2021.12.0 vous aurez accès à la puissance apparente soutirée, et l'unité de l'indexe de consommation sera corrigé (avant on avait des Wh rapportés dans home assistant comme des kWh, ça ne sera plus le cas), mais il manque encore son actualisation automatique. Plus de détails dans mon message [ici](https://github.com/fairecasoimeme/Zlinky_TIC/issues/18#issuecomment-991866807).

Ce que vous décrivez s'apparente plus (du moins je crois) à des problèmes de connexions. Ça m'arrive également de retrouver mon ZLinky éteint, je dois le débrancher et rebrancher pour qu'il reparte. Dans les modifications que j'ai faite qui sont dans la version 2021.12.0 il n'y a rien qui touche à la connexion en soi.

nlhomme, 2021-12-13:
J'avoue à avoir du mal à comprendre, quand je débranche et rebranche le Zlinky je vois qu'il s'intègre très bien au maillage du réseau Zigbee, puis quelques heures plus tard il ne répond plus, je dois le débrancher et rebrancher pour que ça reparte.

Mais alors que veut dire les 7 clignotements qu'on voit dans la vidéo? J'ai eu ça quand je l'ai branché pour la première fois à peine sortie de l'enveloppe?

Et est-ce que je dois demander impérativement à mon fournisseur d'électricité ou Enedis d'activer le relevé de consommation à la seconde de mon compteur?

nlhomme, 2021-12-13:
Et non seulement il ne répond plus mais quand je retourne au compteur la led est totalement éteinte, comme dans cette issue
:https://github.com/fairecasoimeme/Zlinky_TIC/issues/26

fairecasoimeme, 2021-12-14:
Bonjour,
Les clignotements signifient qu'il n'arrive pas à décoder votre linky comme indiquer dans le README.
Pouvez-vous me contacter par mail pour échange standard ? contact[at]lixee.fr

nlhomme, 2021-12-14:
Bonjour,

J'ai également testé sur le compteur Linky d'un ami et j'obtiens le même résultat.
Je pense donc que l'échange est la meilleure chose à faire.

Merci, je vous contacte de suite.

L'homme

fairecasoimeme, 2021-12-15:
> Android [5].

Bonjour,

Pouvez-vous retourner votre module à l'adresse suivante :

Frédéric DUBOIS
19 avenue du 8 Mai
66300 LLUPIA
FRANCE

Dès réception je procèderai à l'échange standard.

Cordialement,
Fred

Links:
------
[1] https://github.com/fairecasoimeme/Zlinky_TIC/issues/26
[2] 
https://github.com/fairecasoimeme/Zlinky_TIC/issues/31#issuecomment-992730186
[3] 
https://github.com/
[4] 
[5]

nlhomme, 2021-12-15:
Bonjour,

Merci pour votre retour, je vous ai fait une réponse par mail à contact[at]lixee.fr.

Je propose qu'en parallèle de notre échange par mail de laisser cette issue ouverte le temps du changement de Zlinky pour confirmer que tout est rentré dans l'ordre.

Bonne journée

fairecasoimeme, 2021-12-15:
Bonjour,
Désolé mais je n'ai pas reçu de mail. Pouvez-vous renvoyer votre mail à contact@lixee.fr et/ou contact@zigate.fr

nlhomme, 2021-12-15:
Bonjour, je viens de vous renvoyer un mail aux deux adresses.

Si vous ne le recevez pas, je vous invite à contacter votre fournisseur de messagerie, je l'ai envoyé à partir d'une adresse icloud

Bonne fin de journée

nlhomme, 2021-12-15:
Au pire si vous n'avez toujours rien dans la soirée dites-le-moi, je vous renverrai le mail depuis un autre fournisseur de messagerie

nlhomme, 2022-01-15:
Bonjour,

Je viens d'installer le nouveau Zlinky.
La led est enfin fixe et s'intègre au réseau Zigbee, j'attends que les premières données arrivent pour clore définitivement l'issue

nlhomme, 2022-01-16:
Le nouveau Zlinky semble bien fonctionner correctement.

Il y a des problèmes de remontées de données mais je vois que c'est en traitement dans la #18.

Je me permets dans ce cas de clore l'issue, merci pour votre assistance!

closed 2022-01-16 by nlhomme