fairecasoimeme/Zlinky_TIC #32: Detect Linky mode

vk496, 2021-12-14:
Hello,

Most of the attributes exposed by Zlinky allow us to detect which kind of configuration is currently active (if we have 0/MAX values in Production, we assume is not enabled). However, instead of trying to read multiple attributes in order to discard possibilities, it would be more comfortable to expose directly that values somewhere (maybe liXee cluster?)

vk496, 2022-01-04:
This can be deduced from STGE attribute, so there is no need to do it from the firmware. However, I'm not sure if is possible to deduce the phase (single or three) and if is producer too

fairecasoimeme, 2022-01-04:
Yep, 
you can deduce the phase or producer mode with asked the right label index and if the value is 0, the mode is not OK

for example:
historique / triphase : --> PPOT
standard / triphase : --> SMAXN2
standard / producer:  --> SMAXIN

vk496, 2022-01-04:
Just checked. My current config is: standard and single phase (no producer):

PPOT: Never get a value
SMAXN2: Never get a value
SMAXIN: Never get a value

However, trying to read triphase label index of historique, I get a MAXint16 (65535). I don't see a clear way to deduce it, since a non readed value doesn't mean it doesn't exist (network delay for example)

vk496, 2022-01-10:
@fairecasoimeme any idea to how to detect the standard/historique?

I thought using `STGE` would be perfectly fine, but it seems that is only available in standard mode following the Enedis specification (even if it reports the mode????)

fairecasoimeme, 2022-01-10:
What is the final goal of this feature ?
if you get PPOT, you're in historic mode with 3 phases
if you get OPTARIF but not PPOT, you're in historic mode mono phase
if you get SMAXN2, you're in standard mode with 3 phases
if you get SMAXIN, you're in standard mode with product feature
if you get SMAXN but not SMAXN2, you're in standard mode with mono phase.

It's sure that is complicated.
If it's really important, I can add a new command in aim to give you the linky mode.

vk496, 2022-01-10:
My main problem to support Zlinky in Z2M is that I'm not reading the **labels** (PPOT, OPTARIF, etcl) but instead, clusters and attributes from a zigbee perspective.

That means, for example, a user in historique will have `OPTARIF`, but in standard he will have `NGTF`. Both come from the same place (`0xFF66/0x0000`) and in both cases they will always have a value.

For that reason I need to make sure that detecting the correct configuration of Linky works properly, since I will expose a set of labels based on that. If I expose just all of them without doing a filter, it cause a huge DB overload, since EVERY change of EVERY label is registered. Problems of that was already reported in Home Assistant

fairecasoimeme, 2022-01-10:
Fixed with next version

https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#linky-mode--from-v4

fairecasoimeme, 2022-01-14:
I'll try to do this ASAP