fairecasoimeme/Zlinky_TIC #33: Firmware OTA

vk496, 2021-12-14:
Hello,

Since this repo already published a release with the OTA firmware, I would like to propose a homogeneous way to make future release (specially, the filename and the tag). That would allow us to implement effective OTA updates detection based on the current firmware of the zlinky and the last published in this repo.

Br,
Valentín

silversmarty, 2022-01-04:
Great idea ! I must admit I'm quite a noob and I have no idea in how to update my Zlinky_TIC to v3 that has just been released (despite the kind of tutorial that is mentionned in the readme, talking about ZiGate).
If any OTA feature can be added to this device, it would be awesome!

Thanks

fairecasoimeme, 2022-01-04:
The feature is available on ZLinky_TIC.
Please see with plugins to see how to proceed.

vk496, 2022-01-04:
Hello @fairecasoimeme 

Currently I'm developing support of Zlinky in Z2M. I tried to use OTA and it seemed to worked correctly (FIleVersion in the firmware is reported as 3), but firmware version nor firmware date was updated. And the bug of Date is still present (https://github.com/fairecasoimeme/Zlinky_TIC/issues/38), so I assume that my zlinky still in the same status.

Could you verify that OTA issue is just on my side and not a firmware problem?

seblang, 2022-01-04:
@vk496 
Thank you for the converter! it's working fine. 
Can you explain us how to update the firmware by OTA?
I have the file ZLinky router v3.0.ota, but what should I do with it?
sorry it's new for me :-)

vk496, 2022-01-04:
In Z2M, you had to modify the code (URL of the index.json) and manually deploy a http server (I use python3 http.server) of the OTA repo of Z2M.

Anyway, I'm not sure if it worked properly, so I prefer to make sure is working before formally supporting OTA in the device.

seblang, 2022-01-05:
Ha somebordy done the update from Firmeware ?

what do you have in Z2M interface now?

[image]

fairecasoimeme, 2022-01-05:
Indeed, I think there is a problem. I see what is not ok.

vk496, 2022-01-06:
@fairecasoimeme I tried to wipe all firmware and put it in v1.0. Then I did a successful OTA update.

The device reports a `fileVersion` of 3, like in the OTA header file (so I'm not able to update anymore). But the rest of the device remains the same.

[image]

Flashing the firm manually over TTL, remains all exactly the same:

[image]

There is something wrong there

vk496, 2022-01-06:
When flashing manually, the last commits are working correctly, even if the device report old firmware version

fairecasoimeme, 2022-01-06:
After some tests, I think there is a problem with OTA and BIN files

fairecasoimeme, 2022-01-06:
Please reupload new Bin or OTA files and retest ?

vk496, 2022-01-06:
@fairecasoimeme I confirm that is working correctly. Z2M Frontend doesn't update the values, but a read report the last version (fileVersion 3, {"swBuildId":"4000-0003","dateCode":"20220104"}).

Just as reminder, Date is broken in this version: https://github.com/fairecasoimeme/Zlinky_TIC/issues/38#issuecomment-1006444101

closed 2022-01-12 by vk496