fairecasoimeme/Zlinky_TIC #34: Need to power off/on LiXee_TIC to reset ADPS value after a consumption spike

AdrienBigot, 2021-12-20:
Hi,

By default ADPS (id 0x0005 in cluster 0xFF66) is set to 0 if we don't exceeed to contrat limit value.
If we exceed the limit and then return to a normal situation, the ADPS value remains blocked at the last value above the limit.

Step to reproduce the problem : 
1. I turn on many devices and ADPS value increase to 38A.
2. I turn off one device, the value change to 34A in a few seconds.
3. I turn off another device, the value change to 31A.
4. I turn off all unecessary devices, the value is stucked to the last 31A.
5. Unpluged and plug my Zlinky, my ADPS value is back to 0 
 
It seems the zlinky_tic store this value in memory and never reset it when we return to a normal situation.

Adrien

fairecasoimeme, 2022-01-10:
sorry for delay.
ZLinky_TIC don't store any values. I think this is the linky which keep the last value. 
So, I'll ask Enedis

fairecasoimeme, 2022-01-10:
Ok, I understand.
Indeed, ADPS is a command which appear only when there is an exceed value. In this case, the value is collected.
But when the power value is ok, ADPS doesn't exist anymore and the module don't update the value. I'll see how to change this.

fairecasoimeme, 2022-02-04:
fixed with v4

closed 2022-02-04 by fairecasoimeme