fairecasoimeme/Zlinky_TIC #36: Couleur du lendemain

chrisalex-dev, 2022-01-03:
Bonjour,
Mon Zlinky fonctionne bien et remonte bien les info sauf une qui est importante pour moi à savoir la couleur de lendemain. Je suis sous jeedom et avec une abonnement Tempo (jour bleu, blanc et rouge avec heures pleines et creuses pour chacune des couleurs des jours).
je joins une "capture d'écran" afin d'aider à identifier d'où peut venir le souci.
Par avance merci de vos lumières
[image]
.

fairecasoimeme, 2022-01-04:
Bonjour,
le problème provient surement du plugin.
Veuillez leur remonter votre souci.

closed 2022-01-04 by fairecasoimeme

chrisalex-dev, 2022-01-06:
bonjour,
j'ai ouvert un incident chez Jeedom sur votre conseil dont je vous fait part de la réponse :
"Je doute vraiment que le plug in soit en cause car s'il y a 3 tirets c'est le module qui les a envoyé, jeedom ne peux pas les inventer..."