fairecasoimeme/Zlinky_TIC #37: Probleme d'appairage

bensaeba, 2022-01-03:
Bonjour à tous,

Je viens de faire l'acquisition de la Zlinky (depuis le temps que j'attendais un ERL me permettant de me connecter à ma TIC!! mon compteur étant en limite de propriété, il était assez compliqué de tirer un câble).
J'ai aussi reçu sous le sapin la zigateV2, la Zlinky est donc mon premier équipement zigbee (j'utilisais jusque-là le RFXcom).

Problème, je n'arrive pas à appairer le zlinky.
J'ai suivi les procédures décrites ici:
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/fr-fr/Tuto_Appairage-objet.md
https://github.com/fairecasoimeme/Zlinky_TIC
https://bestofcpp.com/repo/fairecasoimeme-Zlinky_TIC

Je passe en mode appairage dans l'interface zigate  http://XXXXXX:9440/ (appairage permanent et avec l'assistant)
Quand je branche le zlinky la led bleue est allumée et fixe (pas de clignotement me disant qu'elle est en attente d'appairage - l'appareil clignote 2 fois toutes les 7 secondes). Est-ce normal?
J'ai quand même réinitialisé l'appareil (appuyer sur le bouton "link" pendant 10 secondes puis relâcher, la Led s'éteint puis se mettre à clignoter).

Étant donné que c'est la première fois que je fais un appairge sur zigate je me dis que je ne dois pas suivre la bonne procédure pour l'appairer, aussi quelle est la bonne méthode? 
J'ai aussi envisagé la distance mais en testant sur un autre compteur proche quelques mètre j'arrive à la même conclusion…

ma config:
	• Raspeberry4
	• domoticz 2021.1
	• Plugin: 5.1.016 
	• Firmware: 0320 (j'ai un message me disant qu'une mise à jour est disponible)

Est-ce qu'il vous faut d'autre information?

Bonne année à tous :)

Ben

fairecasoimeme, 2022-01-04:
problème en cours de traitement par mail

closed 2022-01-04 by fairecasoimeme

bensaeba, 2022-01-06:
Bonjour,

Sauf erreur de ma part je n'ai reçu aucun email.
Merci beaucoup

Ben