fairecasoimeme/Zlinky_TIC #39: Couleur du lendemain

chrisalex-dev, 2022-01-06:
bonjour,
j'ai ouvert sur vos conseil un incident chez Jeedom dont vous trouverez la réponse un peu lapidaire ci dessous :
https://github.com/fairecasoimeme/Zlinky_TIC/issues/36#issue-1092720411

fairecasoimeme, 2022-01-06:
Bonjour, 
Apparemment, les tirets correspondent à une couleur du lendemain inconnue.
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwj-pvahkp31AhWGzoUKHZ6-BAoQFnoECBMQAQ&url=https%3A%2F%2Fwww.enedis.fr%2Fmedia%2F2027%2Fdownload&usg=AOvVaw32Y35kiJxzRPJk9R-jrOzv

La couleur du lendemain (donnée du groupe d’étiquette « DEMAIN ») est codée sur 4 caractères ASCII alphanumériques 
selon la syntaxe suivante :
* « ---- » : couleur du lendemain non connue, (le caractère noté « - » correspond au caractère ASCII « tiret »),
* « BLEU » : le lendemain est jour BLEU,
* « BLAN » : le lendemain est jour BLANC,
* « ROUG » : le lendemain est jour ROUGE.

Il n'y a donc pas de dysfonctionnement du module. Pour le vérifier vous pouvez aussi connecter un module usb TTL en utilisant le connecteur de programmation RX et GND. Puis en ouvrant une console putty sur le port COM, vous pourrez observer ce que renvoi précisément le Linky. Mais si vous avez des tirets, c'est qu'il remonte la bonne valeur

closed 2022-01-06 by fairecasoimeme

chrisalex-dev, 2022-01-17:
bonjour,
vous aviez bien raison le module n'est pas en cause  car depuis la mise à jour du plug-in zigbee en 2022-01-11par Jeedom, la couleur du lendemain fonctionne bien désormais

HoM3r17, 2022-10-14:
> 
Bonjour,
Utilises-tu le plugin officiel Jeedom ? Est-ce que cette donnée remonte a une intervalle particulière ? 

Actuellement le Zlinky_Tic est installé depuis 12h et je n'ai aucune remontée dans les champs "Periode tarifaire en cours" et "Couleur du lendemain", tout le reste semble remonter normalement...