fairecasoimeme/Zlinky_TIC #40: Problem subscribe to attributes

vk496, 2022-01-06:
Hello,

When I try to subscribe to **UMOY1** (0x0B04/0x0511), I receive a error from the zigbee controller saying that is not reportable. Probable the same happens to **UMOY2** and **UMOY3**.

Also, when I try to subscribe to SINSTS (0xB04/0x050F) I get a SOFTWARE_FAILURE

fairecasoimeme, 2022-01-10:
fixed with next version for UMOY1, UMOY2, UMOY3

Could you tell me more with the SINSTS problem. Who give to you the SOFTWARE_FAILURE ?
What is the complete request ?

vk496, 2022-01-10:
the SOFTWARE_FAILURE that I receive is from Z2M (herdsman library). I will send you later the zigbee traces

vk496, 2022-01-11:
```
DI","daysNumberNextCalendar":1,"drawnVAMaxN1":3750,"linkquality":255,"message1":"PAS DE MESSAGE","message2":"","meterSerialNumber":"812061182025","powerThreshold":6,"rmsCurrent":2,"rmsVoltage":235,"siteId":"23276845056102","softwareRevision":2,"statusRegister":"003A0001","update":{"state":"idle"},"update_available":false}'
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +5s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +5s
  zigbee-herdsman:deconz:driver query aps data indication +4ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +7ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 95 +3ms
  zigbee-herdsman:deconz:driver query aps data indication +1ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 95 srcAddr: 0x6f21 destAddr: 0x0 profile id: 0x104 cluster id: 0xb04 lqi: 255 +7ms
  zigbee-herdsman:deconz:frameParser response payload: 24,142,10,5,5,33,236,0 +1ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":142,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1285,"dataType":33,"attrData":236}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":28449,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}' +5s
Zigbee2MQTT:debug 2022-01-11 11:04:52: Received Zigbee message from 'zlinky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":236}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-01-11 11:04:52: Configuring 'zlinky'
  zigbee-herdsman:controller:endpoint Bind zlinky/1 seMetering from '0x00212effff07a5cd/1' +5s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +20ms
Zigbee2MQTT:info  2022-01-11 11:04:52: MQTT publish: topic 'zigbee2mqttDEV2/zlinky', payload '{"activeEnerfyOutD01":2519,"activeEnerfyOutD02":453,"activePower":456,"activePowerMax":1250,"activePowerPhB":804,"apparentPower":374,"availablePower":6,"averageRmsVoltageMeasPeriod":236,"currentDate":"02061182025","currentIndexTarif":1,"currentPrice":"SEMAINE","currentTarif":"SEM WE MERCREDI","daysNumberNextCalendar":1,"drawnVAMaxN1":3750,"linkquality":255,"message1":"PAS DE MESSAGE","message2":"","meterSerialNumber":"812061182025","powerThreshold":6,"rmsCurrent":2,"rmsVoltage":236,"siteId":"23276845056102","softwareRevision":2,"statusRegister":"003A0001","update":{"state":"idle"},"update_available":false}'
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x6f21 EP:0 SeqNr. 97 request id: 28 +4ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 96 +0ms
  zigbee-herdsman:deconz:driver query aps data indication +1ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 28 status: 0 +23ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 98 +2ms
  zigbee-herdsman:deconz:driver query aps data indication +1ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 99 +5ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 96 srcAddr: 0x6f21 destAddr: 0x0 profile id: 0x104 cluster id: 0xff66 lqi: 255 +12ms
  zigbee-herdsman:deconz:frameParser response payload: 24,145,10,3,2,35,254,113,38,0 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":145,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":515,"dataType":35,"attrData":2519550}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":28449,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}' +33ms
Zigbee2MQTT:debug 2022-01-11 11:04:52: Received Zigbee message from 'zlinky', type 'attributeReport', cluster 'liXeePrivate', data '{"activeEnerfyOutD01":2519550}' from endpoint 1 with groupID null
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +7ms
Zigbee2MQTT:info  2022-01-11 11:04:52: MQTT publish: topic 'zigbee2mqttDEV2/zlinky', payload '{"activeEnerfyOutD01":2519,"activeEnerfyOutD02":453,"activePower":456,"activePowerMax":1250,"activePowerPhB":804,"apparentPower":374,"availablePower":6,"averageRmsVoltageMeasPeriod":236,"currentDate":"02061182025","currentIndexTarif":1,"currentPrice":"SEMAINE","currentTarif":"SEM WE MERCREDI","daysNumberNextCalendar":1,"drawnVAMaxN1":3750,"linkquality":255,"message1":"PAS DE MESSAGE","message2":"","meterSerialNumber":"812061182025","powerThreshold":6,"rmsCurrent":2,"rmsVoltage":236,"siteId":"23276845056102","softwareRevision":2,"s [... cut ...] ceNumber":null,"writeUndiv":false}) failed (Status 'SOFTWARE_FAILURE')
    at Endpoint.checkStatus (/home/vk496/zigbee-herdsman/src/controller/model/endpoint.ts:310:23)
    at Endpoint.configureReporting (/home/vk496/zigbee-herdsman/src/controller/model/endpoint.ts:644:22)
    at Object.configure (/home/vk496/zigbee-herdsman-converters/devices/lixee.js:522:13)
    at Configure.configure (/home/vk496/zigbee2mqtt/lib/extension/configure.ts:115:13))
```

Let me know if you need more low level debug

vk496, 2022-01-11:
Btw, **SMAXN** is also reported as unreportable (SMAXN)

If I'm correct, the source doesn't represent the documentation (https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/ZLinky/SDK_2_6_4_JN5189DK6/middleware/wireless/zigbee/ZCL/Clusters/MeasurementAndSensing/Source/ElectricalMeasurement.c#L267)

fairecasoimeme, 2022-01-11:
There are only max 20 slots for reportable configuration, this is why, you get a software failure.

fairecasoimeme, 2022-01-11:
For SMAXN, this is possible there is a mistake in source code. I'll see and fix

vk496, 2022-01-11:
oh, thanks for clarifying. Is a limitation of the zlinky device? or defined by standard?

In any case, I will review if is a issue because of playing too much with all modes during dev or I need to limit what should be suscribed

fairecasoimeme, 2022-01-11:
I think i can increase the value but I need to test

vk496, 2022-01-11:
> For SMAXN, this is possible there is a mistake in source code. I'll see and fix

Maybe just update the docs? I'm not sure if SMAXN make sense as reportable