fairecasoimeme/Zlinky_TIC #41: Partially working with Home Assistant

cedsib, 2022-01-09:
Hello,

I just installed my ZLinky_TIC to replace my Atome key.

When I look into HomeAssistant (ZHA, not mqtt), the value sensor.lixee_zlinky_tic_8585d205_electrical_measurement_apparent_power seems to work properly :

[image]

But :
-  I don't have HP/HC values, 
- and the sensor for energy mesurement doesn't seem to send any value so my energy mesurement is flat.

What can I do to correct it ?

Thanks a lot !

ElieDeloumeau, 2022-01-10:
Hello, it's related to #18

Hedda, 2022-05-11:
FYI, there is also a related firmware issue causing issues for ZHA as discussed here -> https://github.com/fairecasoimeme/Zlinky_TIC/issues/46 

There are more discussions on the ZHA side here (in English language only) -> https://github.com/zigpy/zha-device-handlers/issues/1146