fairecasoimeme/Zlinky_TIC #42: Arrêt de communication fréquente

seblang, 2022-01-12:
Bonjour, 

Cela fait maintenant quelques temps que j'utilise le zlinky-tic et jusque là il marchait très bien. 

Une utilisation avec z2m sous ha. 

Seulement depuis une semaine il arrête d'émettre régulièrement. La led bleue reste fixe comme si tout allait bien avec le contrôleur mais celui ci ne reçoit plus d'informations et des interrogations dans la console dev retourne une erreur de non réponse. 

Une déconnexion du module avec quelque minutes d'attente avant reconnexion resou le soucis mais pour une durée plus ou moins variable de renfonctionnement ! 

D'où vient le problème ?
Une solution ?

Crocmagnon, 2022-01-12:
J'ai également la même configuration et le même problème. J'attends avec impatience cette PR : https://github.com/Koenkk/zigbee-herdsman-converters/pull/3493

pdecat, 2022-01-14:
J'ai un problème similaire niveau commandes et attributs, mais ma LED s'éteint complètement par contre. :confused:

fairecasoimeme, 2022-01-20:
Pour le problème de coupures d'alimentation, nous avons avancé et une modification matérielle permet de limiter les problèmes.
contactez-moi par mail : contact[at]lixee.Fr

0r1g1ne, 2022-01-20:
Bonjour, j'ai le même problème. La led s'éteint, la communication s'interrompt. Je vous ai envoyé un mail à contact(at)lixee.fr. Merci d'avance.

guix77, 2022-01-22:
Comment faites-vous marcher le Z_Linky avec z2m sachant que la PR n'est pas mergée? Sauf si bien sûr vous utilisez la PR telle qu'elle.

Crocmagnon, 2022-01-23:
Je viens également d'avoir le problème de l'alimentation qui s'éteint. Je vous envoie un mail.

AdrienBigot, 2022-01-28:
Idem, de mon côté le module s'éteint régulièrement. Ca me rassure de voir que je ne suis pas le seul dans ce cas.

pdecat, 2022-02-04:
Les problèmes de communication sont résolus pour moi [avec la modification apportée par @fairecasoimeme à mon module ZLinky_TIC](https://github.com/fairecasoimeme/Zlinky_TIC/issues/48#issuecomment-1027767123).

closed 2022-02-04 by fairecasoimeme