fairecasoimeme/Zlinky_TIC #44: OTA seems not working

max5962, 2022-01-14:
Hello @fairecasoimeme ,

J'ai tenté une OTA via la commande : 

service: zha.issue_zigbee_cluster_command
data:
  ieee: "XXXX"
  endpoint_id: 1
  cluster_type: out
  command_type: client
  cluster_id: 0x0019
  command: 0
  args:
    - 0
    - 100

Et j'ai depuis 1H ces logs : 

`2022-01-14 13:02:12 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA upgrade progress: 0.0
2022-01-14 13:02:26 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA query_next_image handler for 'LiXee ZLinky_TIC': field_control=0, manufacture_id=4151, image_type=1, current_file_version=1, hardware_version=None
2022-01-14 13:02:26 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA image version: 3, size: 239198. Update needed: True
2022-01-14 13:02:26 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA image_block handler for 'LiXee ZLinky_TIC': field_control=0, manufacturer_id=4151, image_type=1, file_version=3, file_offset=0, max_data_size=48, request_node_addr=Noneblock_request_delay=None
2022-01-14 13:02:26 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA upgrade progress: 0.0
2022-01-14 13:02:26 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA image_block handler for 'LiXee ZLinky_TIC': field_control=0, manufacturer_id=4151, image_type=1, file_version=3, file_offset=40, max_data_size=48, request_node_addr=Noneblock_request_delay=None
2022-01-14 13:02:26 DEBUG (MainThread) [zigpy.zcl] [0xf9be:1:0x0019] OTA upgrade progress: 0.0`

Mais rien ne semble bouger.
Une idée ?

pdecat, 2022-01-14:
Hello, le ZLinky répond toujours aux autres commandes et remonte ses attributs ?

Lors de ma première tentative, il avait progressé jusqu'à 3% puis a perdu la connexion (LED éteinte).

À ma seconde tentative après l'avoir débranché / rebranché, c'est passé (~30 minutes pour atteindre les 100%).

max5962, 2022-01-14:
Oui, cela fonctionne toujours :-)
D'ailleurs ma zigate est à 2 mètre  donc je pense que le LQI est au top ^^

max5962, 2022-01-14:
Yes ! J'ai mes valeurs qui remontent :)

J'ai tout essayé pour l'OTA :
 - redémarrer Hass
- redémarrer la zlinky (débrancher / rebrancher)
- supprimer / rejoindre le réseau ZigBee.
Je reste à 0.0 ....

J'ai téléchargé et mis dans le dossier comme conseillé par zigpy le fichier routeur V3.ota

max5962, 2022-01-17:
J'ai tenté la version 4 qui vient de sortir mais malheureusement pas mieux ...

``` 2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] ZCL request 0x0001: [0, 4151, 1, 1, None]
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] OTA query_next_image handler for 'LiXee ZLinky_TIC': field_control=0, manufacture_id=4151, image_type=1, current_file_version=1, hardware_version=None
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] OTA image version: 4, size: 239982. Update needed: True
2022-01-17 14:34:10 INFO (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] Updating: LiXee ZLinky_TIC
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=3>
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] ZCL request 0x0003: [0, 4151, 1, 4, 0, 48, None, None]
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] OTA image_block handler for 'LiXee ZLinky_TIC': field_control=0, manufacturer_id=4151, image_type=1, file_version=4, file_offset=0, max_data_size=48, request_node_addr=Noneblock_request_delay=None
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] OTA upgrade progress: 0.0
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=6 command_id=3>
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] ZCL request 0x0003: [0, 4151, 1, 4, 40, 48, None, None]
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] OTA image_block handler for 'LiXee ZLinky_TIC': field_control=0, manufacturer_id=4151, image_type=1, file_version=4, file_offset=40, max_data_size=48, request_node_addr=Noneblock_request_delay=None
2022-01-17 14:34:10 DEBUG (MainThread) [zigpy.zcl] [0xf8e8:1:0x0019] OTA upgrade progress: 0.0
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:10 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-17 14:34:11 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
```

En tout cas, zigpy reconnait mon fichier, sait le lire car me dit clairement "je privilégie la V4 vs la V3", mais je sais pas pourquoi rien ne démarre :(

PS : je suis avec une zigate V1

max5962, 2022-01-17:
Bon j'en ai marre, je suis passé par un update manuel en physique  : 
Après avoir lancé un "./DK6Programmer -s COM4 -P 115200 -Y -p .\ZLinky_router_v4.0.bin", j'ai eu ceci au bout de quelques secondes : 

[image]

Malheureusement, après avoir TOUT remonté et rebranché : 

[image]

Je sais plus quoi faire :(

fairecasoimeme, 2022-01-17:
Peux-tu re-télécharger le binaire ?
Peux tu interroger dans gérer les clusters, le cluster basic 0x0000 attribut 0x0001 et attribut 0x0006 ?

max5962, 2022-01-17:
Re-télecharger le binaire OTA : 

```
 wget "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v4.0/ZLinky_router_v4.0.ota"
--2022-01-17 20:08:57--  https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v4.0/ZLinky_router_v4.0.ota
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/406058893/f35de9c7-b77c-4c1a-a04f-5db932ee164a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220117T190857Z&X-Amz-Expires=300&X-Amz-Signature=6fded419fc54db2dc29576bba4402927b7d0b59f36c4f2d49f7d41b0049438fe&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=406058893&response-content-disposition=attachment%3B%20filename%3DZLinky_router_v4.0.ota&response-content-type=application%2Foctet-stream [following]
--2022-01-17 20:08:57--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/406058893/f35de9c7-b77c-4c1a-a04f-5db932ee164a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220117T190857Z&X-Amz-Expires=300&X-Amz-Signature=6fded419fc54db2dc29576bba4402927b7d0b59f36c4f2d49f7d41b0049438fe&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=406058893&response-content-disposition=attachment%3B%20filename%3DZLinky_router_v4.0.ota&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 240046 (234K) [application/octet-stream]
Saving to: 'ZLinky_router_v4.0.ota'

ZLinky_router_v4.0.ota                               100%[===================================================================================================================>] 234.42K  --.-KB/s    in 0.04s

2022-01-17 20:08:58 (5.53 MB/s) - 'ZLinky_router_v4.0.ota' saved [240046/240046]

```

max5962, 2022-01-19:
@fairecasoimeme besoin de plus d'informations ?

kfran78, 2022-01-22:
existe il un tuto simple pour faire les MAJ via un PC, ou directement sur HA en OTA avec ZHA ?

kfran78, 2022-01-22:
> Re-télecharger le binaire OTA :
> 
> ```
>  wget "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v4.0/ZLinky_router_v4.0.ota"
> --2022-01-17 20:08:57--  https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v4.0/ZLinky_router_v4.0.ota
> Resolving github.com (github.com)... 140.82.121.4
> Connecting to github.com (github.com)|140.82.121.4|:443... connected.
> HTTP request sent, awaiting response... 302 Found
> Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/406058893/f35de9c7-b77c-4c1a-a04f-5db932ee164a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220117T190857Z&X-Amz-Expires=300&X-Amz-Signature=6fded419fc54db2dc29576bba4402927b7d0b59f36c4f2d49f7d41b0049438fe&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=406058893&response-content-disposition=attachment%3B%20filename%3DZLinky_router_v4.0.ota&response-content-type=application%2Foctet-stream [following]
> --2022-01-17 20:08:57--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/406058893/f35de9c7-b77c-4c1a-a04f-5db932ee164a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220117%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220117T190857Z&X-Amz-Expires=300&X-Amz-Signature=6fded419fc54db2dc29576bba4402927b7d0b59f36c4f2d49f7d41b0049438fe&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=406058893&response-content-disposition=attachment%3B%20filename%3DZLinky_router_v4.0.ota&response-content-type=application%2Foctet-stream
> Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
> Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 240046 (234K) [application/octet-stream]
> Saving to: 'ZLinky_router_v4.0.ota'
> 
> ZLinky_router_v4.0.ota                               100%[===================================================================================================================>] 234.42K  --.-KB/s    in 0.04s
> 
> 2022-01-17 20:08:58 (5.53 MB/s) - 'ZLinky_router_v4.0.ota' saved [240046/240046]
> ```
> 
> Bonne prise en compte au redémarrage :
> 
> ```
> 2022-01-17 20:11:29 DEBUG (SyncWorker_3) [zigpy.ota.provider] ImageKey(manufacturer_id=4151, image_type=1): /config/zigpy_ota/ZLinky_router_v4.0.ota, version: 4, hw_ver: (None, None), OTA string: OM15081-RTR-JN5189-0000000000000
> ```
> 
> AppVersion : 2022-01-17 20:12:52 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [0] cluster_type: [in] endpoint_id: [1] attribute: [1] manufacturer: [None] response: [3] failure: [{}],
> 
> DateCode : 2022-01-17 20:13:51 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [0] cluster_type: [in] endpoint_id: [1] attribute: [6] manufacturer: [None] response: [20220117] failure: [{}],
> 
> Voici ce que me donne l'affichage : [image]

Perso ayant reçu la clé aujourdhui et après Plug et config ZHA, le tout avec ma clé Conbee 2

J'ai mis dan smon fichier de config 
zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
      ikea_provider: false
      ledvance_provider: flase

#Pour le temps de la mise à jour
logger:
  default: info
  logs:  
    homeassistant.components.zha: debug
    zigpy: debug

Créer le dossier /config/zigpy_ota et j'y ai mis le fichier V4.ota dedans.

J'ai re démarré, attendu 10mn au moins et sans rien lancé comme ligne de commande la mise à jour s'est faite toute seule :)

[image]

Constat dans les logs.

J'ai viré du fichier de config

logger:
  default: info
  logs:  
    homeassistant.components.zha: debug
    zigpy: debug

et re démarré.

ça à l'air nickel

Peut être qua ta commande à doubler le processus et à fait bugger ...

cedsib, 2022-01-22:
Je confirme que ça a marché également :-) 

Je colle le code ici avec la bonne indentation :

```
zha:
  zigpy_config:
    ota:
      ikea_provider: false                        # Auto update Trådfri devices
      ledvance_provider: false                    # Auto update LEDVANCE devices
      otau_directory: /config/zigpy_ota           # Utilize .ota files to update everything else
```

Par contre, question con : elle apporte quoi l'update. Il y a le changelog quelque part ?

max5962, 2022-01-22:
[image]

Je vais tenter en désactivant Ikea et ledvance mais j'ai un doute ^^

pdecat, 2022-01-22:
> Je vais tenter en désactivant Ikea et ledvance mais j'ai un doute ^^

J'ai les deux activés, ça ne vient pas de là...

pdecat, 2022-01-22:
As-tu tenté la réinitialisation en appuyant longtemps sur le bouton d'appairage ?

kfran78, 2022-01-23:
> [image]
> 
> Je vais tenter en désactivant Ikea et ledvance mais j'ai un doute ^^

As tu bien mis les lignes pour le debug dans le config yml comme j ai mis plus haut. A mettre le temps de la maj pour pouvoir suivre

pdecat, 2022-01-23:
@max5962 As-tu déjà vu l'OTA fonctionner pour d'autres appareils zigbee avec ta Zigate et ZHA ?

drlibre, 2022-01-23:
Pas de soucis pour flasher la v4.0 en OTA avec le firmware `.ota` dans un répertoire et la configuration ad-hoc dans le fichier de config Home Assistant (clé ConbeeII et intégration ZHA).
L'ajout des info de debug dans le log permet de suivre l'avancement de la mise à jour qui prend plusieurs minutes (compter une dizaine de minutes).

Suggestion d'amélioration pour la doc: indiquer le lien vers les firmware... ([Firmware](https://github.com/fairecasoimeme/Zlinky_TIC/releases))

max5962, 2022-01-23:
> @max5962 As-tu déjà vu l'OTA fonctionner pour d'autres appareils zigbee avec ta Zigate et ZHA ?

Je n'ai jamais fait attention à cela. Donc impossible à dire.
Peut être que le soucis viens de zigpy-zigate ?

max5962, 2022-01-23:
Ok

> > [image]
> > Je vais tenter en désactivant Ikea et ledvance mais j'ai un doute ^^
> 
> As tu bien mis les lignes pour le debug dans le config yml comme j ai mis plus haut. A mettre le temps de la maj pour pouvoir suivre

Yes of course :
[image]

Même si ça ne devrait pas influencer :)

fairecasoimeme, 2022-02-04:
works good on ZLinky firmware side

closed 2022-02-04 by fairecasoimeme

sebPomme, 2022-02-17:
@max5962 petite question, ton homeassistant est installé dans docker ?

max5962, 2022-02-17:
@sebPomme non plus maintenant. (DSM7 de Synology a coupé la maintenance des libraires USB non stockage)

sebPomme, 2022-02-17:
@max5962 ok merci

SylvainPer, 2022-02-18:
> @sebPomme non plus maintenant. (DSM7 de Synology a coupé la maintenance des libraires USB non stockage)

Il y a plusieurs solutions pour cela, j'utilise docker et indifféremment zigate/zzh (TI).
Il faut le driver qui correspond, par exemple, pour la zigate v2, il est présent, il suffit de le charger au startup (ftdio) : https://www.cachem.fr/synology-dsm-7-0-drivers-usb/

Pour la zigate v1, il faut du cp210x et pour les clés TI, ça fonctionne avec du ch341. Pour les ajouter : http://www.jadahl.com/