fairecasoimeme/Zlinky_TIC #45: MAJ OTA avec Domoticz et PiZiGate V2

dranreb38, 2022-01-17:
Bonjour, 

Je ne possède pas de clé USB TTL CP2102, afin de mettre à jour le firmware en v4, peux t'on faire la MAJ en OTA avec domoticz ?, ou je dois forcement acquérir une clé USB / TTL

zigbeefordomoticz, 2022-01-24:
je te recommande de venir sur le forum Domoticz .
https://easydomoticz.com/forum/viewforum.php?f=28&sid=41179ed98ef4f8795637e46cc205fdcd

Tu trouveras egalement beaucoup d'information ici:
https://github.com/zigbeefordomoticz/wiki/blob/master/fr-fr/Home.md

dranreb38, 2022-01-31:
Merci, mais entretemps j'ai commandé une clé USB TTL CP2102. 
Donc mise à jour faite, je suis bien en firmware 4 mais, depuis après de nouveau un appareillage je ne reçois plus de données en historique comme auparavant.
[image]

[image]

[image]
[image]

 Je n'ai en graphe que le cluster **ZLinky_TIC_Power & ZLinky_TIC_Ampere3** d'ailleurs le graphe ZLinky_TIC_Ampere3 ne correspond pas à des ampères  ou VA

closed 2022-02-08 by dranreb38