fairecasoimeme/Zlinky_TIC #47: Augmenter la portée du Zlinky

bensaeba, 2022-01-18:
Bonjour à tous,

Plus une question trucs et astuces qu'un bug mais je voudrais savoir s'il est possible d'augmenter la portée du Zlinky avec une antenne déportée qui sort du coffret EDF.
A priori c'est possible, je vois que @pdecat l'a fait ici : https://github.com/fairecasoimeme/Zlinky_TIC/issues/19#issuecomment-1003382649
Idéalement j'aimerais faire la même chose :)
Est-qu'il serait de fournir un tuto svp?
Je pensais que la portée du zigbee serait suffisamment puissante mais mon compteur triphasé est en limite extérieure de propriété dans le coffret EDF à environ 30 mètre de mon zaspeberry+zigate+ usb est visiblement trop loin/protégé.

Merci bcp!
Ben

pdecat, 2022-01-19:
L'antenne que j'ai est celle fournie [en option sur la boutique officielle Lixee](https://lixee.fr/produits/37-72-zigate-usb-ttl-3770014375148.html#/26-antenneexterne-oui) :
[image]

bensaeba, 2022-01-19:
Hello, 
Merci pour le retour. Je n'avais pas vu cette option.
Désolé pour la question ...

closed 2022-01-19 by bensaeba

pdecat, 2022-01-19:
Pas de soucis, il y a peut-être un moyen d'en ajouter une soi-même.

fairecasoimeme, 2022-01-20:
Nop pas moyen de faire les modifs soit même sauf si on maitrise la soudure et les composants en 0402 de dimension.

bdherouville, 2024-09-11:
Bonjour,

je viens de commander un lixee sans antenne pour remplacer mon précédent qui a pris l'eau pensant que le connecteur y était. Est-ce possible de savoir si il y a plusieurs composants à ajouter ?

fairecasoimeme, 2024-09-12:
Voici la marche à suivre pour rajouter l’antenne externe : 
 
 
[image]

 
1.	Faire une rotation du condensateur de 90° vers la gauche.
2.	Souder un connecteur IPEX : ex : https://www.lcsc.com/search?q=GT-RF1125A-01
 
Grâce à cette modification, vous pourrez brancher votre antenne avec un câble ufl.