fairecasoimeme/Zlinky_TIC #48: Utilisation d'une alimentation externe

pdecat, 2022-01-19:
Mon module ZLinky_TIC [arrêtant systématiquement de communiquer (et LED éteinte) au bout de quelques heures](https://github.com/fairecasoimeme/Zlinky_TIC/issues/42#issuecomment-1013297953) après avoir été branché, j'aimerais écarter la piste d'une alimentation trop faible due à la distance avec le reste de mon réseau Zigbee.

Est-il possible d'alimenter le module ZLinky_TIC avec une alimentation externe, au moins pour ces tests ?

PS: j'ai placé un routeur filaire (une [prise IKEA](https://zigbee.blakadder.com/Ikea_E1603.html)) juste au dessus du compteur et le ZLinky_TIC s'y connecte avec un LQI local de 255 entre les deux, mais cela ne change rien. Il essaie probablement de passer en direct pour obtenir un LQI total meilleur. La prise IKEA ne perd jamais la connexion elle.

[image]

[image]

fairecasoimeme, 2022-01-20:
Salut,
pour régler le problème des coupures systématiques, j'ai surement une solution matérielle. Peux-tu me contacter par mail ?
Fred

pdecat, 2022-02-02:
Hello @fairecasoimeme,

j'ai bien reçu le module ZLinky_TIC suite à ta modification, et je l'ai aussitôt branché. Pour le moment, tout est ok :crossed_fingers:

[image]

Le LQI semble plus élevé, peut-être dû à la nouvelle position dans laquelle j'ai mis l'antenne.

En quoi consistait la modification par curiosité ?

fairecasoimeme, 2022-02-02:
J'ai rajouté un condensateur et un superviseur de tension.

Fred

pdecat, 2022-02-02:
So far, so good. Merci :)

[image]

pdecat, 2022-02-04:
Je confirme que ça tout est ok désormais.

closed 2022-02-04 by pdecat