fairecasoimeme/Zlinky_TIC #50: Problem update Firmware with USB TTL

sdeluce, 2022-02-04:
Hello,
I wanted to update the firmware to version **0004** via USB to TTL, I followed the doc, and tried to do it via a Pi 3B with a Raspbian Buster. The firmware of the Zlinky is in version **0001**. 

`sudo ./DK6Programmer -s /dev/ttyUSB0 -P 115200 -e FLASH -Y -p ZLinky_router_v4.0.bin`

[image]

I reconnect the Zlinky but the firmware version remains 0001. The result is the same whether I use the `-e FLASH` flag or not.
The zigbee key used is a ZiGate+ USB if it helps.

Thanks for your help

fairecasoimeme, 2022-02-08:
Hello,

you can do this command to get more informations when updating : 
`sudo ./DK6Programmer -s /dev/ttyUSB0 -V6 -P 115200 -e FLASH -Y -p ZLinky_router_v4.0.bin`
if the update is ok. I think you have to try to delete device on HA and re do a new pairing. 
Fred

sdeluce, 2022-02-08:
Thanks Fred,
I just tried to update the firmware by adding the `-V6` flag and everything seems to work fine:

```
sudo ./DK6Programmer -V6 -s /dev/ttyUSB0 -P 115200 -Y -p ZLinky_router_v4.0.bin

dev/ttyUSB0: Connected at 115200
/dev/ttyUSB0: Detected JN5189 with MAC address 00:15:8D:00:05:BE:86:EE

 Status : 0
/dev/ttyUSB0: Chip ID: 0x1140C686
/dev/ttyUSB0: Bootloader Version: 0xCC000014
/dev/ttyUSB0: Selected memory: FLASH
/dev/ttyUSB0: Programming FLASH at 0x0
/dev/ttyUSB0: Partial erase required on memory FLASH, addr=0x00000000, length=239984
/dev/ttyUSB0: The area to erase is not an exact multiple of the erase block size. Erase data from 0x00000000 to 0x0003aa00?
/dev/ttyUSB0: Forcing operation due to command line argument
/dev/ttyUSB0: Erasing FLASH
Progress: 0
/dev/ttyUSB0: Completed
/dev/ttyUSB0: Programming FLASH
Progress: 0
/dev/ttyUSB0: Programming FLASH
[...]
Progress: 99
/dev/ttyUSB0: Programming FLASH
Progress: 100
/dev/ttyUSB0: Memory programmed successfully
```

I had previously removed the Zlinky from my Home-Assistant, after reconnecting it, the firmware version remains unchanged (v0001) :sob: 

[image]

sdeluce, 2022-02-12:
I tried to update to firmware **003**, with and without the `-e FLASH` flag. Before unplugging the TIC module, I deleted it from Home Assistant.

```shell
sudo ./DK6Programmer -s /dev/ttyUSB0 -V6 -P 115200 -Y -p ZLinky_router_v3.0.bin
```
As before, nothing abnormal in the logs, after pairing the module again, the version remains unchanged (**001**)

If you have any ideas I'm interested.

sdeluce, 2022-03-15:
This issue is similar to #53

closed 2022-03-15 by sdeluce