fairecasoimeme/Zlinky_TIC #51: Zlinky neuf, clignote 8 fois, puis 9 fois, et recommence

T0ytoy, 2022-02-08:
Bonjour, 

Je ne sais pas trop si une issue Github est le bon endroit pour poser la question, n'hésitez pas à me rediriger le cas échéant.
Ma Zlinky toute neuve se connecte sans soucis à Home Assistant, mais elle remonte toujours des valeurs aberrantes et qui ne changent jamais : 

[image]

Connectée au compteur Linky, la ZLinky clignote 8 fois, puis quelques secondes fixe, puis 9 fois, puis fixe à nouveau et recommence. Si j'en crois la documentation, elle devrait soit clignoter 4 fois, soit être tout le temps allumer, donc je ne sais pas quoi en penser?

J'ai également tenté de la mettre à jour avec le firmware 4 : le flash se passe bien sur un rpi 3B+ avec la commande suivante : 
`sudo ./DK6Programmer -s /dev/ttyUSB0 -P 115200 -Y -p ZLinky_router_v4.0.bin ZLinky_router_v4.0.bin`

mais l'affichage dans Home Assistant indique toujours une version 01... Y a-t-il un moyen de vérifier la version du firmware directement via l'utilitaire DK6Programmer par exemple ?
[image]

Est-ce un signe que l'appareil est défectueux, ou bien un problème de mon côté ? Pour info mon Linky est en mode historique, mais ça ne devrait pas poser de soucis.

fairecasoimeme, 2022-02-09:
Bonjour,
Si le ZLinky_TIC ne cesse de clignoter c'est qu'il n'arrive pas à décoder les données du Linky.
Contactez-moi par mail pour un échange standard.
Fred

T0ytoy, 2022-02-09:
@fairecasoimeme mail envoyé hier soir à contact[AT]lixee.fr, merci d'avance ;)

fairecasoimeme, 2022-02-10:
Désolé mais je n'ai rien reçu. Essayez sur contact[at]zigate.fr

T0ytoy, 2022-02-10:
C'est renvoyé, merci ;)

fairecasoimeme, 2022-02-10:
cette fois-ci, bien reçu

T0ytoy, 2022-02-18:
Un A/R par la poste plus tard, le ZLinky fonctionne maintenant correctement et remonte des informations (testé sur 30 minutes pour le moment, sans soucis).
Merci !

closed 2022-02-18 by T0ytoy

dansch1973, 2022-03-13:
Bonjour,
J'ai le même problème. Mon Zlinky neuf, clignote 8 fois, puis 9 fois, et recommence
Cdt,

T0ytoy, 2022-03-13:
@dansch1973 problème réglé pour moi suite à un retour et réparation du zlinky. Il s'agissait à priori d'une diode mal soudée ou disfonctionnelle, je te recommande de contacter @fairecasoimeme ;)