fairecasoimeme/Zlinky_TIC #53: Firmware version with USB TTL

sebPomme, 2022-02-17:
Salutations, 
J'ai un soucis de version avec ma zigate et homeassistant.
Je suis passé de la version  v2, v3,et maintenant v4 (via USB TTL), mais Homeassitant indique : `Firmware: 0x00000001`. 
[image]

Pourtant les cluster 0x0001 & 0x0006 m'indiquent que le firmware est bien à jour.
[image]

[image]

Une idée de comment arranger ça ?

Seb

pdecat, 2022-02-18:
Est-ce que ça aide de lancer une [recherche de mise à jour OTA](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#manually-initiating-ota-firmware-update) ?

```
service: zha.issue_zigbee_cluster_command
data:
  ieee: 00:00:00:00:00:00:00:00 # À mettre à jour
  endpoint_id: 1
  cluster_id: 25
  cluster_type: out
  command: 0
  command_type: client
  args:
    - 0
    - 100
```

sebPomme, 2022-02-18:
> Est-ce que ça aide de lancer une [recherche de mise à jour OTA](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#manually-initiating-ota-firmware-update) ?
>
> ```
> service: [...]

J'ai bien peur que mon installation ne me permette pas de faire de l'OTA pour le moment, j'ai déjà essayé mais les logs de progressions ne dépassent pas les 0.0% (je pense que le soucis provient du fait que mon installation soit dans docker, pure suppositions)

```
2022-02-17 19:18:23 DEBUG (SyncWorker_27) [zigpy.ota.provider] ImageKey(manufacturer_id=4151, image_type=1): /config/zigpy_ota/ZLinky_router_v5.0.ota, version: 5, hw_ver: (None, None), OTA string: OM15081-RTR-JN5189-0000000000000
2022-02-17 19:19:57 DEBUG (MainThread) [zigpy.zcl] [0x8361:1:0x0019] OTA query_next_image handler for 'LiXee ZLinky_TIC': field_control=0, manufacture_id=4151, image_type=1, current_file_version=1, hardware_version=None, model=ZLinky_TIC
2022-02-17 19:19:57 DEBUG (MainThread) [zigpy.zcl] [0x8361:1:0x0019] OTA image version: 5, size: 240478. Update needed: True
2022-02-17 19:19:57 DEBUG (MainThread) [zigpy.zcl] [0x8361:1:0x0019] OTA image_block handler for 'LiXee ZLinky_TIC': field_control=0, manufacturer_id=4151, image_type=1, file_version=5, file_offset=0, max_data_size=48, request_node_addr=Noneblock_request_delay=None
```

Donc je ne peux pas répondre à ta question

pdecat, 2022-02-18:
L'idée n'était pas de vraiment faire l'OTA via ce service, mais juste de lancer un check, et voir si ça rafraîchissait la valeur.

Mais dans la log, on voit qu'il y a bien un problème avec ce que retourne le ZLinky: `current_file_version=1`

sebPomme, 2022-02-18:
@pdecat autant pour moi. Les logs précédents date de quand mon zlinky était en v3.
J'ai fait le test et je peux confirmer qu'en v5 c'est toujours pareil:
```
2022-02-18 18:30:11 DEBUG (MainThread) [zigpy.zcl] [0x8361:1:0x0019] OTA query_next_image handler for 'LiXee ZLinky_TIC': field_control=0, manufacture_id=4151, image_type=1, current_file_version=1, hardware_version=None, model=ZLinky_TIC
2022-02-18 18:30:11 DEBUG (MainThread) [zigpy.zcl] [0x8361:1:0x0019] No OTA image is available
```

sdeluce, 2022-02-19:
Hola,
J'ai exactement le même problème, j'ai ouvert l'issue #50 sur le problème que je rencontre pour la mise à jour via le TTL quelque soit la version de firmware, en voyant cette issue, j'ai fait le test via l'OTA et j'ai exactement la même chose que @sebPomme.

```log
2022-02-19 15:37:28 DEBUG (MainThread) [zigpy.zcl] [0x4a33:1:0x0019] OTA query_next_image handler for 'LiXee ZLinky_TIC': field_control=0, manufacture_id=4151, image_type=1, current_file_version=1, hardware_version=None, model=ZLinky_TIC
[...]
2022-02-19 15:37:28 DEBUG (MainThread) [zigpy.zcl] [0x4a33:1:0x0019] OTA image version: 5, size: 240478. Update needed: True
2022-02-19 15:37:28 INFO (MainThread) [zigpy.zcl] [0x4a33:1:0x0019] Updating: LiXee ZLinky_TIC
[...]
2022-02-19 15:37:29 DEBUG (MainThread) [zigpy.zcl] [0x4a33:1:0x0019] OTA upgrade progress: 0.0
```

sebPomme, 2022-02-19:
@sdeluce j'aurais probablement du vérifier qu'une issue ne soit pas déjà ouverte, mes excuses. mes excuses pour ça.

En passant, petite aparté j'ai utilisé cette astuce https://github.com/fairecasoimeme/Zlinky_TIC/issues/18#issuecomment-1012368631 pour pouvoir mettre à jour la dernière information présente sur votre capture d'écran. Ça permet à d'indiquer à home-assistant d'aller questionner manuellement zlinky pour récupérer l'information (et ça permet de faire fonctionner les fonctionnalités `ENERGY` d'home-assistant).

---
Quelqu'un à des idées pour corriger le souci de version de firmware ?

sdeluce, 2022-03-15:
@sebPomme, merci pour l'astuce.
Pour le numéro de version, sur un autre module TIC commandé resemant, après mise à jour via OTA (car il n'y a pas de prise TTL sur ce dernier) le numéro de version affiché est bien celui de la version installée.

JeroLajHub, 2022-04-15:
Bonjour, je suis dans exactement la même situation avec un ZLinky de première génération acheté mi novembre 2021 et que j'ai initialisé seulement maintenant. 

J'ai pu faire une mise à jour OTA directement en V5 avec TTL, après des tentatives par OTA infructueuses, et je suis maintenant dans la même situation que signalé dans cette conversation.

https://github.com/fairecasoimeme/Zlinky_TIC/issues/61#issue-1200434787

sebPomme, 2022-05-12:
Pour résumer.
Mise à jour via USB-TTL: fonctionnelle mais le numéro de version n'est pas mit à jour.
Mise à jour via OTA : Tout va bien

@fairecasoimeme il y a des chances pour que les deux firmware ne soient pas construit de la même façon ce qui expliquerait le problème ?

fairecasoimeme, 2022-05-13:
je vais faire des tests et je reviens vers vous

fairecasoimeme, 2022-05-13:
ma procédure : 
J'ai flashé un ZLinky_tic en v4. Je l'ai appairé et demandé une requête au coordinateur pour récupérer la version (Cluster 0x0000 / propriété 0x0001). Il me retourne la valeur 4
J'ai ensuite téléchargé le firmware (v5 binaire) sur le github. J'ai flashé le firmware puis reappairé au coordinateur. J'ai relancé la requête pour récupérer la version et il me retourne bien 5

fairecasoimeme, 2022-05-13:
> Bonjour, je suis dans exactement la même situation avec un ZLinky de première génération acheté mi novembre 2021 et que j'ai initialisé seulement maintenant.
> 
> J'ai pu faire une mise à jour OTA directement en V5 avec TTL, après des tentatives par OTA infructueuses, et je suis maintenant dans la même situation que signalé dans cette conversation.
> 
> [#61 (comment)](https://github.com/fairecasoimeme/Zlinky_TIC/issues/61#issue-1200434787)
> 
> PS : avant la mise à jour du firmware, le ZLinky me remontait le même type de données parce que le compteur était en MODE TIC HISTORIQUE. Une fois passé en MODE TIC STANDARD il n'y avait plus de valeurs négatives.
Je pense qu'il y a un problème de rafraichissement côté HA, si la requête pour demander la version fonctionne avec l'outil de cluster.

mospaeda, 2022-10-18:
Idem pour moi : Achat récent en version 9

ZLinky_TIC
par LiXee
Connecté via [Zigbee Coordinator]
Firmware: 0x00000001

From the cluster details it seems that it is different (in my case firmware 9 as I received last week)
reference from offcial doc : https://github.com/fairecasoimeme/Zlinky_TIC#clusters

Basic (Endpoint id: 1, Id: 0x0000, Type: in)
Valeur 9

sw_build_id (id: 0x4000)
4000-0009

date_code (id: 0x0006)
20220930

I don't know from where the value 0x00000001 is coming but it is probably not pointing to right register

sebPomme, 2023-05-10:
Je suis passé sous z2mqtt et je ne suis plus confronté au souci. Je ferme l'issue. Merci à tous

closed 2023-05-10 by sebPomme