fairecasoimeme/Zlinky_TIC #54: unsupported device with Zigbee model 'ZLinky_TIC' and manufacturer name 'LiXee'

almostlunatic, 2022-02-22:
Hi,

Today, I received the long waited module for my Linky and pairing was successful. It appeared on Zigbee2MQTT (v.1.17.1-dev (zStack 12 20180815) (commit #084d6ba)) on my Domoticz v.2021.1 (build 14117) on an Intel NUC with Mint 20.3 as follows:

[image]

Clicking on it, no devices found:

[image]

On the Devices list with filtering, only one entry has appeared, and it's addition to the (I don't know why preseted) api_transort and includes the next info:

`{"dateCode": "20220117",
"description": "-", "friendly_name": "0x00158d0005d29a5e",
"hardwareVersion": 1, "ieeeAddr": "0x00158d0005d29a5e", "lastSeen":
1645537055443, "manufacturerID": 4151, "manufacturerName": "LiXee",
"model": "ZLinky_TIC\u0000", "modelID": "ZLinky_TIC\u0000",
"networkAddress": 47337, "powerSource": "Mains (single phase)",
"softwareBuildID": "4000-0004", "type": "Router", "vendor": "-"}`

When logging into Linux, the warnings for the Zegbee2MQTT are shown periodically:

`Zigbee2MQTT:warn  2022-02-22 14:42:49: Received message from unsupported device with Zigbee model 'ZLinky_TIC' and manufacturer name 'LiXee'`
`Zigbee2MQTT:warn  2022-02-22 14:42:49: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

All stacks are loaded successfully and are working:

`Zigbee2MQTT:info  2022-02-01 23:48:38: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-02-01.23-48-31' filename: log.txt`
`Zigbee2MQTT:info  2022-02-01 23:48:39: Starting Zigbee2MQTT version 1.17.1-dev (commit #084d6ba)`
`Zigbee2MQTT:info  2022-02-01 23:48:39: Starting zigbee-herdsman (0.13.68)`
`Zigbee2MQTT:info  2022-02-01 23:48:45: zigbee-herdsman started`
`Zigbee2MQTT:info  2022-02-01 23:48:45: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20180815,"transportrev":2},"type":"zStack12"}'`
`Zigbee2MQTT:info  2022-02-01 23:48:46: Currently 14 devices are joined:`

What could be a problem?

fairecasoimeme, 2022-02-22:
Hi,
Please see this :
https://gist.github.com/ralmn/02de46e193bf111062f15be07d3e9bd2

almostlunatic, 2022-02-22:
Thank you @fairecasoimeme !

I read from the first link:
>
>1. Dans le dossier de votre configuration de zigbee2mqtt, creer un dossier avec converters
>
>2. Placer le fichier lixee.js dans le dossier converters
>3. Dans le fichier configuration.yaml ajouter:
>`external_converters:`
>`  - converters/lixee.js`
>4. Relancer zigbee2mqtt

Where could I get the file "lixee.js" and what shoul it continue?

almostlunatic, 2022-02-22:
Ok, but how about updating zigbee-herdsman to include your pull request? The mine is 0.13.68, the newest is 0.14.18.

antoinelibert, 2022-02-22:
ZLinky is supported natively by Z2M starting from version 1.23.x-DEV. Just upgrade Z2M to the latest 1.23.x series DEV version and you should have everything working right of the box.

almostlunatic, 2022-02-24:
Thank you vey much for your time and assistance! 

What I did was 
`git checkout HEAD -- npm-shrinkwrap.json`
`npm ci`

Among the results, I had many errors like:
`npm WARN EBADENGINE Unsupported engine {`
`npm WARN EBADENGINE   package: '@eslint/eslintrc@1.1.0',`
`npm WARN EBADENGINE   required: { node: '^12.22.0 || ^14.17.0 || >=16.0.0' },`
`npm WARN EBADENGINE   current: { node: 'v12.18.2', npm: '8.4.0' }`
`npm WARN EBADENGINE }`
`npm WARN EBADENGINE Unsupported engine {`
`npm WARN EBADENGINE   package: '@typescript-eslint/eslint-plugin@5.12.0',`
`npm WARN EBADENGINE   required: { node: '^12.22.0 || ^14.17.0 || >=16.0.0' },`
`npm WARN EBADENGINE   current: { node: 'v12.18.2', npm: '8.4.0' }`

And so 14 warnings. After it, I ran:
`nvm install 12.22.0`

Then I repeated `npn ci` and this time with no errors. So I decided to upgrade npm from 6.14.11 → 8.5.1:
`npm install -g npm`

And as result, I got no errors from ZLinky:

`Zigbee2MQTT:info  2022-02-24 20:57:37: MQTT publish: topic 'zigbee2mqtt/0x00158d0005d29a5e', payload '{"active_enerfy_out_d01":18870,"active_enerfy_out_d02":15257,"active_power":296,"active_power_max":1910,"active_power_ph_b":322,"apparent_power":278,"apparent_power_ph_b":65535,"available_power":9,"average_rms_voltage_meas_period":236,"current_date":"H220224205702","current_index_tarif":1,"current_price":"BASE","current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier3_summ_delivered":1941,"current_tier4_summ_delivered":2471,"current_tier5_summ_delivered":389,"current_tier6_summ_delivered":488,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"drawn_v_a_max_n1":2010,"linkquality":42,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"021775131322","power_threshold":9,"rms_current":1,"rms_current_ph_b":65535,"rms_current_ph_c":65535,"rms_voltage":235,"site_id":"24253690259961","software_revision":2,"status_register":"003A0001","update":{"state":"available"},"update_available":true}'
`

Now I don't have so many errors on Z2M output, only this one time after time:
`Zigbee2MQTT:error 2022-02-24 21:01:40: Failed to configure '0x00158d0005d29a5e', attempt 2 (Error: ConfigureReporting 0x00158d0005d29a5e/1 liXeePrivate([{"attribute":"activeEnerfyOutD01","minimumReportInterval":60,"maximumReportInterval":900,"reportableChange":100}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'SOFTWARE_FAILURE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22))`

I read from @fairecasoimeme links that it's necessary to replace poll interval in lixee.js from 120 to 60, was it wrong? There was 2 places to chenge it:
`const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 60;`

As a result, I have now one device appaired for the HW ZLinky_TIC:

[image]

But not any other attributes to be subscribed. What should I do next?

vk496, 2022-02-28:
Hello @almostlunatic 

I'm the guy who spend my money and time to make a great support for this device (patching Z2M project, OTA, non-standard clusters of this device, etc). Currently all my changes are in DEV. I suggest you to wait for tomorrow and install the stable version 1.24.0

Your problem for SOFTWARE_FAILURE probably come from here: https://github.com/fairecasoimeme/Zlinky_TIC/issues/40

almostlunatic, 2022-02-28:
Thank you @vk496! I'm very grateful for your efforts, I hope that it will wotk out!

almostlunatic, 2022-03-04:
Hi @vk496,

I updated to v 1.24.0, but nothong has changed.

I still have only the Ling Quality on the devices list for ZLink_TIC:

[image]

And errors on Logs:

` 2022-03-04 15:25:42.037 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.037 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.037 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "NTARF"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "VTIC"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EAST"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF01"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF02"
2022-03-04 15:25:42.038 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF03"
2022-03-04 15:25:42.039 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF04"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF05"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF06"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF07"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF08"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF09"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASF10"
2022-03-04 15:25:42.040 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASD01"
2022-03-04 15:25:42.043 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASD02"
2022-03-04 15:25:42.043 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASD03"
2022-03-04 15:25:42.043 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "EASD04"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "IRMS1"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "URMS1"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "PREF"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "PCOUP"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "CCASN"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "CCASN-1"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "UMOY1"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "SINSTS"
2022-03-04 15:25:42.044 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "SMAXN"
2022-03-04 15:25:42.045 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "SMAXN-1"
2022-03-04 15:25:42.045 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.045 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.048 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.048 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "DPM1"
2022-03-04 15:25:42.048 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "FPM1"
2022-03-04 15:25:42.048 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "DPM2"
2022-03-04 15:25:42.048 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "FPM2"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "DPM3"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "FPM3"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "RELAIS"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "NJOURF"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process numeric item "NJOURF+1"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text"
2022-03-04 15:25:42.049 Error: ZigbeeHW: 0x00158d0005d29a5e: can not process feature type "text" `

What shlud I do next?

vk496, 2022-03-04:
Did you removed the old external-converter?
Also, I suggest to manually set the device config (specific options) and restart Z2M to take effect, since auto detection is not fully implemented

almostlunatic, 2022-03-04:
I always restart Z2M after any change, but never manually set the device config... Can you, please, remember me, how to do it, and how to remove the old external-converter and having the new one? My head seems to blow up with everythin, because there are many other issues too to reslove...

vk496, 2022-03-04:
well, I don't know what exactly do you use, since I don't recognize that Z2M frontend. But basically, you had to undo https://github.com/fairecasoimeme/Zlinky_TIC/issues/54#issuecomment-1047929265 

And set the specifc option in the tab near to other options like Binding, Reporting or Exposes

almostlunatic, 2022-03-04:
Ok, I stopped Z2M, renamed lixee.js to lixee.js.old, removed
`external_converters:`
`- converters/lixee.js`

from the configuration.yaml and restarted Z2M. But all is the same!

almostlunatic, 2022-03-04:
Zlincy TIC sends always correct data, but Z2M doesn't seem to recognize it.

`Zigbee2MQTT:info  2022-03-04 16:14:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0005d29a5e', payload '{"active_enerfy_out_d01":18910,"active_enerfy_out_d02":15257,"active_power":164,"active_power_max":700,"active_power_ph_b":190,"apparent_power":221,"apparent_power_ph_b":65535,"available_power":9,"average_rms_voltage_meas_period":233,"current_date":"H220304161347","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":34167,"current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier4_summ_delivered":2471,"current_tier5_summ_delivered":389,"current_tier6_summ_delivered":488,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"drawn_v_a_max_n1":1580,"linkquality":44,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"021775131322","power_threshold":9,"rms_current":1,"rms_current_ph_b":65535,"rms_current_ph_c":65535,"rms_voltage":234,"site_id":"24253690259961","software_revision":2,"status_register":"003A0001","update":{"state":"idle"},"update_available":false}'
`

vk496, 2022-03-04:
What I see there is a zlinky working correctly.... Maybe you just need to set your specific config

almostlunatic, 2022-03-04:
I don't know how to set the specific config :-(

antoinelibert, 2022-03-04:
I think there is some misunderstanding here. @almostlunatic seems to use Domoticz (not Home Assistant) with Z2M. While HomeAssistant+Z2M+ZLinky is fully compatible, I'm not sure it works with Domoticz+Z2M+ZLinky. My guess is that you need support for ZLinky for Domoticz too? You should ask on the Domoticz project.

antoinelibert, 2022-03-04:
Ok, I have looked around and it seems like Domoticz doesn't support ZLinky via Z2M yet. Someone else has the same issue: https://github.com/domoticz/domoticz/issues/5183.

almostlunatic, 2022-03-04:
Yes, thank you! That's cpuld be the problem, of course! There are some blogs that assure that ZLinky is supported for Domoticz with Zigate. I have one, so I will give it a try. And let you know. Thank you take my attention on it!

almostlunatic, 2022-03-07:
So, after installing ZiGate+, ZLinky started to add some properties to domoticz. Although, not everything and not very well, but still. So, thank you very much!

hergonoway, 2022-03-10:
Hello, currently using HomeAssistant+Z2M+ZLinky 

On my Linky, contract is set to "HC SEM WE MERCR", TIC is set to "HISTORIQUE" and all infos on my Linky are split between HC and HP, for Semaine, Mercredi and WE. My contract is "Mes Jours Zen Plus" by EDF

but whatever  I choose in device config (specific options), I always end up with : 
OPTARIF : BASE
PAPP : <the right value>
BASE : <which correspond to 'INDEX MID' on my Linky>
ISOUSC : 30 (<- instead of 6 on my Linky)

every other info like, HP Semaine, HC semaine, HP WE, HC WE, or any other split value from my linky doesn't show up anywhere whatever the device config I choose.

I've tried to avoid any 'auto' config my choosing "historique", "single_phase" and "false" (production), but the result is the same.

vk496, 2022-03-10:
Set your current TIC mode manually (mode,phase,production) in specific options and restart Z2M. I suppose you're looking for current_price, which tells you if you are in low or high prices. I don't remember if that attribute is available in historique (you could try setting up standard just for test purpose and see if you have that info, in which case the code should be fixed). If is not available, that means a limitation from the mode itself, and you need call/request your energy provider in order to switch mode to standard (you can't request it directly to Enedis via web)

hergonoway, 2022-03-10:
> Set your current TIC mode manually (mode,phase,production) in specific options and restart Z2M. I suppose you're looking for current_price, which tells you if you are in low or high prices. I don't [...]

already tried all the modes in auto or manually set on mode, phase and production, and ofc with a restart of Z2M each time -> same result.

I'm not looking for "current_price" especially, but I'd like to be able to monitor my consumption in HC/HP during the week, HC/HP on wednesday and HC/HP during the weeking. Currently I only have the sumup of everything...

changing TIC mode has to be request by  EDF to Enedis irrc and they would certainly charge me for it... but if I need to switch to Standard TIC in order to gather data from an "HC SEM WE MERCR" contrat, I'd do it.

any other thing to try before trying to change TIC mode? I'd like to be sure that nothing's wrong with my Lixee nor Z2M before doing that.

vk496, 2022-03-10:
Take in account that the tarif mode is just a filter to reduce useless attributes for your mode (if I'm in HC/HP, I don't care about colour days for example).

In your case, could you please setup all your config manually (without the tarif filter) and share some screenshot of what you have and what you miss exactly? Thnks

hergonoway, 2022-03-10:
> Take in account that the tarif mode is just a filter to reduce useless attributes for your mode (if I'm in HC/HP, I don't care about colour days for example).
>
> In your case, could you please setup [...]

all in manual without tarif filter like : 
linky_mode : historique
energy_phase : single_phase
production : false
tarif : auto

?

vk496, 2022-03-10:
Ok, what you're looking for is current_tierX values, which are 0 on your side. In your Linky screen you can see that values? Do you have them in the reportings section (subscription)? If you try to read them manually with the console tab, they still reports 0?

hergonoway, 2022-03-10:
> Ok, what you're looking for is current_tierX values, which are 0 on your side. In your Linky screen you can see that values? Do you have them in the reportings section (subscription)? If you try to read them manually with the console tab, they still reports 0?

**In your Linky screen you can see that values?**
--> yes , I've got the following screens : 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
(in that order after the contrat type "HC SEM WE MERCRE")

**Do you have them in the reportings section (subscription)?**
 --> no (never touched that section)
[image]

but if you meant if there're available, it seems to be the case for the cluster seMetering (I didn't add any, I've just made a screenshot)
[image]

**If you try to read them manually with the console tab, they still reports 0?** 
--> not for current_tier**1**SummDelivered but yes from current_tier**2**SummDelivered to current_tier**10**SummDelivered : 
`Read result of 'seMetering': {"currentTier1SummDelivered":[0,11960503],"currentTier2SummDelivered":[0,0],"currentTier3SummDelivered":[0,0],"currentTier4SummDelivered":[0,0],"currentTier5SummDelivered":[0,0]...`

From current_tier**11**SummDelivered up to current_tier**15**SummDelivered it returns:
`failed (Status 'UNSUPPORTED_ATTRIBUTE')`
same error goes for all current_tierXSumm**Received**. (Received not Delivered)

fairecasoimeme, 2022-03-11:
Hi,
You have to change to the standard mode. The Historic mode is not compatible with this kind of  subscription. Historic mode is an old mode and only BASE, HCHP et tarif bleu is compatible.

Fred

hergonoway, 2022-03-11:
> Hi, You have to change to the standard mode. The Historic mode is not compatible with this kind of subscription. Historic mode is an old mode and only BASE, HCHP et tarif bleu is compatible.
> 
> Fred

thanks for the feedback, I'll ask to change TIC mode then. Maybe you could add this to the main page, it may help others with exotic subscription like me