fairecasoimeme/Zlinky_TIC #55: 4 commandes seulement pour le ZLinky TIC avec Zigate sur Jeedom

echaplet, 2022-02-25:
Bonjour,
ayant acheté récemment un ZLinky TIC appairé à une Zigate, je n'obtiens que 4 commandes dispo.
- manufacturer
- type
- current_delivered
- refresh

Je ne vois pas ce qui cloche, c'est décevant pour un produit à 50€...
Un peu d'aide serait le bienvenue, merci !
Eric

https://community.jeedom.com/t/peu-de-commandes-pour-le-zlinky-tic-souci-matos-ou-plugin/78435

fairecasoimeme, 2022-02-28:
Bonjour,

Sur jeedom, il faut utiliser le plugin officiel zigbee. 
Le plugin ZiGate n'est plus maintenu par son développeur. (Je ne suis pas le développeur)
Il y a aussi zigbee2mqtt avec zigbeelinker il me semble.
Voir peut-être avec la communauté Abeille pour voir la compatibilité avec le plugin

Fred

echaplet, 2022-03-10:
fonctionnement ok avec le plugin officiel zigbee et la zigate mise à jour en OPDM

closed 2022-03-10 by echaplet