fairecasoimeme/Zlinky_TIC #56: Appairage impossible

elstringer, 2022-02-26:
Bonjour, 
J’utilise une plateforme ‘zigbee smart gateway’

Il m’est impossible de la relié avec le module sur mon compteur linky 
Aucune reconnaissance 

[image]

Pouvez-vous me dirigé sur du matériel comptable ? 

mon compteur ce trouve à 250m de la maison la plateforme zigbee est alimentée en wifi a 2m du compteur 

mon serveur Domotique HA ce  trouve à 300m sur une machine dédiée fibré  

Je ne vois que le module zigate pour palier à mon problème cela serais pratique de pouvoir utilisée des plateformes prête a l’emploi

fairecasoimeme, 2022-02-28:
Bonjour, 
Le module ZLinky_TIC est compatible avec les plateformes : Jeedom, Domoticz, Home-assistant, EEdomus. Je ne sais pas si votre passerelle est compatible avec ces plateformes.
Si votre compteur Linky se trouve à 250 mètres de la passerelle Zigbee, ça ne sera pas possible.
Je ne comprends pas très bien votre architecture.

Fred

elstringer, 2022-03-09:
J’utilise home- assistant !
Mon compteur est à cotée d’un de mes routeurs wifi  et la box zigbee est branchée sur ce routeur ! 

Mon instance home assistant est sur une vm dans la maison, 

La question est juste savoir si ma passerelle est compatible avec le module sur le linky

fairecasoimeme, 2022-03-09:
Ok pour l'architecture.

Vous utilisez quoi pour piloter votre passerelle zigbee sur home assistant ? ZHA ? zigbee2mqtt ? ou un plugin propriétaire à la passerelle ?

le module est compatible avec les "plugins" libres et ouverts (ZHA / Zigbee2mqtt / Deconz)