fairecasoimeme/Zlinky_TIC #58: Problème de ré-appairage - finit par réussir - erreurs protocole

mdeweerd, 2022-03-19:
Je relate ma problématique d'appairage du ZLinky avec des éléments techniques .
Alors que j'étais en train de décrire ceci, l'appairage a miraculeusement réussi.

# Problématique d'appairage

Après une première association, une mise à jour et une suppression du ZLinky du réseau pour forcer un nouvel appairage, je ne parvenais plus à appairer.

Visiblement il assurait encore un rôle de Routeur (observé par sniffer), mais les tentatives d'appairage n'ont pas menés à une ré-intégration au niveau de ZHA, ou c'est zigpy qui contrôle un CC2531 avec un ZStack 3.0 de KKoenk.

Je n'ai pas ce genre de comportement avec d'autres objets qui ont subit le même traitement, mais pour ces objets je peux initier un appairage par appui d'un bouton quelconque, alors que le ZLinky n'en a pas.

Historique des opérations depuis la réception du ZLinky:

- Installation sur Linky - OK;
- Association dans HA - OK;
- "Scan device" avec zha-toolkit - semi-ok car pas de réponse à Discover Attributes, Discover Commands;
- Constat que la version était à 01 - expliquant éventuellement les valeurs impossibles : >65000V, -32000W, etc.
- Mise à jour par OTA - OK
- La version était confirmé à 05 avec une date_code en février, mais toujorus des valeurs impossibles.
- Enlèvmeent et remise du ZLinky dans le Linky pour provoquer un redémarrage - OK
- Lz ZLinky est à nouveau interrogéable, mais pas d'évolution au niveau des valeurs "extrêmes".
- Je constate bon nombre de requêtes Zigbee de type "Configure Reporting" où les réponses du ZLinky sont défavorables (la configuration n'est pas acceptée).
- Le "Scan device" avec zha-toolkit ne donne toujours pas la liste des attributs - semi-ok (pas d'erreurs, mais pas le détail des attributs);
- Je décide de tenter une commande "Factory Reset" (=cluster 0/commande 0).  Le ZLinky confirme la commande ("Success"), mais je ne constate aucun effet: le ZLinky est toujours dans le réseau et son comportement semble le même.
- Pour le réassocier, sous Homa Assistante je fais "Remove Device".  Le ZLinky disparrait de la liste des objets Zigbee associés.

La dernière information que j'ai dans la trace de HomeAssistant pour le ZLinky est (après le factory reset):

```
2022-03-19 00:51:55 DEBUG (MainThread) [custom_components.zha_toolkit.utils] registry_device DeviceEntry(area_id='cuisine', config_entries={'e876e09df16c5319d7a09650f914276c'}, configuration_url=None, connections={('zigbee', '00:15:8d:00:05:d2:99:ea')}, disabled_by=None, entry_type=None, id='e36d5820afc30475064d5abe45afaa6e', identifiers={('zha', '00:15:8d:00:05:d2:99:ea')}, manufacturer='LiXee', model='ZLinky_TIC', name_by_user=None, name='LiXee ZLinky_TIC', suggested_area=None, sw_version='0x00000005', hw_version=None, via_device_id='26851989eb8070e67ce5d95a26c6a241', is_new=False)
```

Dans les traces de mon sniffer j'ai bien une demande "Leave":
```
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
            ...0 1... = Key Id: Network Key (0x1)
            ..1. .... = Extended Nonce: True
        Frame Counter: 26168
        Extended Source: Jennic_00:05:d2:99:ea (00:15:8d:00:05:d2:99:ea)
        Key Sequence Number: 0
        Message Integrity Code: a15075a9
        [Key: XXXXX]
        [Key Label: 2]
    Command Frame: Leave
        Command Identifier: Leave (0x04)
        ..0. .... = Rejoin: False
        .0.. .... = Request: False
        0... .... = Remove Children: False
```

Et une confirmation:
```
ZigBee Application Support Layer Command
    Frame Control Field: Command (0x01)
        .... ..01 = Frame Type: Command (0x1)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Counter: 96
    Command Frame: Update Device
        Command Identifier: Update Device (0x06)
        Device Address: Jennic_00:05:d2:99:ea (00:15:8d:00:05:d2:99:ea)
        Device Address: 0x145d
        Device Status: Device left (0x02)
```

Juste après je vois une demande d'association, confirmée par le coordinateur

```
    Sequence Number: 131
    Destination PAN: 0x880a
    Destination: Jennic_00:05:d2:99:ea (00:15:8d:00:05:d2:99:ea)
    Extended Source: SiliconL_ff:fe:6d:7f:8a (bc:33:ac:ff:fe:6d:7f:8a)
    Command Identifier: Association Response (0x02)
    Association Response
        Short Address: 0x7546
        Association Status: 0x00 (Association Successful)
    Frame Check Sequence (TI CC24xx format): FCS OK
        RSSI: 20dB
        FCS Valid: True
        LQI Correlation Value: 107
```

Et c'est suivi par:
```
    ZigBee Security Header
        Security Control Field: 0x20, Key Id: Link Key, Extended Nonce
            ...0 0... = Key Id: Link Key (0x0)
            ..1. .... = Extended Nonce: True
        Frame Counter: 90580
        Extended Source: SiliconL_ff:fe:6d:7f:8a (bc:33:ac:ff:fe:6d:7f:8a)
        Message Integrity Code: 4c120659
        [Key: 5a6967426565416c6c69616e63653039]
        [Key Label: Std Trust Center link key]
    Command Frame: Update Device
        Command Identifier: Update Device (0x06)
        Device Address: Jennic_00:05:d2:99:ea (00:15:8d:00:05:d2:99:ea)
        Device Address: 0x7546
        Device Status: Standard security, unsecured join (0x01)
```

C'est le même type de paquet et la même addresse courte que j'observe se matin: "Update Device".
Je pense que le problème est là:
1. Le "Device Announement" est absent, c'est un élément important du processus d'appairage.
2. La clef standard est utilisé pour la commande "Update Device".

Lors de la première Association il y a bien eu un Device Announcement, et ce avec la clef dédié au réseau:
```
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
    Frame Control Field: Data (0x08)
        .... ..00 = Frame Type: Data (0x0)
        .... 10.. = Delivery Mode: Broadcast (0x2)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 0
    Device Announcement (Cluster ID: 0x0013)
    Profile: ZigBee Device Profile (0x0000)
    Source Endpoint: 0
    Counter: 229
ZigBee Device Profile, Device Announcement, Nwk Addr: Jennic_00:05:d2:99:ea
    Sequence Number: 165
    Nwk Addr of Interest: 0x145d
    Extended Address: Jennic_00:05:d2:99:ea (00:15:8d:00:05:d2:99:ea)
    Capability Information: 0x8e
        .... ...0 = Alternate Coordinator: False
        .... ..1. = Full-Function Device: True
        .... .1.. = AC Power: True
        .... 1... = Rx On When Idle: True
        .0.. .... = Security Capability: False
        1... .... = Allocate Short Address: True
```

~Me voilà don très embêté pour utiliser ce poroduit.~

Alors que j'étais en train d'associer un autre appareil qqs heures plus tard, le Device Announcement s'est mystérieusement réalisé:
```
ZigBee Device Profile, Device Announcement, Nwk Addr: Jennic_00:05:d2:99:ea
    Sequence Number: 70
    Nwk Addr of Interest: 0x06a7
    Extended Address: Jennic_00:05:d2:99:ea (00:15:8d:00:05:d2:99:ea)
    Capability Information: 0x8e
        .... ...0 = Alternate Coordinator: False
        .... ..1. = Full-Function Device: True
        .... .1.. = AC Power: True
        .... 1... = Rx On When Idle: True
        .0.. .... = Security Capability: False
        1... .... = Allocate Short Address: True

```

lui même apparu après une "Association Request" suivi d'une "Association Response".

C'est pas l'idéal tout cela, mais au moins je ne suis pas bloqué.

J'imagine qu'à l'aide d'une trombone j'aurai pu forcer une réinitialisation à travers le trou le plus près des broches.

fairecasoimeme, 2022-03-21:
Bonjour,

Vous avez effectuer un leave avec l'option not rejoin ce qui explique que le module ne rejoigne pas le coordinateur.
Pour remettre à zéro le ZLinky_TIC il faut tout simplement appuyer 10 sec sur le bouton. l'appairage devient alors possible.

closed 2022-03-21 by fairecasoimeme

mdeweerd, 2022-03-21:
C'est peut-être une explication, mais ce n'est pas comme cela que le module devrait fontionner:
- Il n'a pas quitté le réseau;
- Il continuait à jouer son rôle de routeur;
- La clef secrete - au moins pour la commande update_device - est incorrect.