fairecasoimeme/Zlinky_TIC #59: Certaines valeurs ne remontent pas

batchmanfr, 2022-03-29:
Bonjour,

En mode Standard Triphasé, je constate que seules certaines valeurs remontent dans Home Assistant (via Z2M) :
[image]

Donc aucune info sur la consommation en kWh... et par conséquent pas de suivi de la consommation dans l'onglet Energie de Home Assistant donc, même si j'imagine qu'une bidouille dans le style de ce que propose @skanx pour sha doit être possible (https://github.com/fairecasoimeme/Zlinky_TIC/issues/18#issuecomment-1021186192)

En cherchant bien, j'ai réussi à trouver la consommation totale dans les attributs d'états des différentes entités (exemple ici pour l'entité IRMS1) :
[image]
Mais l'info n'est pas identifiée comme entité à part entière, et surtout, je n'ai aucun détail (je suis en HP/HC, je devrais donc avoir selon la doc EASF01 pour le compteur HP et EASF02 pour le compteur HC)

Une idée de ce qui bloque ? Un paramétrage de la Zlinky ou du module Zigbee2MQTT?

Merci !

fairecasoimeme, 2022-03-30:
Bonjour,
L'information EASF01 et EASF02 est nommé : 
currentTier1SummDelivered et currentTier2SummDelivered dans Z2M

vous pouvez vérifier si ces valeurs existent dans votre fichier.

Dans tous les cas cela ne concerne pas le ZLinky mais plutôt le module Zigbee2MQTT 

@vk496 une idée ?

vk496, 2022-03-30:
@batchmanfr could you check what is exposed in Z2M (not Home Assistant)?

Standard+Triphase should report EASFXX as attribute

vk496, 2022-03-30:
btw, did you forced the zlinky config acording to your linky and rebooted Z2M?

batchmanfr, 2022-03-30:
Thx @fairecasoimeme @vk496 for your answer.

I've forced the zlinky config his way :
[image]
Btw I've tried "auto" configuration and the values that show up in Z2M are the same.

This is what I see on the z2m logs :
`MQTT publish: topic 'zigbee2mqtt/ZLinky', payload '{"active_power_max":3599,"active_power_max_ph_b":4402,"active_power_max_ph_c":4162,"apparent_power":133,"apparent_power_ph_b":231,"apparent_power_ph_c":155,"average_rms_voltage_meas_period":235,"average_rms_voltage_meas_period_ph_c":234,"average_rms_voltage_measure_period_ph_b":235,"current_summ_delivered":48825.2,"current_tarif":"H PLEINE/CREUSE","drawn_v_a_max_n1":3465,"drawn_v_a_max_n1_p2":2017,"drawn_v_a_max_n1_p3":2979,"linkquality":90,"rms_current":1,"rms_current_ph_b":1,"rms_current_ph_c":1,"rms_voltage":236,"rms_voltage_ph_b":235,"rms_voltage_ph_c":233,"status_register":"003A4401","update":{"state":"idle"},"update_available":false}'`

So I've a `current_summ_delivered` value (that is not exposed in Z2M), but no EASFXX attribute :
[image]
[image]

batchmanfr, 2022-04-01:
Hi,
This issue have been resolved with Z2M version 1.25.0-1 released today.
Seems to be the @SilenT-FR pulls on zigbee-herdsman-converters repository that solved this :
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3946
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3950
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3951

Thank you for your help. 

A bientôt !
Florian

closed 2022-04-01 by batchmanfr

SilentT19, 2022-04-02:
your welcome

SilentT-FR, 2022-04-02:
your welcome