fairecasoimeme/Zlinky_TIC #60: Deconnexion du Zlinky_TIC de mon réseau Zigbee

gd35, 2022-04-06:
Bonjour,

Je l'utilise avec un dongle USB Conbee depuis quelques semaines. J'ai pu le connecter avec mon réseau Zigbee sans problème et ai configuré Deconz de façon a récupérer les données. Tout fonctionnait correctement jusqu'à il y a quelques jours. Mon ZLinky_TIC est complètement déconnecté du réseau Zigbee et donc ne remonte plus les données. La lampe bleu reste allumée mais plus de communication. Y-a-t-il une solution pour éviter cette déconnexion ?

D'avance merci.

fairecasoimeme, 2022-04-06:
Bonjour,
Est-ce qu'il se reconnecte lorsque vous le débranchez puis rebranchez physiquement ?

gd35, 2022-04-08:
Bonjour,

Oui, je l'ai fait hier et il s'est reconnecté après à mon réseau. J'ai constaté parfois que pendant plusieurs heures les infos ne remontaient pas. Voir ici: https://forum.phoscon.de/t/lixee-zlinky-tic-ddf-file/1164/60

Je ne sais pas si cela vient de ma configuration avec Deconz. Sinon, produit super qui remplace mon Ecodevice avec sans aucun doute une consommation moindre.

fairecasoimeme, 2022-04-12:
Bonjour, 
de quand date votre achat ? pouvez-vous me contacter par mail ? Si c'est un matériel acheté avant mi Février, il est possible qu'il faille faire une modification matérielle ?
Fred

gd35, 2022-04-13:
Ok je vous contacte via le formulaire de votre site web. Merci de votre aide.

Smanar, 2022-04-13:
Bonjour, il y aurait des valeurs conseillées pour le polling / reporting ?
Pour ne pas surcharger l'appareil ?

fairecasoimeme, 2022-04-13:
Bonjour,
il est conseillé de poll toutes les minutes.
Pour le reporting, on peut en créer 20 maximum et pareil que le poll, éviter de passer en dessous de la minute.
Cependant, s'il manque le condensateur sur le ZLinky_TIC, le problème peut quand même apparaître.
Pour vérifier s'il y a le condensateur voir  : https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware
Fred

gd35, 2022-04-13:
Bonjour,
Je viens de le démonter pour vérifier la présence du condensateur. Il est bien présent:
[image]

fairecasoimeme, 2022-04-13:
Ok donc pas de lien physique avec la deconnexion

gd35, 2022-04-16:
Très bien. Je pense que cela confirme un problème de configuration. Avec les réglages suggérés par @Smanar, je n'ai pas observé de déconnexion pour le moment. Donc c'est sur la bonne voie. Merci de votre aide.

gd35, 2022-04-30:
Bonjour,
Pas de chance avec mon ZLinky, il ne communique plus avec mon réseau zigbee. Aucune lumière allumée. Je l'ai déconnecté du Linky. Reconnecté, fait un reset. Et plus rien... Y-a-t-il un moyen de le tester par un autre moyen N

KipK, 2022-05-14:
Hello , je rencontre le meme soucis de deconnection, avec condensateur.
J'en demande sûrement trop au module, mais c'est problématique dans mon cas.
J'ai besoin d'avoir les changements de puissance et courant quasi instantané. Je fais du delestage avec ma borne de recharge, et si je passe à un update toutes les 60 sec pour les push , j'ai bien peur que cela fasse sauter mon compteur avant.

SylvainPer, 2022-05-14:
> Hello , je rencontre le meme soucis de deconnection, avec condensateur. Je retrouve le module led etteinte au bout d'un certain temps. J'en demande sûrement trop au module, mais c'est problématique dans mon cas. J'ai besoin d'avoir les changements de puissance et courant quasi instantané. Je fais du delestage avec ma borne de recharge, et si je passe à un update toutes les 60 sec pour les push , j'ai bien peur que cela fasse sauter mon compteur avant.

Cette méthode est contre productive, inutile de saturer son réseau zigbee en polling.
Il suffit de faire un config report sur un des champs:
 - ADPS en historique mono, 
 - ADIRx en tri
 - en standard, je laisse @fairecasoimeme compléter
 la variable changera si dépassement. Pour la config, suffit de mettre des temps long et un changement à 1.

KipK, 2022-05-14:
C'est exactement ce que je fais, le polling est repoussé à 600 sec.
Je ne mets que le courant et la puissance en config report ( de 1s à 600sec max ) , pour qu'il me renvoie au plus vite la valeur en cas de changement. Le reste tourne à peu près dans les 300sec.

SylvainPer, 2022-05-14:
Ce n'est pas ce que j'ai écrit. Avoir le changement de puissance ou d'intensité rapidement n'est pas utile à des fins de suivi de consommation, il faut allonger les temps ou la valeur du changement.
Pour du délestage en mono par exemple, il faut utiliser ADPS, celui vaut 0 en temps normal et prends la valeur consommée si dépassement, il suffit donc d'attendre son changement (!=0) et d'appliquer un délestage immédiatement.

KipK, 2022-05-14:
je comprends bien mais dans mon cas ca fonctionne comme ça, le delestage est particulier. J'ai différents scenarios, selon il va delester le chauffage electrique dans différentes pièces, l'eau chaude etc, ou envoyer des commandes à l'EVSE pour dimminuer le  courant de charge, ou remonter à fond quand tout est libre. 
J'ai besoin de connaitre ces valeurs en direct pour calculer combien j'attribue à l'EVSE et au reste de la maison selon les differents scenarios prévus. 
L'ADPS ne me permetque de constater qu'on est déjà en surcharge. 

Pour l'instant ça à l'air de tenir en l'état en enlevant tout les polls  et en limitant les reports sur la plupart des valeurs. Je laisse tourner pour voir.

SylvainPer, 2022-05-14:
Justement, par peur de tout faire sauter, ça devient une usine à gaz.
Lorsque l'on a debug l'ADPS avec Lixee, j'ai eu le voyant au Linky un moment sans que ça ne saute.
L'information utile est vraiment l'ADPS, tant qu'il n'arrive pas, il ne faut rien faire, ou du moins, une régulation lente. Il faut commencer à réagir de manière rapide lorsqu'il déclenche (c'est le fonctionnement du BMS schneider que j'ai, il fonctionne en slow tant que le seuil n'est pas atteint et passe en fast report ensuite).
Donc ce que je fais chez moi, tant que je ne dépasse pas, je laisse vivre et ensuite, si dépassement, je coupe tout le monde et je remets par zone progressivement avant de tout remettre si la conso le permet.

Egglestron, 2022-06-05:
> Bonjour, il est conseillé de poll toutes les minutes. Pour le reporting, on peut en créer 20 maximum et pareil que le poll, éviter de passer en dessous de la minute. Cependant, s'il manque le [...]

Même limite atteinte de mon côté : j'ai réglé les bindings de puissance apparente entre 5 et 120 secondes, avec "Reportable change" à 50 (Watts) pour avoir du *quasi* temps réel et je perds la connexion tous les 1 ou 2 jours... Je vais passer sur du 30 secondes en intervalle minimum (voire 60 comme préconisé, si cela ne suffit pas).

Y a-t-il une explication technique pour ces déconnexions ?

KipK, 2022-06-06:
Ca a tenu 20j de mon coté avant de lâcher. Obligé de redebrancher le module, attendre un peu puis le rebrancher pour qu'il reparte.

Je me demande si le fait que les fonctionalités de routing zigbee soit activée ne ferait pas susrconsommer.

Ca serait possible d'essayer un firmware sans routage pour comparer ? 
Comme vous pouvez voir ici il est maillé un peu partout je me demande si ça peut jouer 

[image]

fairecasoimeme, 2022-06-14:
Bonjour,
Je vais regarder pour faire un firmware en limitant la fonction "routeur" et voir l'impact. je vous tiens au courant.
Fred

KipK, 2022-06-14:
> Bonjour, Je vais regarder pour faire un firmware en limitant la fonction "routeur" et voir l'impact. je vous tiens au courant. Fred

Ah merci. N'hésites pas à le publier que je puisse tester aussi.

pdecat, 2022-06-14:
> Bonjour, Je vais regarder pour faire un firmware en limitant la fonction "routeur" et voir l'impact. je vous tiens au courant. Fred

@fairecasoimeme Serait-il possible de rendre cela configurable via le cluster spécifique fabricant ? Car j'ai peur que la suppression du routage impacte négativement la durée de vie des capteurs sur pile que j'ai placé à côté de mon compteur Linky. Et je n'ai plus de perte de connectivité depuis le condensateur à été rajouté à mon module Zlinky_TIC par tes soins.

fairecasoimeme, 2022-06-14:
Le rendre configurable, à mon avis c'est compliqué. Mais peu importe, si j'y arrive ce sera un firmware spécial pour ceux qui ne veulent pas de routage, genre une nouvelle branche. Ce sera la même version qu'actuellement. 
En gros, c'est juste des directive de compilation qui vont changer

fairecasoimeme, 2022-06-14:
Pouvez-vous essayer avec cette version ?
[ZLinky_router_v5.0_noRoute.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/8900821/ZLinky_router_v5.0_noRoute.zip)

KipK, 2022-06-14:
> Pouvez-vous essayer avec cette version ? [ZLinky_router_v5.0_noRoute.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/8900821/ZLinky_router_v5.0_noRoute.zip)

Tu l'as compilé pour ota ? Cureusement ça reste bloqué à 0.04% chez moi. Je l'avais déjà mis à jour avant.

fairecasoimeme, 2022-06-14:
La version OTA
[ZLinky_GpProxy_Ota_OM15081_V5.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/8901195/ZLinky_GpProxy_Ota_OM15081_V5.zip)

KipK, 2022-06-14:
merci. C'est installé. On verra bien.
Il est toujours identifié comme routeur, mais il faut peut etre que je le reappaire.

Tolriq, 2022-06-14:
@KipK tu as fait comment pour l'OTA? Meme avec le force a true il veut pas update via Z2M.

KipK, 2022-06-14:
J'ai du incrementer le numero de version dans le package ota sinon ca upgradait pas.

KipK, 2022-06-14:
@tolriq, essaye celui là
[ZLinky_GpProxy_Ota_OM15081_V5.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/8903101/ZLinky_GpProxy_Ota_OM15081_V5.zip)

KipK, 2022-06-14:
C'est le même qu'au dessus mais edité en V6 au lieu de V5 avec deconz.
Une fois chargé, le n: de version reste 5, c'est juste pour tromper le moteur ota

Tolriq, 2022-06-14:
Pareil ca continue de me dire que ca utilise le normal :(

```
ota:
  zigbee_ota_override_index_location: index.json
```

```
[
    {
        "url": "ZLinky_GpProxy_Ota_OM15081_V5.ota",
		"force": true
    }
]
```

```
Debug 2022-06-14 21:04:02Is new image available for 'xxxxxxx', current '{"fieldControl":0,"manufacturerCode":4151,"imageType":1,"fileVersion":5}', latest meta '{"fileVersion":5,"fileSize":240478,"url":"https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v5.0/ZLinky_router_v5.0.ota"}'
Debug 2022-06-14 21:04:02Update available for 'xxxxx': NO
Info 2022-06-14 21:04:02No update available for 'Linky'
```

Je doit rater un truc a la con.

fairecasoimeme, 2022-06-15:
> merci. C'est installé. On verra bien. Il est toujours identifié comme routeur, mais il faut peut etre que je le reappaire.

Oui, il restera identifié en tant que routeur mais n'acceptera plus de routes. Si les tests sont concluants, je ferai une version qui sera reconnu en endDevice.

fairecasoimeme, 2022-06-16:
:crossed_fingers:

Tenez moi au courant

Egglestron, 2022-06-20:
Pas d'amélioration de mon côté... J'ai perdu la connexion après un peu plus d'un jour :(

Egglestron, 2022-07-12:
Toujours pas de nouvelles pistes de résolution ? Je ne vais pas tarder à demander un retour au vendeur si aucune amélioration n'est en vue...

fairecasoimeme, 2022-07-12:
Avec la version sans route ?
Pouvez-vous faire une capture d'écran de votre config ?
Votre ZLinky contient bien le condensateur ? https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware

fairecasoimeme, 2022-07-18:
Bonjour, on peut tenter un échange standard si vous le souhaitez. Contactez moi sur mon mail contact[at]lixee.fr

fairecasoimeme, 2022-07-19:
Vous utilisez quoi comme coordinateur ?

Tolriq, 2022-07-19:
Conbee 2, je n'avais pas de déco avant de passer en mode standard. Total à mis 15 jours à faire le changement et pendant ses 15 jours ca marchait.

pdecat, 2022-09-01:
> Le rendre configurable, à mon avis c'est compliqué. Mais peu importe, si j'y arrive ce sera un firmware spécial pour ceux qui ne veulent pas de routage, genre une nouvelle branche. Ce sera la même version qu'actuellement. En gros, c'est juste des directive de compilation qui vont changer

Bonjour @fairecasoimeme, je vois que la [version 6.0 du firmware "router"](https://github.com/fairecasoimeme/Zlinky_TIC/releases/tag/v6.0) limite le routage à 1 :

> Add Route limitation to 1. (Save energy)

J'ai deux capteurs Zigbee sur pile à côté du ZLinky, cela signifie-t-il qu'un seul ne pourra être routé en permanence ou est-ce que cela alternera ?

fairecasoimeme, 2022-09-01:
non un seul sera routé.

la version officielle sera sans routage car beaucoup de SAV lié à des réseaux trop bavards consommateur d'énergie.

Je sortirai la semaine prochaine, une version pour ceux qui ont besoin du routage.

Pour le moment, si tu n'a pas de problème, ne passe pas en v6

pdecat, 2022-09-01:
> Pour le moment, si tu n'a pas de problème, ne passe pas en v6

Trop tard :p

Tolriq, 2022-09-01:
Mis à jour en V6 ici et toujours re stop au bout de 1h :(

fairecasoimeme, 2022-09-02:
Est-il possible de faire une mesure avec un multimètre en mode voltmètre AC et de tester les bornes A et I1 ou I2 du Linky ?

Tolriq, 2022-09-02:
Je doit pouvoir oui, il est sensé y avoir combien ?

fairecasoimeme, 2022-09-02:
Normalement 6Vrms minimum. Non faut enlever le ZLinky

Tolriq, 2022-09-02:
Je suis une bille en elec :)

J'ai ca:
https://images-eu.ssl-images-amazon.com/images/I/71eBjCbTnPL.jpg

J'ai mis sur le V et en AC a priori c'est ce que tu demande.

Ca me dis environ 1.7V entre A et i2 et rien entre A et i1

fairecasoimeme, 2022-09-02:
Voici 2 exemples de mesure sur un Linky triphasé.
Il faut bien insérer les sondes en appuyant sur le bouton pour que le contact se fasse correctement.
Normalement I1 ou I2 ça doit donner les mêmes valeurs.

[image]
[image]

Tolriq, 2022-09-02:
Bon bah j'ai bien vérifier et c'est bien enfoncer à fond (meme profondeur que ce qui rentre sur le ZLinky)

Suis bien en V / AC et j'ai bien que 1.7 sur I2 et 0 sur I1. Soit le multi est pourri soit y'a un soucis sur le Linky ?

Le ZLinky pourrais marcher avec just 1.7V sur le i2 ?

fairecasoimeme, 2022-09-02:
En effet, pas normal. Le ZLinky ne pourrait pas s'allumer avec cette tension

Tolriq, 2022-09-02:
Il marche plusieurs heures donc c'est le multi, c'est un pas cher qui m'as servit a trouver des defaults de terre chez moi.

Tolriq, 2022-09-02:
La ca fait 4h et ca tiens toujours. C'est possible que le ZLinky ait besoin de plus de jus en fonction de la connectivité avec le coordinateur? Je n'ai pas remis le cache sur le Linky ca joue peut être?

Egglestron, 2022-09-20:
Je ne pense pas que cela vienne du multimètre : j'ai fait les mêmes mesures sur mon compteur Landis+Gyr et j'obtiens à peu près ces ordres de grandeur pour la tension AC (quasi identiques sur I1 et I2 cependant).

KipK, 2022-12-28:
Hello.

J'avais oublié de faire un retour mais depuis que je suis passé en no route et que j'ai réglé aux petits oignons mes reporting ( pourtant reporting instantané de mes valeurs Conso Inst , Amp et Tension ) mais j'ai supprimé complètement la lecture forcée de ces paramètres ( je suis sur Deconz ), mon Zlinky n'a pas buggé depuis le mois d'Aout.
Je n'ai pas mis à jours le fw depuis.

PS: ça a rien à voir mais @Tolriq , c'est bien toi l'auteur du genialissime Yatse pour Kodi?
Je dois être un de tes tout premiers clients :)

Tolriq, 2022-12-28:
Oui c'est moi ;) Et oui vraiment rien à voir :p Plus de 11 ans maintenant pour être dans les 1er.

KipK, 2022-12-28:
Ça fait plus de 20 ans que je tourne avec XBMC/Kodi , j'ai même commencé avec la Xbox :) 
Ton client Kodi m'a solutionné beaucoup de choses et je m'en sers encore pas mal.
J'arrête de polluer le thread, désolé pour l'ellipse 😁

Egglestron, 2022-12-29:
> Hello.
>
> J'avais oublié de faire un retour mais depuis que je suis passé en no route et que j'ai réglé aux petits oignons mes reporting ( pourtant reporting instantané de mes valeurs Conso Inst , Amp [...]

Bonjour KipK,
Serait-ce possible de partager ton DDF, s'il te plaît ?

KipK, 2022-12-29:
Le voilà. Pour HC/HP

jiji45, 2024-09-13:
Bonjour à tous.
Je suis sur HA 2024.9.1 et utilise un Zlinky_TIC avec un coordinateur Sonof.
Depuis quelques temps, mon Zlinky_TIC se déconnecte, la led bleue reste fixe, parfois il va se reconnecter sans intervention, et parfois je suis obligé de refaire un appairage. avez-vous des informations sur le problème SVP
Merci d'avance pour votre aide.
Cordialement.