fairecasoimeme/Zlinky_TIC #61: Pour feedback : appairage très difficile et mise à jour OTA impossible

JeroLajHub, 2022-04-11:
Voici mon expérience pour feedback au fabricant du Zlinky et du Zigate v1 que j'encourage en achetant ses produits.

Installation - rien a priori de problématique :
- Compteur Linky Iltron monophasé en mode TIC "Historique" 
  (passage en mode Standard demandé par EDF à ENEDIS a priori dans 24H)
- Clé Zigate V1 en USB à 2-3 m de la Zlinky
- Home Assistant dernière version 2022.04 avec ZHA

La Zlinky a été achetée il y a plusieurs mois a priori selon Home Assistant en version firmware 0001 ... sauf que dans le cluster Basic sw-id affiche "4000"... J'attendais que Home Assistant sache la gérer pour l'initialiser.

Appairage inopérant en suivant la consigne du README : 
- Quand appui de 10 s et relâchement du bouton d'appairage, la led s'éteint 1 sec mais ne clignote pas à la suite contrairement à ce qui est annoncé dans le README. ZHA affiche une fenêtre "Appareil trouvé Démarrage du processus d'interrogation" mais qui ne mène à rien (plusieurs messages en log zha montrant que l'appareil est trouvé mais sans aller jusqu'au bout et sans créer l'appareil dans Home Assistant)
- Quelquefois en tenant de réappuyer sur le bouton pendant 10 sec, la led se met à clignoter très rapidement dès 1 sec d'appui, ceci pendant peut 2-3 sec puis une pause puis 2-3 sec ou encore 5 sec non stop, sans effet sur l'appairage visible.

Appairage obtenu par deux fois avec la manipulation suivante après plusieurs dizaines de tentatives : 
(avec suppression de l'appareil entre deux pour tenter de résoudre le problème de mise à jour OTA)
- appui 10 sec pour faire éteindre la led 1sec afin de faire quitter l'appareil du network pendant une recherche d'appareil par ZHA (message "left network" en log - qqfois avec identifiant 0x000 de l'appareil qui n'est pas celui aléatoire avec lequel il avait rejoint le réseau dans les logs)
- Redémarrage de Home Assistant qui a sans doute vider le cache de ZHA et relance d'une recherche d'appareil dans la configuration de ZHA
- Appui 10 sec sur le bouton d'appairage pour faire éteindre la led 1 sec puis dans la seconde ou elle se rallume de nouveau appui de 1 sec ou moins sur la bouton d'appairage ce qui lance le clignotement rapide 2-3 sec dans la foulée de l'extinction de la led 1 sec.
- La Zlinky est alors immédiatement trouvée par ZHA et créée dans Home Assistant

Les clignotements constatés sont de même fréquence dans les manipulations évoquées, bien plus rapide que ceux évoqués dans d'autres "issues" (2 x 8 clignotements visibles dans une vidéo).

Mise à jour OTA impossible : 
- J'ai suivi toutes les recommandations et consignes des forums pour permettre une mise à jour manuelle en tenant les deux fichiers V4 et V5 séparément
- Comme dans un autre "issue" sur ce forum, "update progress" apparaît dans les logs 3 à 8 fois la même seconde toutes les 5 minutes environ mais reste scotché à 0.0
- Nombreuses tentatives de redémarrage du Home Assistant, temps long laissé en marche, suppression et réappairage pour vérifier si le changement de version avait besoin de cela pour être acté... sans succès

J'ai commandé sur la boutique Zigate un CP2102 USB pour tenter une mise à jour du firmware par USB mais ça devra attendre l'arrivée du colis. Affaire à suivre !

J'espère que cette expérience pourra aider ceux qui galèrent un peu comme moi pour l'appairage... ou que quelqu'un qui me lira aura des conseils à me donner.

JeroLajHub, 2022-04-15:
Voici des nouvelles !

USB-TTL reçu. Mise à jour réussie par USB... non sans difficulté pour faire reconnaître le CP2102 sous Windows 11 (besoin de téléchargé le driver du CP2102 et de faire clic droit puis installer sur le fichier .inf une fois les pilotes extraits du .exe).

Le premier appairage a ressemblé aux échec précédents cités plus haut. Cependant la seconde pressions du bouton d'appairage pendant 10 sec a pour la première fois provoqué l'extinction de la LED suivant de son clignotement 2-3 sec sans devoir rappuyer et alors l'appairage s'est fait immédiatement.

Les valeurs du clusters montrent que le changement de version est effectif. Par contre le numéro de version vu par l'entité Home Assistant est toujours 0x00000001 et Home Assistant croit toujours devoir faire une mise à jour par OTA qui reste toujours bloquée à 0.0 de progress.

Mega-Touk, 2022-04-30:
Pour ma part j'ai aussi galérer a appairer mon device. je pense que j’étais en version 3 ... a confirmer... enfin c'est trop tard .... bref !

J'ai je l'ai flasher avec le binaire de la version 5.0. au moment de le brancher il s'est appairer instantanément sur mon zigbee2mqtt

A l'heure actuelle il me dit être en version 
```
Firmware build date  20220117
Firmware version 4000-0004
```
pourtant je suis sur d'avoir mis la version 5.0  (le build date correspond mais le firmware version me parais étrange)

une OTA m'a été proposée. (31mn d'attente) je lai vu aller a 16% mais je n'ai pas laisser les logs ouvert. Elle a du crasher puisque 5mn après le module n’étais plus en mode update et on me proposais de nouveau une mise a jour OTA. ma version de firmware n'ayant pas bouger

je suspecte que tout nos soucis viennent du condensateur manquant, problème qui a été résolu dans les nouvelles versions de la clef apparemment. d’après le readme il faut contacter le dev https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware j’attends un retour :)

Minims, 2022-05-05:
Idem plein de soucis de perte de connexion, je viens de les contacter. J'ai une des 1ere version également.

fairecasoimeme, 2022-05-13:
En effet, le fait de ne pas avoir le condensateur peut provoquer des coupures d'alimentations surtout quand il est sollicité. Ceci explique que les mise à jour OTA peuvent cracher.
Pour ceux qui ont des premières versions, vous pouvez me contacter pour que je fasse la modification.

Concernant le problème de version suite à mise à jour, je pense que c'est un problème de rafraichissement du logiciel. Tentez de refaire un appairage pour voir si la version est mise à jour.

JeroLajHub, 2022-05-14:
> En effet, le fait de ne pas avoir le condensateur peut provoquer des coupures d'alimentations surtout quand il est sollicité. Ceci explique que les mise à jour OTA peuvent cracher. Pour ceux qui ont [...]

Bonjour, 

Je vais tenter un retour, j'imagine en vous contactant par le site d'achat lixee. 

J'ai acheté l'USB-TTL sur le site Zigate un peu pour rien. Tant pis.

Je vais avant cela retenter une suppression et réappairage pour avoir du retour d'expérience (j'hésitais à faire des tests multiples car difficile d'appairer) et je vous dirai.

lraulet, 2022-05-25:
Bonjour,
Je viens de recevoir mon ZLinky acheté directement sur le site lixee.fr. J'ose espérer qu'il a bien le condensateur.
J'utilise une ZGate en version 3.21 sur HomeAssistant 2022.5.5 (HomeAssistant OS 8.0).
J'ai installé sans trop de problème le ZLinky dans HomeAssistant.
Cependant il semble être en version 0x00000001 ; l'ayant acheté la semaine dernière sur le site lixee.fr je m'attendais à ce qu'il soit déjà en version 5.
J'ai donc tenté de le mettre à jour en OTA mais sans succès. Le Firmware est bien detecté mais le progress "boucle" en restant à 0.0
Une idée pour me dépanner ?