fairecasoimeme/Zlinky_TIC #62: consommation a 0VA

sheep300, 2022-04-30:
Bonjour,
Je suis sur un compteur Linky en historique (passage en standard demandé)
mon Zlinky est en version 4000-0005 sur un Z2MQTT.
je possède un petit panneau photovoltaïque de 300VA, ce matin ma consomation est tombé sur le linky a 0VA hors je me suis rendu compte que le Zlinky m'annonçait toujours 260VA soit la dernière mesure avant le passage a 0VA dans les log du Z2M : "apparent_power":260 apparait toujours la trame.
c'est comme si le 0 n'etait pas une valeur pour le Zlinky
merci par avance

sheep300, 2022-04-30:
voila ce que j'ai dans les log j'ai l'impression que c'est contradictoire
```
Debug Received Zigbee message from 'zlinky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"apparentPower":0}' from endpoint 1 with groupID null

Debug Received Zigbee message from 'zlinky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"apparentPower":310}' from endpoint 1 with groupID null

Info MQTT publish: topic 'zigbee2mqtt/zlinky', payload '{"active_register_tier_delivered":"HC..","apparent_power":310,"available_power":30,"current_tarif":"HC..","current_tier1_summ_delivered":12501.313,"current_tier2_summ_delivered":12287.491,"device":{"applicationVersion":5,"dateCode":"20220217","friendlyName":"zlinky","hardwareVersion":1,"ieeeAddr":"0x00158d0005d2a090","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":9143,"powerSource":"Mains (single phase)","softwareBuildID":"4000-0005","stackVersion":2,"type":"Router","zclVersion":3},"linkquality":255,"meter_serial_number":"811961759844","rms_current":1,"rms_current_max":90,"update":{"state":"idle"},"update_available":false}'
```

STaN33130, 2022-06-08:
Bonjour, exactement la même config, exactement le même problème, avez vous trouvé une solution ?

fairecasoimeme, 2022-07-12:
Je vais regarder mais c'est bizarre car je ne fais que renvoyer ce que le Linky envoie.

STaN33130, 2022-07-12:
Merci beaucoup. Je n’avais pas ce problème avant lorsque j’utilisais deconz. C’est depuis zigbee2mqtt

fairecasoimeme, 2022-07-12:
Du coup, je pense que c'est côté Z2M. je suis en train de regarder le code mais pas trouvé

fairecasoimeme, 2022-07-12:
Je confirme que le problème ne vient pas du ZLinky, je viens de faire le test en faisant varier la valeur de 0 à 12VA puis à nouveau 0. Le ZLinky envoie bien les bonnes valeurs.

une idée @vk496 ?

sheep300, 2022-07-15:
bonjour ne serais-ce pas les valeur negatives qui lui pose probleme ?
merci en tout cas pour vos retour

STaN33130, 2022-07-20:
Je pensais pareil. Personne n'a d'idées?

vk496, 2022-08-31:
Hi,

The problem is solved?

STaN33130, 2022-08-31:
Hi, yesterday not solved, something new in the code ? thank you for your help

STaN33130, 2022-08-31:
Hi thank you for your help. Update 006 Check but problem is the same. When PAPP=0 last value before 0 is available (and no update) on zlinky_tic

STaN33130, 2022-08-31:
[image]

[image]

STaN33130, 2022-08-31:
[image]
[image]

STaN33130, 2022-08-31:
I think there is a divised by 0 or something like that no? Zlinky_tic seems to be blocked. Because there is another problem. When PAPP = 0, OPTARIF stay to his last value « HC… » whereas true value is now « HP… ». Thank you for your help

arenoux, 2022-09-22:
Hello !
J'ai exactement le même problème ici pour la meme config (Panneau Sunology :p )

Pas de solution pour l'instant, mais je vais essayer de creuser coté Zigbee2MQTT

HomeAssistant
Zigbee2MQTT
Zlinky firmware 4000-0006 (Je viens de voir qu'il y avait une nouvelle version.. A tester :))

STaN33130, 2022-09-22:
J’ai trouvé la solution…. Revenir à Zha. Dès que j’injecte jai bien zero de conso sur le linky.

arenoux, 2022-09-23:
Ca confirmerais que le soucis vienne de Zigbee2MQTT...
Ca m'embête de migrer rien que pour ca... Je vais continuer de creuser...

D'autant plus que j'ai egalement le soucis du Zlinky qui s'eteint... (J'ai pourtant bien le fameux condensateur...)

ddelpire, 2022-09-30:
Bonjour,
exactement le même problème et ça m'embête de revenir à ZHA car  ma clé zlinky ne voulait pas s'appairer.
J'ai du passer par Z2MQTT pour bénéficier de l'OTA et maintenant elle est en version 1.0.8.
Peut-être qu'en 1.0.8, elle voudra bien s'appairer mais clairement le problème vient d'une valeur négative que Z2MQTT ne sait pas gérer.

STaN33130, 2022-09-30:
Pour ceux qui peuvent passez sous Zha en attendant. La mise à jour de la valeur power apparent est faite toutes les 7 secondes

BrunoSpy, 2022-10-03:
Est-ce que le problème a été remonté à z2m ? J'ai pas très envie de migrer sur zha car tout le reste fonctionne bien et j'avais pas mal de souci avec zha.

ddelpire, 2022-10-03:
J'ai tenté de re-passer sur ZHA mais on perd l'OTA et mon télérupteur Legrand n'est plus détecté donc retour sur z2m.
J'ai vérifié sur le github de Z2M et quelqu'un a déjà soumi l'issue.
Une nouvelle version est sortie hier mais je ne sais pas si la correction a été prise en compte.

BrunoSpy, 2022-10-03:
Je suis passé à la dernière version (1.28.0) et pas d'amélioration.
Quelle est la référence de l'issue côté z2m ?

BrunoSpy, 2022-10-03:
Pour ceux qui veulent ajouter des informations : https://github.com/Koenkk/zigbee2mqtt/issues/13991

ddelpire, 2022-10-03:
J'ai ajouté aussi un commentaire.

ddelpire, 2022-10-09:
Salut j'ai eu une réponse aujourd'hui

Voici ce que j'ai constaté
@vk496 Problem of conversion number to string in lib/extension/receive.ts ?

const converters = data.device.definition.fromZigbee.filter((c) => {
const type = Array.isArray(c.type) ? c.type.includes(data.type) : c.type === data.type;
return c.cluster === data.cluster && type;
});

Réponse d'un contributeur
@[vk496](https://github.com/vk496)
Contributor
vk496 commented [11 hours ago](https://github.com/Koenkk/zigbee2mqtt/issues/13991#issuecomment-1272481110)

Hello,

Sorry, difficult days and I can't take a look rn.

The problem is with conversion of data types (or precision with floats) for sure, but not there. You should take a look in the lixee.js

Have a nice day :)
Valentín

----------------------------------------
Donc on en revient au problème que j'ai soulevé, le lixee ne sait pas envoyer une valeur négative ou à 0 dans les messages MQTT. 

@fairecasoimeme une idée ?

fairecasoimeme, 2022-10-10:
Bonjour,
Je ne pense pas que le Linky envoie des valeurs négatives avec l'étiquette PAPP.  (Les données ne sont pas signées)
ZLinky n'a pas de lien direct avec MQTT, il envoie les valeurs brutes du Linky au coordinateur.
De ce que je comprends, c'est que Z2M ne traite pas (ou filtre ou n'enregistre pas) la valeur quand elle est à zéro.

BrunoSpy, 2022-10-10:
Je confirme que le problème est bien du côté de Z2M, lixee envoie bien la bonne valeur à 0 mais il y a un bug dans le convertisseur. Un patch a été proposé et doit encore être testé pour vérifier que cela règle le problème.

ddelpire, 2022-10-10:
Du côté de Z2M mais ce n'est pas Z2M c'est le convertisseur zigbee-herdsman-converters qui intérprête mal les données du fichier lixee.js 
https://github.com/Koenkk/zigbee2mqtt/issues/13991#issuecomment-1272905514

ddelpire, 2022-10-10:
@BrunoSpy  je leur ai demandé comment conserver la persistence de la modification car quand on re-démarre le plugin Z2MQTT, le container est détruit et un nouveau démarre en perdant les modifications. A part faire un fork du prokjet et le tester, il n'y a que eux qui peuvent proposer la modif dans la branche dev pour que l'on puisse tester.

arenoux, 2022-10-11:
En effet, pour tester je pense qu'il faut :

**Methode 1:**
- Forker zigbee-herdsman-converters et faire la modif
- Forker Zigbee2MQTT et mettre a jour le dependance vers [zigbee-herdsman-converters  dans le package.json](https://github.com/Koenkk/zigbee2mqtt/blob/master/package.json#L64)
- Build l'image docker du Zigbee2MQTT forké

**Methode 2 :**
- Copier le lixee.js (vers lixee-test.js par exemple)
- Modifier lixee-test.js
- Utiliser la fonctionnalité de [External converters](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters)  en ajoutant :
```
external_converters:
  - chemin/vers/lixee-test.js
```

ddelpire, 2022-10-11:
Ils m'ont préconisé la méthode 2 mais j'avais essayé ce week-end et le plugin ne démarrait plus. J'étais passé par le menu développement HA pour retrourver le dossier z2mqtt. J'avais déposé le lixee.js dedans et ajouté dans le fichier Yaml 
external_converters:
  - chemin/vers/lixee-test.js

Mais ils embarquent déjà nativement le support du lixee donc du coup je me demande si ça ne rentre pas en conflit.

J'ai pensé à la méthode 1 mais comme tu dis je dois forker les 2 projets et reconstruire l'image et à la prochaine mise à jour, je dois tenir mon fork à jour.

Il faudrait déjà au moins valider ce qu'ils ont proposé pour que ce soit intégré dans la prochaine release du  zigbee-herdsman-converters.

arenoux, 2022-10-12:
Pour info @ddelpire.
J'ai pu utiliser la méthode que tu voulais utiliser initialement :p

- s'attacher au container : `docker exec -it container_number /bin/sh`
- vi sur le fichier lixee : `vi node_modules/zigbee-herdsman-converters/devices/lixee.js`
- modifier le fichier (Perso j'ai juste ajouté des logs pour le moment)
- enregistrer le fichier et sortir du container : `exit` :)
- restart le container avec docker-compose : `docker-compose restart zigbee2mqtt`

J'ai bien mes logs qui apparaissent donc je pense que tu peux faire ton test sans fork ni external converters :p
Mes fichiers sont bien conservés, probablement grace au docker-compose.yml avec la section "volumes"

```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - /home/pi/zigbee2mqtt-data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/ttyACM0
```

ddelpire, 2022-10-12:
@arenoux c'est vrai que je n'ai pas pensé à le relancer par docker, je suis repassé par HA en relancant le plugin.
je n'ai pas eu le temps hier d'y regarder.
Et alors ça donne quoi ça corrige le bug ?
Merci !

arenoux, 2022-10-12:
Hello !
Bonne nouvelle ici !
Je viens de tester via mes logs en effet lixee.js ne renvoie pas les data quand la valeur est égale a 0 !
`if(0)` est évalué à "false" et donc en ne fait le traitement qui suit que pour les valeurs non nulles :(

J'ai pu faire le fix en local et valider que ca fonctionne correctement (Quel plaisir de voir arriver ce 0 dans Home Assistant !!)
Je vais m'occuper d'ouvrir un ticket dans le bon repo et je pousserais moi meme un fix, dans une Pull Request :)

Je ne sais pas à quel point l'équipe en charge de zigbee-herdsman-converters seront efficace dans le traitement de cette PR, mais il y a bon espoir que ce soucis soit corriger dans les jours a venir !!!!

ddelpire, 2022-10-12:
En fait je viens de comprendre toi tu utilises z2mqtt avec docker compose moi je l'ai mis en tant que plugin dans HA
Il faut que je vois si je peux ajouter un volume persistant dans la conf du plugin.
Bonne nouvelle !

arenoux, 2022-10-12:
C'est mergé côté zigbee-herdsnam-converter (dans le master). Donc la prochaine release de zigbee2mqtt devrait résoudre le soucis 😇😇😇😇

Pour les plus pressés vous pouvez build vous même l'image docker ou utiliser [Le bricolage suggéré plus haut ;)](https://github.com/fairecasoimeme/Zlinky_TIC/issues/62#issuecomment-1275851151)

BrunoSpy, 2022-11-02:
Nouvelle version de Z2M publiée, problème réglé : le ticket peut être fermé !

arenoux, 2023-09-06:
Je confirme :) Il faudrait fermer ce ticket :)

closed 2023-09-06 by fairecasoimeme