fairecasoimeme/Zlinky_TIC #64: Difficultés récupération informations

StephanGR, 2022-05-22:
Bonjour,

Je suis en train de paramétrer le Lixee fraichement recu
Il s'est bien appairé au réseau et possède une très bonne connexion avec un LQI de 255 et un RSSI de -25 : 
[image]

Cependant je ne comprends pas trop comment fonctionne les valeurs retournées par l'appareil

J'ai regardé, je suis en mode historique monophasé si j'ai bien compris la doc : https://github.com/fairecasoimeme/Zlinky_TIC#values

[image]

Lorsque je navigue sur le cluster 0x0B04 je n'ai pas d'informations fonctionnelles sur les attributs 0x0505 et 0x0508
[image]

Est-ce normal ?

Cependant la valeur de l'attribut 0x050F bouge bien !
J'ai allumé un four électrique pour voir si ca changeait je suis directement passé à plus de 2100

Sinon serait-possible d'avoir plus d'informations sur les DDF car je crois que c'est grâce à cela que HomeAssistant récupère les datas ?

Bien cordialement

tomj44300, 2022-05-22:
En mode historique ça remonte des résultats bizarre ça faisait pareil pour moi , en mode standard par contre aucun soucis
[image]

StephanGR, 2022-05-22:
Ah merci !

Peux tu me dire comment tu as fait pour mettre en mode standard ?

Merci

tomj44300, 2022-05-22:
J ai appelé mon fournisseur d énergie dans mon cas Total énergie demander à passer mon linky en mode standard quelque heure plus tard à minuit ça a changé

StephanGR, 2022-05-22:
Merci beaucoup pour ces informations 👍 

Je viens de faire la demande à total, je te tiendrai au courant !

Sinon tu utilises un DeconZ ou une autre passerelle ?

fairecasoimeme, 2022-05-23:
Bonjour,
Pour la compatibilité ZLinky_TIC avec deconz, je vous conseil aussi ce forum :
https://forum.phoscon.de/t/lixee-zlinky-tic-ddf-file/1164/103

tomj44300, 2022-05-23:
perso j'utilise ZHA
@fairecasoimeme merci beaucoup pour ton travail