fairecasoimeme/Zlinky_TIC #65: Plus de valeur remontée après quelques heures depuis le passage en mode standard.

Tolriq, 2022-05-23:
Hello,

Z2M + firmware 4000-0005, quand j'ai reçu j'étais en mode historique et tout marchais pas trop mal.
Depuis le passage en mode standard qui a mis des plombes coté total les données ne sont plus remontée après un temps aléatoire entre qq heures et 2 jours.

Unplug / replug et tout repart pour quelque temps.

J'ai essayer de limiter la configuration pour réduire la traffic mais ensuite je n'ai pas la moindre idée de ou et quoi debug pour voir la cause.

[image]

navycrow, 2022-06-09:
Je vois que tu as configuré une liste limitative base, papp, iinst qui sont des valeurs du mode HISTORIQUE. C'est peut-être la raison. As-tu essayé de supprimer ces valeur et de mettre **all** à la place. As-tu essayé de forcer le mode historique ?

Tolriq, 2022-06-10:
J'ai tout refait plusieurs fois avec les bons paramètres et la suppression de quasi tous les reporting mais ca tiens toujours pas plus de 12h :(

Tof-le-Chauve, 2022-06-12:
Bonjour, ça ne serait pas le condo par hasard ? Je crois que j'ai le même problème.

_Sur les premières versions hardware, il n'y a pas de condensateur (voir la photo) engendrant quelques fois, sur certains Linky et en fonction du nombre de requêtes, des coupures ou déconnexion. Si c'est le cas, il convient de contacter par mail le fabricant contact[at]lixee.fr pour effectuer la modification._

Tof

Tolriq, 2022-06-13:
Non y'a le condo et tout :(

fairecasoimeme, 2022-06-14:
Bonjour,
Pouvez-vous montrer ce qu'il y a dans l'onglet rapport (reporting)?
Pouvez-vous afficher le network map pour savoir si le ZLinky_TIC a beaucoup de routes enregistrés ?
Fred

Tolriq, 2022-06-14:
Le reporting avec le strict mini (J'aimerais bien en avoir 2 de plus quand même)

[image]

La map quand il est en état buggé
[image]

La map juste après un unplug
[image]

fairecasoimeme, 2022-06-14:
Pouvez-vous essayer avec cette version ?
[ZLinky_router_v5.0_noRoute.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/8900821/ZLinky_router_v5.0_noRoute.zip)

fairecasoimeme, 2022-06-14:
La version OTA
[ZLinky_GpProxy_Ota_OM15081_V5.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/8901195/ZLinky_GpProxy_Ota_OM15081_V5.zip)

fairecasoimeme, 2022-06-15:
Côté ZLinky, il n'y a pas de différences entre les valeurs, le Zlinky_TIC ne met qu'à dispo ses valeurs.
Donc soit il y a un config report différent entre les valeurs qui fait qu'une demande de donnée est faite plus régulièrement qu'une autre, soit la demande ne se fait pas dans le même laps de temps.
Mais ce n'est pas le ZLinky qui pilote ça. Il faut regarder côté ZHA ou script, je ne sais pas.

pipiche38, 2022-10-04:
meme problème depuis le passage en v9. la remonté des infos s'est simplement terminée.
Seul les read attributes fonctionnent ce qui est complètement contre productif
j'ai refait les configure reporting, qui se passent parfaitement, je récupère les etats des configure reporting et ils sont correctes, mais rien n'y fait.

technarf, 2022-11-20:
Bonjour. J’ai exactement le même problème sur un lixee très récent avec condensateur et simples pastilles pour la mise à jour usb (pas très pratique pour le coup d’ailleurs)… Je suis en mode standard (j’ai pris un lixee pour connaître précisément la puissance instantanée consommée sur chacune de mes 3 phases), j’ai installé la v11 limited et tout fonctionne correctement quelques heures puis ça décroche sans que j’ai trouvé d’élément déclencheur. Obligé de débrancher, rebrancher le lixee pour que ça reparte pour un tour… (le problème existe aussi avec la v11 standard). Mon réseau zigbee est conséquent sans problème réel de portée et j’utilise zigbee2mqtt couplé à un Jeedom si ça peut aider… Si je peux aider (dans le limite de mes capacités), dites le moi, et merci pour votre aide car en l’état, le lixee est presque inutile.

technarf, 2022-11-30:
Bonjour, Je me permets de faire un petit retour, car depuis que j'ai installé la V12 limited, cela fait maintenant plus de 7 jours que je n'ai pas eu la moindre coupure... 
Je ne sais pas si c'est lié, mais de mon côté, je ne touche plus à rien... ;)