fairecasoimeme/Zlinky_TIC #68: Informations erronées ??

navycrow, 2022-06-09:
Bonjour,
J'utilise mon lixee avec avec l'intégration Z2MQTT. Et je viens basculer mon compteur en mode STANDARD.
J'ai plusieurs interrogations
La valeur URMS1 (tension efficace) m'affiche 65535 V ?! Cette valeur me parait erronée d'autant que selon le référencielle ENEDIS (Enedis-NOI-CPT_54E), cette valeur ne comporte que 3 caractères. Je pense qu'il y a un erreur de mappage. Qu'en pensez-vous ?
Concernant la valeur de courant efficace (IRMS1). Cette valeur est codé sur 3 caractères et sa valeur maxi ne devrait pas dépasser 90 A toujours selon le référentiel. J'en déduis que l'un des caractères concerne la décimale. Est-ce exacte et si oui comment l'afficher ?

Tof-le-Chauve, 2022-06-12:
Bonjour, j'ai commandé le module il y 15 jours et je l'ai reçu avec un firmware V1. Déjà c'est pas franchement sympa. Du coup, galère de la mise a jour. Bref... Je suis sur Home Assistant + ZHA sur une Conbee II et les valeurs remontées sont complètement farfelues et très (trop) binaires (32768 et 65535) ci-joint mon screenshot. Et pour couronner le tout, le module "Energie" de HA remonte les infos quand il a le temps :)
Avez-vous des explications a tout ça ?
Merci d'avance
Tof.

[image]

navycrow, 2022-06-13:
De mon côté j’ai finalement résolu mon souci en restaurant les paramètres par default de la clé et en relançant Z2MQTT. Depuis tout est ok.

Tolriq, 2022-06-13:
@navycrow tu as fait quoi comme reset pour repartir propre?

navycrow, 2022-06-13:
Après avoir demandé le basculement en mode STANDARD à mon fournisseur, mon compteur a commencé à recevoir de nouvelles infos le lendemain matin mais d'autres infos étaient erronés (comme toi).
J'ai d'abord, sous Zigbee2MQTT, cliquer sur le bouton RECONFIGURER
[image]

Comme ça n'a pas suffit. J'ai redémarré le module Z2M.
[image]

Depuis tout marche nickel !!

Tolriq, 2022-06-13:
Ha le petit bouton caché :) Ca m'a enfin virer une ligne de la partie reporting qui revenait meme en supprimant / rajoutant le device au réseau. Ca à même réparé la map.

On va voir si ca tient dans le temps, mais merci je pense que c'était le truc manquant.

Tolriq, 2022-06-14:
Bon bah non ca marche pas après qq heure ca stop.

Tof-le-Chauve, 2022-06-14:
Exactement la même... Ca fonctionne de temps en temps.

fairecasoimeme, 2022-06-14:
on est plus trop dans le thème de l'issue. Merci de basculer sur le bon sujet.

closed 2022-06-14 by fairecasoimeme