fairecasoimeme/Zlinky_TIC #71: smartenergy_metering_summation_delivered rafraichi seulement au reboot de HA

azodev, 2022-06-27:
Bonjour,
Je rencontre un problème avec le zlinky, la valeur smartenergy_metering_summation_delivered, ne semble pas se rafraichir correctement, en effet elle ne se met à jour que lorsque je reboot HA ou débranche/rebranche le zlinky. le reste du temps seules les valeurs : 
 - electrical_measurement_apparent_power
 - electrical_measurement rms_current
semblent évoluer dans le temps, même si parfois les valeurs semble complètement hors limites, ex : pendant la nuit un pic à 2200va et 9A alors que quasiment tout est coupé, la journée  5370 va et 23 A ... mais le reste du temps les valeurs semblent ok.

Mon installation : HA en docker (sur un Synology) avec Conbee 2 en ACM0, les autres capteurs (prises et capteurs de temp son ok, valeurs bien récupérées régulièrement).
L'intégration est ZHA, et j'ai pu faire la maj du firmware en OTA vers la version 5 , la conbee est aussi à jour (par windows).
Avez-vous une idée ?  Ai-je un autre moyen de tester en branchant la conbee sur windows avec un soft de debug ? dois-je le renvoyer en RMA? Y a t'il le moyen de forcer un demande dans ZHA?
ps : la valeur sur compteur électrique change bien elle.
Merci d'avance.

azodev, 2022-07-03:
pour palier au problème, j'ai forcé la récupration de la valeur grace à une automatisation avec zha_toolkit vers un nouveau sensor, en forcant l'intérogation du cluster 1794:0 la valeur de smartenergy_metering_summation_delivered se met aussi à jour.

gsemet, 2022-07-06:
J'ai le même problème. J'ai l'impression que c'est un bug de ZHA plus que de zlinky

azodev, 2022-07-06:
Bonjour,
il faut installer zha_toolkit depuis HACS
ensuite dans configuration.yaml
```
#ajouter la ligne
zha_toolkit:

#creer un sensor (template)
template: 
  - sensor:
    - name: lixee_zlinky_tic_metering_custom
      unit_of_measurement: "Wh"
      device_class: energy
      state_class: total_increasing
      state: unavailable

#ajouter une automatisation
automation linky:
  - alias: Read ZLinky_TIC every minute
    description: ''
    trigger:
      - platform: time_pattern
        hours: '*'
        minutes: /1
        seconds: '0'
    condition: []
    action:
      - service: zha_toolkit.execute
        data:
          command: attr_read
          ieee: 00:15:8d:00:05:d2:a8:91
          cluster: 1794
          attribute: 0
          state_id: sensor.lixee_zlinky_tic_metering_custom
          allow_create: false
    mode: single
```
Je crois que j'ai rien oublié.
Ps : la solution n'est pas de moi, mais je ne retrouve pas la source.

closed 2022-07-06 by fairecasoimeme