fairecasoimeme/Zlinky_TIC #72: Valeurs inconsistantes en Mode Standard / Triphasé

zen2, 2022-07-03:
J'ai des valeurs inconsistantes remontées par zlinky via zigbee2mqtt:

```
{
    "active_enerfy_out_d01": 9696.091,
    "active_enerfy_out_d02": 12816.58,
    "active_power": 1310,
    "active_power_max": 1190,
    "active_power_max_ph_b": 3310,
    "active_power_max_ph_c": 2570,
    "active_power_ph_b": 1620,
    "apparent_power": 7904,
    "apparent_power_ph_b": 7911,
    "apparent_power_ph_c": 16550,
    "available_power": 12,
    "average_rms_voltage_meas_period": 234,
    "average_rms_voltage_meas_period_ph_c": 238,
    "average_rms_voltage_measure_period_ph_b": 238,
    "current_date": "E220702131414",
    "current_index_tarif": 2,
    "current_price": "HEURE PLEINE",
    "current_summ_delivered": 28745.389,
    "current_tarif": "H PLEINE/CREUSE",
    "current_tier1_summ_delivered": 12331.317,
    "current_tier2_summ_delivered": 16414.072,
    "drawn_v_a_max_n1": 1610,
    "drawn_v_a_max_n1_p2": 3140,
    "drawn_v_a_max_n1_p3": 2630,
    "last_seen": "2022-07-03T11:03:28+02:00",
    "linkquality": 171,
    "message1": "PAS DE MESSAGE",
    "message2": "",
    "meter_serial_number": "021976758997",
    "power_threshold": 12,
    "rms_current": 3,
    "rms_current_ph_b": 9,
    "rms_voltage": 235,
    "rms_voltage_ph_b": 240,
    "rms_voltage_ph_c": 239,
    "site_id": "25317944963012",
    "software_revision": 2,
    "status_register": "003A4401",
    "update": {
        "state": "idle"
    },
    "update_available": false
}
```

Tout semble correct sauf les valeurs suivantes;
```
    "apparent_power": 7904,
    "apparent_power_ph_b": 7911,
    "apparent_power_ph_c": 16550,
```

Alors que la consommation réelle est à ce moment de cet ordre:
P1: ~750/850 W
P2: ~30 W
P3: ~3 W

A noter:
- installation et appairage du zlinky avec compteur linky en mode historique
- upgrade du firmware via OTA vers version 4000-0005
- puis passage en standard du compteur linky via EDF/Enedis
- reinitialisation du zlinky
- effacement du device dans zigbee2mqtt
- réappairage, il s'est reconnecté tout de suite

Configuration zigbee2mqtt v. 1.25.2 [9b2a9210]
```
  friendly_name: zlinky
  linky_mode: standard
  energy_phase: three_phase
  production: 'false'
  kWh_precision: 3
  measurement_poll_interval: 20
  measurement_poll_chunk: 4
  tic_command_whitelist: ''
  debounce: 2
  homeassistant: {}
  optimistic: true
  tarif: Standard - Heure Pleine Heure Creuse
```

fairecasoimeme, 2022-07-05:
Bonjour,
Attention la puissance apparente est en VA et non pas W. La différence est le cos phi liés aux appareils.
Concernant les valeurs, le ZLinky ne fait que lire les données du Linky et les remonte sans traitement. Je viens de tester sur un compteur triphasé et de mon côté je n'ai pas de valeurs bizarres.

Pour en être sur (si c'est possible), il vous faut connecter un module usb ttl au zlinky et ouvrir une console pour valider les données.

TROM83, 2022-07-07:
Bonjour,
Je rencontre exactement le même problème de valeurs incohérentes sur la puissance apparente par phase avec la configuration suivante : 

`0x00158d0005d299c0':
    friendly_name: '0x00158d0005d299c0'
    linky_mode: standard
    energy_phase: three_phase
    production: 'false'
    tarif: auto
    kWh_precision: 2
    measurement_poll_chunk: 1`

`{
    "active_enerfy_out_d01": 30063.93,
    "active_power": 1406,
    "active_power_max": 3440,
    "active_power_max_ph_b": 1800,
    "active_power_max_ph_c": 250,
    "active_power_ph_b": 1682,
    "apparent_power": 52118,
    "apparent_power_ph_b": 11901,
    "apparent_power_ph_c": 2229,
    "available_power": 15,
    "average_rms_voltage_meas_period": 233,
    "average_rms_voltage_meas_period_ph_c": 233,
    "average_rms_voltage_measure_period_ph_b": 234,
    "current_date": "E220707112720",
    "current_index_tarif": 1,
    "current_price": "BASE",
    "current_summ_delivered": 34524.16,
    "current_tarif": "BASE",
    "current_tier1_summ_delivered": 29455.26,
    "current_tier2_summ_delivered": 3686.02,
    "drawn_v_a_max_n1": 2340,
    "drawn_v_a_max_n1_p2": 2110,
    "drawn_v_a_max_n1_p3": 2450,
    "linkquality": 255,
    "message1": "PAS DE MESSAGE",
    "message2": "",
    "meter_serial_number": "021776032394",
    "power_threshold": 15,
    "rms_current": 7,
    "rms_current_max": 65535,
    "rms_current_ph_b": 3,
    "rms_current_ph_c": 1,
    "rms_voltage": 235,
    "rms_voltage_ph_b": 235,
    "rms_voltage_ph_c": 235,
    "site_id": "25305209770218",
    "software_revision": 2,
    "status_register": "003A0001",
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "relais": null
}`

Zigbee2MQTT Current version: 1.26.0-1

fairecasoimeme, 2022-07-07:
@vk496 : est-il possible qu'il y ait un traitement sur ces valeurs côté zigbee2mqtt ?

"apparent_power"
"apparent_power_ph_b"
"apparent_power_ph_c"

zen2, 2022-08-19:
It seems that everything is working fine now...
By looking at my logs it seems that a new version of zigbee2mqtt (1.27.0) have make it works correctly.
I think a reconfiguration at zigbee level of zlinky device was necessary.

closed 2022-08-19 by zen2