fairecasoimeme/Zlinky_TIC #73: Lecture Injection

xlyric, 2022-07-04:
Bonjour, 

Je viens de passer en mode TIC standard, pour récupérer les informations d'injection 

0xFF66 - ( 0x0207 à 0x0209 ) 

en essayant de récuperer les valeurs, j'ai semble t'il des valeurs de lecture incorrecte d'après ce que remonte les développeurs du Pizigate 

https://easydomoticz.com/forum/viewtopic.php?f=28&t=12364&start=50
`Alors donc le Configure Reporting se passe pas bien. En effet on devrait avoir
"xxxx": "00" ou xxxx est l'attribut et "00" le statut

2 hypothèses, l'attribut de supporte le reporting, où les paramètres donnés ne sont pas correct.

0xc1: An invalid or out-of-range parameter has been passed"

De la doc https://github.com/fairecasoimeme/Zlinky_TIC ,
0207 c'est un uint16 --> DataType 0x21
0208 c'est un int16 --> DataType 0x29
0209 ne peut pas être reportable`

une idée ? 

Cordialement

fairecasoimeme, 2022-07-05:
Bonjour,
Il y a une erreur dans le tableau que je viens de corrigé. Le cluster 0xFF66 / 0x207 --> int16 --> data type : 0x29

Attention aussi au max interval du config report qui doit être à 61 secondes (0x3d)

pipiche38, 2022-07-05:
cela n'explique en rien les erreur 0xc1 pour l'attributs 0x0208 ainsi que pour l'attribut 0x0008 

Qu'est-ce qui oblige un Max à 0x3d ( 1 minute et 1 seconde ) ?  Car le Max c'est bien que si aucune Change n'a été detecté il faut quand meme faire un report d'une valeur qui a déjà été communiquée 1 minute et 1 seconde avant !!!

fairecasoimeme, 2022-07-05:
je ne peux pas expliquer le 0xc1 car je n'ai pas la trame de config report, mais comme l'erreur l'indique, c'est que le config report n'est pas bien formé.
exemple d'une trame config report qui fonctionne sur le ZLinky pour le cluster 0xff66 et l'attribut 0x208 / short address : 0xf397
01 02 11 20 02 10 17 D6 02 12 F3 97 02 11 02 11 FF 66 02 10 02 10 02 10 02 10 02 11 02 10 29 02 12 02 18 02 10 02 10 02 10 3D 02 10 02 10 02 10 03 

Quant au max report interval, je me suis mal exprimé, je voulais dire qu'il ne fallait pas mettre une valeur en dessous de 61 secondes

pipiche38, 2022-07-05:
le 0xc1 est du au fait que le Datatype mis etait conforme à la doc avec 0x21 . Donc quand on mettra 29 ce devrait passer. Par contre rien n'explique le 0xc1 pour le 0x0208 et pour l'attribut 0x0008

De toute façon du coté plugin Zigbee4Domoticz j'ai arreté le support du ZLinky . 

Bonne soirée

fairecasoimeme, 2022-07-05:
non, le 0xc1 n'a pas de lien avec le datatype. Lorsque le data type est incorrect, le ZLinky ne répond tout simplement pas.
Donc il risque d'y avoir la même erreur en changeant le datatype. Ce qui est logique vu la même erreur sur les autres attributs.

Si vous me donnez le raw data que vous envoyez pour le config report, je pourrais investiguer. Sans cela, c'est difficile.
Il serait utile aussi de donner la version du ZLinky.

Mais si le produit n'intéresse pas les utilisateurs de domoticz et si ça peut rendre service, je l'enlève des plateformes compatibles.

Merci pour la bonne humeur.

xlyric, 2022-07-05:
Voilà le raw data : 
[export (5).json.txt](https://github.com/fairecasoimeme/Zlinky_TIC/files/9048985/export.5.json.txt)

la version du Zlinkly, c'est une récente, ( achat le 20 mai ) 

Perso, ce qui m'intéresse dans le produit, c'est l'info d'injection, pour le reste, j'ai déjà les infos par d'autre moyen. 
cdt

fairecasoimeme, 2022-07-06:
Bonjour, 
quand je parlais du raw data, je parlais de la trame complète envoyée au Zlinky pour faire le config report.
Avec la trame je pourrai voir ce qui ne va pas

fairecasoimeme, 2022-07-08:
For information this is the real error code error table : 

0xC1 = E_ZCL_CMDS_SOFTWARE_FAILURE
0x8C = E_ZCL_CMDS_UNREPORTABLE_ATTRIBUTE

l'erreur 0xc1 peut provenir de : 
* Data type inconnu
* Trame mal formée
* buffer overflow ( trop de données dans la trame)
* pas assez de données dans la trame
* Plus de place dans la table des reports (limitée à 20)

// ZCL Command status (from ZCL spec)
typedef enum 
{
    E_ZCL_CMDS_SUCCESS =0x00,
    E_ZCL_CMDS_FAILURE,
    E_ZCL_CMDS_NOT_AUTHORIZED =0x7e,
    E_ZCL_CMDS_RESERVED_FIELD_NOT_ZERO,
    E_ZCL_CMDS_MALFORMED_COMMAND =0x80,
    E_ZCL_CMDS_UNSUP_CLUSTER_COMMAND,
    E_ZCL_CMDS_UNSUP_GENERAL_COMMAND,
    E_ZCL_CMDS_UNSUP_MANUF_CLUSTER_COMMAND,
    E_ZCL_CMDS_UNSUP_MANUF_GENERAL_COMMAND,
    E_ZCL_CMDS_INVALID_FIELD,
    E_ZCL_CMDS_UNSUPPORTED_ATTRIBUTE,
    E_ZCL_CMDS_INVALID_VALUE,
    E_ZCL_CMDS_READ_ONLY,
    E_ZCL_CMDS_INSUFFICIENT_SPACE,
    E_ZCL_CMDS_DUPLICATE_EXISTS,
    E_ZCL_CMDS_NOT_FOUND,
    E_ZCL_CMDS_UNREPORTABLE_ATTRIBUTE,
    E_ZCL_CMDS_INVALID_DATA_TYPE,
    E_ZCL_CMDS_INVALID_SELECTOR,
    E_ZCL_CMDS_WRITE_ONLY,
    E_ZCL_CMDS_INCONSISTENT_STARTUP_STATE,
    E_ZCL_CMDS_DEFINED_OUT_OF_BAND,
    E_ZCL_CMDS_INCONSISTENT,
    E_ZCL_CMDS_ACTION_DENIED,
    E_ZCL_CMDS_TIMEOUT,
    E_ZCL_CMDS_HARDWARE_FAILURE =0xc0,
    E_ZCL_CMDS_SOFTWARE_FAILURE,
    E_ZCL_CMDS_CALIBRATION_ERROR,
    E_ZCL_CMDS_UNSUPPORTED_CLUSTER,
    E_ZCL_CMDS_ENUM_END
} teZCL_CommandStatus;

JSteunou, 2022-07-18:
Je ne sais pas si c'est le même problème, si besoin j'ouvrirai un autre ticket, mais je n'arrive pas à lire ma production non plus. J'utilise deconz (peut être que le problème est de leur côté) et que ça soit sur 702 0x0001 ou du côté du cluster spécial LiXee je n'ai que 0

gsemet, 2022-07-26:
j'ai également 0 tout le temps sur la valeur 0x702 0x0001

fairecasoimeme, 2022-07-28:
> Je ne sais pas si c'est le même problème, si besoin j'ouvrirai un autre ticket, mais je n'arrive pas à lire ma production non plus. J'utilise deconz (peut être que le problème est de leur côté) et que ça soit sur 702 0x0001 ou du côté du cluster spécial LiXee je n'ai que 0

Bonjour, pas de lien avec l'issue en cours

JSteunou, 2022-07-28:
> j'ai également 0 tout le temps sur la valeur 0x702 0x0001

Go ticket #78 ;)