fairecasoimeme/Zlinky_TIC #75: Compatibility with domoticz and zigbee receiver

BenGost, 2022-07-08:
Hello,

I couldn't get it to work properly with MQTT with a "CCxxx" key. I only have the network quality indicator.
I've been looking everywhere and can't find a clear explanation.

Can you tell me, if you managed to make it work with Domoticz?
What is the complete configuration of your domoticz?

Thanks

pipiche38, 2022-08-03:
@BenGost it works with Domoticz with the ZigbeeforDomoticz plugin either via the zigpy radio librairies or the Zigate native one.
Only basic reporting works

For more complex especially Energy Production this is too complex as it looks we have to use the ZLinky in an other way.

[image]

BenGost, 2022-08-03:
Thank you pipeche38! I will try that!

pipiche38, 2022-08-30:
@BenGost did you try ?

BenGost, 2022-09-02:
Hi @pipiche38 ,

Yes thank you it works!

closed 2022-09-02 by BenGost