fairecasoimeme/Zlinky_TIC #77: Inconsistence in attributes

pipiche38, 2022-07-20:
Some time Attributes are reportable, some time they are not 

EASF05 | 0x702 | 0x0108 | RP
BBRHCJR | 0x0702 | 0x0108 | RO

Last from the foirmware it looks like it is not reportable at all, while 0106 is !

fairecasoimeme, 2022-07-20:
> Some time Attributes are reportable, some time they are not
> 
> EASF05 | 0x702 | 0x0108 | RP BBRHCJR | 0x0702 | 0x0108 | RO
> 
> Last from the foirmware it looks like it is not reportable at all, while 0106 is !

It's just a doc issue. I fixed the doc

fairecasoimeme, 2022-07-20:
> PTEC | 0x0702 | 0x0020 | RO Which is something changing over the time should be

This value change with fixed hour. Not impredictible update. This could be update for next version.

closed 2022-09-12 by pipiche38