fairecasoimeme/Zlinky_TIC #78: Problème lecture EAIT à 0

fairecasoimeme, 2022-07-28:
le cluster 0x0702 attribut 0x0001 remonte 0. https://github.com/fairecasoimeme/Zlinky_TIC/issues/73

fairecasoimeme, 2022-07-28:
Pouvez-vous donnez la valeur STGE cluster 0xff66 attribut 0x217 ?

JSteunou, 2022-07-28:
J'allais créer un ticket quand j'ai vu votre réponse.

Je ne vois pas de 0x217.

J'ai 0x0300 qui est à 1 (mode standard monophasé)

fairecasoimeme, 2022-07-28:
Vous ne le voyez pas avec Deconz ? peut-être le rajouter dans le DDF

JSteunou, 2022-07-28:
Il n'apparait pas non.

[image]

Pas sûr que le DDF change quelque chose, c'est plutôt pour mapper des attributs avec l'API REST non ?

Smanar, 2022-07-28:
Hello, sur un OS classique tu as un fichier XML `/usr/share/deCONZ/zcl/general.xml`
Vers la ligne 3948 tu as le bloc lixee, la ligne a rajouter est cele du millieu

```
        <attribute id="0x0207" type="u16" name="Puissance app. Instantanée injectée" access="r" required="m" default="0x00"/>
        <attribute id="0x0217" type="cstring" name="Registre de Statuts" access="r" required="m"/>
        <attribute id="0x0300" type="u8" name="Protocole Linky" access="r" required="m" default="0x00"/>
```

fairecasoimeme, 2022-07-28:
Merci @Smanar pour les infos.

JSteunou, 2022-07-28:
Oui merci, j'ai pu avoir l'info du coup. Ça me dit : `003A0001`

fairecasoimeme, 2022-07-29:
Ok, la valeur en bit de 003A0001 est :
1110100000000**0**00000001 avec le bit 8 correspondant au fonctionnement de votre linky. Vous pouvez retrouver ces informations sur le doc Enedis-NOI-CPT_54E.pdf (dispo sur le net)

Apparemment, votre Linky est en mode consommateur (même s'il remonte des étiquettes de production).

Je suis en contact avec Enedis qui me dit que c'est une erreur possible. Pour régler le problème, il m'a demandé votre PDL (numéro unique de votre linky) et votre ville. Vous pouvez me les envoyer par mail (contact[at]lixee.fr) et je ferais suivre.

gsemet, 2022-07-29:
Je pense être dans le même problème, est ce que je peux contacter enedis directement ?

JSteunou, 2022-07-29:
Ok c'est peut-être *normal* tant que le dossier CAC n'est pas entièrement validé (mais que c'est long...). Mais c'est dommage cette façon de faire car le Linky est capable de comptabiliser l'injection peu importe le mode‚ donc pourquoi ne pas juste transmettre l'info tout le temps ?

JSteunou, 2022-08-02:
@fairecasoimeme est-ce que ça veut dire que la TIC ne contient pas forcément tout ce qu'on peut lire physiquement sur son linky et que le ZLinky ne peut remonter qu'une partie des informations ?

fairecasoimeme, 2022-08-03:
@JSteunou 
je communique actuellement avec Enedis, voici un début de réponse sur votre cas : 
```
Le compteur Linky associé au PDL xxxxxxxxxxxxxx est programmé en mode consommateur.

Dans ce mode, comme indiqué dans la note 54E, l'étiquette  EAIT (Energie active injectée totale) n'est pas transmise. Par ailleurs, le client ne dispose pas de contrat avec injection, uniquement soutirage.

Il est nécessaire que les clients aient un compteur configurés en mode producteur. Nous échangerons à le rentrée sur les conditions de basculement en configuration producteur.
```

JSteunou, 2022-08-03:
Ok donc si je comprends bien, pas de CACSI ou de CRAE pas de mode producteur et pas d EAIT transmise.

Aucun autre moyen de récupérer l'information ? Car côté Linky, je peux le lire, mais c'est juste péniable de faire du relevé manuel plusieurs fois par jour :/

JSteunou, 2022-08-03:
Et encore même pas sûr pour la CACSI du coup ça se trouve, vous avez l'info ?

fairecasoimeme, 2022-08-03:
je fais suivre la question

gsemet, 2022-08-03:
Alors pour ma part, j'ai la valeur toujours à zero et pourtant j'ai bien confirmé avec Enedis que mon compteur est en mode producteur. 
Est-ce que je me serais pas trompé dans mon automatisation pour lire la valeur:
```
automation:
  - alias: Read ZLinky_TIC every minute
    description: ""
    trigger:
      - platform: time_pattern
        hours: "*"
        minutes: /1
        seconds: "0"
    condition: []
    action:
      - service: zha_toolkit.execute
        data:
          command: attr_read
          ieee: 00:15:8d:00:05:d2:a9:c5
          cluster: 1794 # 0x702
          attribute: 0 # 0x0000 : Energie active soutirée totale (Wh)
          state_id: sensor.lixee_zlinky_tic_consumed_metering_forced_refresh
          allow_create: false
      - service: zha_toolkit.execute
        data:
          command: attr_read
          ieee: 00:15:8d:00:05:d2:a9:c5
          cluster: 2820 # 0xB04
          attribute: 1295 # 0x050F # Puissance app. Instantanée soutirée (VA)
          state_id: sensor.lixee_zlinky_tic_consumed_power_forced_refresh
          allow_create: false
      - service: zha_toolkit.execute
        data:
          command: attr_read
          ieee: 00:15:8d:00:05:d2:a9:c5
          cluster: 1794 # 0x702
          attribute: 1 # 0x0001 : Energie active injectée totale (Wh)
          state_id: sensor.lixee_zlinky_tic_injected_metering_forced_refresh
          allow_create: false
      - service: zha_toolkit.execute
        data:
          command: attr_read
          ieee: 00:15:8d:00:05:d2:a9:c5
          cluster: 65382 # 0xFF66
          attribute: 519 # 0x0207 # Puissance app. Instantanée injectée (VA)
          state_id: sensor.lixee_zlinky_tic_injected_power_forced_refresh
          allow_create: false
    mode: single
```

fairecasoimeme, 2022-08-03:
@gsemet pouvez vous remonter la valeur STGE ? cluster 0xff66 attribut 0x217 ?

gsemet, 2022-08-04:
Alors avec la commande home assistant:
```
service: zha_toolkit.attr_read
data:
  ieee: 00:15:8d:00:05:d2:a9:c5
  cluster: 65382  # 0xFF66
  attribute: 535  # 0x217
  csvout: zlinky
```

Je lis la valeur suivante:
```
2022-08-04T11:41:49.645837+00:00,Manufacturer Specific,535,b'\x00\x00\x00\x00\x00\x00\x00\x00',0x0217,0xFF66,1,00:15:8d:00:05:d2:a9:c5,
```

JSteunou, 2022-08-21:
Bon j'ai eu des infos de la part du support Enedis, pas de Linky en mode producteur si pas de CRAE. En CACSI ils ne touchent à rien donc la valeur ne remontera pas.

Pour tout ceux qui ont du PV ça veut dire EDF OA ou rachat par un autre qui veut bien (perso je n'en connais pas pour de l'auto installation) ou batterie virtuelle avec Urban Solar par exemple. Et si vous ne voulez (comme moi) pas vous emmerder avec Consuel et paperasse bin retour à la bonne vieille pince ampermetrique pour avoir son injection.

Smanar, 2022-08-21:
Heu ...
L'info m'interesse, mais tu pourrait me dire ce que veut dire CRAE et CACSI  ?

JSteunou, 2022-08-21:
CACSI : Convention d’AutoConsommation Sans Injection, c'est la demande de raccordement hyper basique dans laquelle Enedis se protège en demandant de ne rien injecter selon leur interprétation de la loi alors qu'on peut se permettre d'injecter un peu. Dans les faits, avec ou sans CACSI je crois que personne ne voit la différence.

Contrat de Raccordement, d'Accès et d'Exploitation : là on arrive dans de l'injection officielle dans le réseau, soit avec revente soit cédé à titre gratuit, et apparemment il faudrait au minimum ça pour avoir son Linky en mode producteur.

fairecasoimeme, 2022-08-21:
D'après mon contact chez Enedis, il y a des discussions sur le passage en mode production. On a une réunion mi-septembre où on doit en parler. Je vous tiendrai au courant

gsemet, 2022-09-05:
Bonjour. J'ai eu le support Enedis et ils m'ont bien confirmé que mon Linky était bien configuré en mode "producteur" (depuis 2018, j'ai mes facturations etc, pas de pbl), et pourtant je n'arrive toujours pas à lire de valeur injectée sur le réseau

fairecasoimeme, 2022-09-07:
Bonjour,
la valeur de STGE que vous m'avez fourni plus haut semble être NULL. Serait-il possible de redemander la valeur correspondante à cluster: 0xFF66 attribut: 0x217 ? (ne pas mettre le manufacturer specific si possible)

fairecasoimeme, 2022-09-23:
@gsemet Oui pas normal de pas avoir la valeur STGE. C'est l'occasion de mettre à jour en effet.

ddelpire, 2022-11-02:
Bonsoir,
post très intéressant, je viens de comprendre pourquoi j'ai 
[image]
alors que mon compteur m'affiche plus de 300kw de ré-injection mais j'ai le choix d'un contrat CACSI.

Donc ce n'est pas la sonde qui est en cause mais je croyais que la sonde Zlinky prenait ses infos sur le compteur pas en téléinfo. Là, j'ai du mal à comprendre, pourquoi allez récupérer les valeurs sur leurs serveurs alors qu'elles sont disponibles en locales.

Il n'y a donc aucune solution avec le Zlinky pour suivre sa ré-injection ? mon but était de pouvoir faire des scenari en cas de ré-injection.

Merci.

pabaillon, 2023-01-12:
Je pense qu'il serait pas-mal de mettre cette information sur la boutique, ce qui éviterait des déconvenues comme la mienne.  Surtout que certains sites véhiculent l'idée que l'on peut faire remonter cette information : https://www.lesalexiens.fr/labo-test-alexa/test-du-lixee-zlinky-votre-consommation-electrique-sur-home-assistant-en-temps-reel/ .

fairecasoimeme, 2023-01-12:
Bonjour, 
je ne l'avais pas rajouté parce que c'était en discussion. Ils peuvent le débloquer actuellement uniquement à la main (au cas par cas) et malgré discussions, il ne semble pas que l'automatisation puisse se faire prochainement.
Je vais donc rajouter dans la description du produit que les contrats CACSI ne permette pas de récupérer l'injection au travers de la téléinformation.

Certains sites véhiculent l'idée parce que ce sont des utilisateurs qui ont surement un contrat le permettant et qu'ils ont testé. Ils ne sont pas forcément au fait de tous les cas particuliers.

nyroDev, 2023-01-12:
@fairecasoimeme Je suis actuellement en attente de mon consuel pour valider mon contrat de vente et passer officiellement en producteur.
Pour le moment, je ne reçois pas l'information via le TIC d'injection (mode standard) alors que le Linky l'affiche bien (total injecté et instantané)

Dès réception du consuel et validation par le service OA de mon compte producteur, j'espère avoir les informations qui remontent dans le TIC.
Je ne manquerai pas de revenir ici pour confirmer ou infirmer ce point là.

fairecasoimeme, 2023-01-12:
Dans tous les cas, si PB, je pourrais alors contacter directement Enedis pour leur signaler.

navycrow, 2023-02-20:
je croyais également le zlinky prenais l'info direct sur le compteur. Je suis dans le même cas. Injection affiché à l'écran mais rien qui remonte. C'est balot !

bubar26, 2023-03-18:
Bonjour, voici ma config: RPI3, HAOS 2023.2.4,  ZHA Mosquitto, Cobee2, Lixee 

Je ne suis pas du tout a l'aise toutes les possibilités de HA mais grace aux echanges comme ici , j'avance tout de meme, MERCI!

Apres avoir lu ici et la qu'il fallait être déclaré producteur auprès d'Enedis pour exploiter le cluster 0xff66, c'est enfin chose faite. l' attribut 0X0300 linky mode remonte = 5 :-)
je peux enfin lire les attributs d'injection 0x0207 - 209 manuellement.

comment a present remonter cela simplement dans une carte jauge pour mon dashboard?

D'avance merci

bubar26, 2023-03-26:
Bonjour, un petit UP, au cas où une bonne âme passerait par la :-)

Bonne journée

VianneyHENON, 2023-04-04:
Pour info si cela peut aider, j'arrive bien à récupérer l'index d'injection EAIT avec le Zlinky sous home assistant.
[image]

[image]

JSteunou, 2023-04-05:
Votre contrat d'auto-consomation est-il un CRAE ? Avez-vous dû passer le consuel ?

bubar26, 2023-04-05:
@JSteunou je ne sais pas ce qu'est CRAE ni ou le voir. Aussi je n'ai pas eu a faire passer de Consuel (je crois par ce que inferieur a 3Kw de panneau PV). Mon Linky est en mode  "standard producteur" .

@VianneyHENON Merci beaucoup pour ce retour plein d'espoir :-)
je n'arrive pas a avoir ce genre de compteur, je n'ai pas de sensor correspondant a l'injection...
L'injection n'est dispo que via les interrogation de cluster...
Si je comprend bien ma différence apriori est que je passe par ZHA qui ne propose pas tous les sensor disponibles sur Lixee ?

JSteunou, 2023-04-05:
@VianneyHENON et @bubar26 quelles ont été vos démarches pour passer en mode "producteur". J'ai fait une CACSI (simple, 5 min, validé très rapidement) mais je ne vois pas comment passer en producteur sans que ça soit tout de suite les gros dossiers, le consuel, ...

JSteunou, 2023-04-05:
@bubar26 c'était quelle démarche cette déclaration de production, une CACSI ?

bubar26, 2023-04-05:
j'ai recherché pour vous et en effet j'ai retrouvé le contrat "producteur" CRAE : Contrat D'accès et d'Exploitation.
il avait fallu prendre une photo des panneau au sol et fournir l'attestation des onduleur disposant de la protection des intervenant en cas de coupure pour éviter le passage du CONSUEL et donc une facture.
a titre d'info, sur mon espace client Enedis on peut voir que tous les documents echangés sont tagués  " attestation de conformité visée par le CONSUEL"

JSteunou, 2023-04-05:
C'est bien ce qui me semblait, vous faîtes parti des chanceux qui sont passés entre les mailles sans consuel ;)

bubar26, 2023-04-05:
je ne suis pas sur que cela soit de la chance, il y a cette histoire de production de moins de 3Kw, installé a moins d'1.8m de hauteur, protection électrique intégrée au micro onduleur = procédure simplifié et gratuite du Consuel.

Ensuite les histoires de producteur standard ou standard simple ou historique c'est un peu la bataille, et j'ai peut être eu de la chance en jouant les imbéciles qui demande gentiment a ENEDIS de l'aide.

Le discours que j'ai tenu :  le monsieur qui a installé les panneau et la box domotique me dit que ca marche pas a cause du mode linky que ENEDIS gere, il faut leur demander. Du coup allo ENEDIS help me... 

Les personnes ne st pas toute au courant de cela chez Enedis, mais ont toutes essayer de faire au mieux et cela a finalement marché, ca a été long.

bubar26, 2023-04-05:
j'aurai aimé avoir comme sur la page ENEDIS mais ds HA
[image]

bubar26, 2023-04-05:
@JSteunou  la declaration se fait 
non pas sur  l'espace Enedis classic  ( https://mon-compte-particulier.enedis.fr/) 
mais sur le portail de raccordement >>>    **https://connect-racco.enedis.fr/**

JSteunou, 2023-04-05:
@bubar26 je ne connais pas assez ZHA, moi je passe par Deconz mais même déjà là c'est compliqué, suivant le profil du Linky il faut charger une config différente. À cause de la manière dont a été fait l interface zigbee, apparemment ce n'est pas possible d'avoir quelque chose d'universelle, du coup il y a un profile pour ceux en Historique, un pour Standard, un pour Standard HC/HP, etc. ZHA est peut être moins souple là dessus. J'ai entendu beaucoup de bien de Zigbee2MQTT, le support sera peut-être meilleur.

gsemet, 2023-04-05:
Il faut demander a votre fournisseur energetiques (EDF, Total,..) de passer le compteur en mode Standard, ca se fait par téléphone ça prend 10 minutes, et en 24-48h, Enedis a fait la transition sur votre compteur.

bubar26, 2023-04-05:
est ce que je peux installer l'integration Z2Mqtt juste pour le Lixee et garder tous les autre device sous ZHA ?

VianneyHENON, 2023-04-05:
@JSteunou, oui c'est un contrat CRAE, panneaux solaire installé par un installateur agréé et un électricien est venu faire le raccordement, donc j'imagine que le consuel est passé ;)

Un Contrat CACSI = (Convention d'Autoconsommation Sans Injection)
Donc vous ne faites pas d'injection sur le réseau.

@bubar26, je n'ai utilisé que zigbee2mqtt sur home assistant, l'index d'injection n'était pas récupéré quand le linky était en mode historique, dès que j'ai eu le mode standard, j'ai réapparait le zlinky et l'index EAIT est apparu. :)
Le plus difficile et le plus longs ;) a été faire comprendre à EDF (mon fournisseur) de faire une demande F185 auprès d'énédis. Quelques jours après le linky est passé en mode Standard.

VianneyHENON, 2023-04-05:
> est ce que je peux installer l'integration Z2Mqtt juste pour le Lixee et garder tous les autre device sous ZHA ?

Pas sûr car, tout dépend si ton dongle/clé zigbee peut-être utilisé avec 2 intégrations zigbee en même temps.

gsemet, 2023-04-05:
Bon, il faut refaire tous tes inclusions....

bubar26, 2023-04-05:
Merci pour vos retours. Je m'aperçois que je n'ai pas "réappairer" le lixee après le passage en mode producteur. Je voudrais donc tenter. Que dois faire, bien noté son nom, le supprimer et rester 10sec sur le bouton pour le redetecter/ renommer et hop ?

JSteunou, 2023-04-06:
@gsemet ça ne suffit pas, il faut le mode Standard producteur

@VianneyHENON en fait il manque un entre 2 côté enedis car la CACSI est *censée* être sans injection sauf que la loi nous autorise à injecter jusqu'à 3kW et à moins d'avoir un onduleur qui coupe / bride automatiquement, ce qui est non consommé est forcément *injecté* dans le réseau. Le pire c'est que le compteur Linky le comptabilise, mais côté TIC ça ne remonte pas.

VianneyHENON, 2023-04-06:
> e après le passage en

Oui, tu peux tu faire la procédure de réappairer, sinon supprimes complétement le lixee en avec la coche forcer la suppression. Tu réappaires, le renommer je dois pas jouer, car homeassistant s'appuie sur l'id du sensor.

VianneyHENON, 2023-04-06:
> @gsemet ça ne suffit pas, il faut le mode Standard producteur
>
> @VianneyHENON en fait il manque un entre 2 côté enedis car la CACSI est _censée_ être sans injection sauf que la loi nous autorise à [...]

Si tu ne parviens pas à récupérer les informations du TIC, essai avec une pince ampèremétrique connecté ou Sonoff Pow Elite, ou un plugin sur Domoticz ou Home assistant qui récupère les infos depuis le site ENEDIS, tel que [DomoticzLinky](https://github.com/guillaumezin/DomoticzLinky) que j'utilise depuis qqes années sur Domoticz.

VianneyHENON, 2023-04-06:
> Merci pour vos retours. Je m'aperçois que je n'ai pas "réappairer" le lixee après le passage en mode producteur. Je voudrais donc tenter. Que dois faire, bien noté son nom, le supprimer et rester 10sec sur le bouton pour le redetecter/ renommer et hop ?

Oui, tu peux tu faire la procédure de réappairer, sinon supprimes complétement le lixee en avec la coche forcer la suppression. Tu réappaires, le renommer ne dois pas jouer, car homeassistant s'appuie sur l'id du sensor.

bubar26, 2023-04-07:
erf j'ai supprimé le lixee et impossible de le réappairer , plus rien de marche....
N'ayant pas de piste suppl. j'ai installé Z2MQTT, le lixee a pu être détecté et appairé après x tentative .
actuellement rien ne semble en vie, z2mqtt me déroute bcp je ne retrouve pas mes petits, et tout est en vrac... question d'habitude.
Très frustrant ce lixee.

VianneyHENON, 2023-04-07:
> erf j'ai supprimé le lixee et impossible de le réappairer , plus rien de marche.... N'ayant pas de piste suppl. j'ai installé Z2MQTT, le lixee a pu être détecté et appairé après x tentative . [...]

Il faut trouver ses repères, c'est que je me disait il y a qqes semaines, quand j'ai commencé à utiliser Home Assistant. :)

bubar26, 2023-04-07:
Je pense avoir une instabilité quelque part, les perf sont tres mauvaise depuis z2mqtt. 
je ne suis pas bien certain du cheminement et des fonction de chaque module ou integration.
J'avais initialement le module Deconz pour la conbeeii
puis je l'ai désactivé pour mètre a la place l'intégration ZHA qui apriori nécessitait le module Mosquitto et l'intégration MQTT
Avec tout cela ca fonctionnait de maniere plutot fluide.
Maintenant j'ai désactivé ZHA  et mis le module Z2Mqtt et Mosquitto est tjrs actif. est ce ok ?
par ou passer vous pour appairrer les device? L'UI de Z2Mqtt est tres tres lente...

VianneyHENON, 2023-04-07:
> Je pense avoir une instabilité quelque part, les perf sont tres mauvaise depuis z2mqtt. je ne suis pas bien certain du cheminement et des fonction de chaque module ou integration. J'avais [...]

Je ne pas vous dire pour l'instabilité je n'utilise que le lixee en zigbee. Effectivement il faut installer mqtt + z2mqtt comme indiqué ici 
https://www.lesalexiens.fr/labo-test-alexa/test-du-lixee-zlinky-votre-consommation-electrique-sur-home-assistant-en-temps-reel/
Et
https://m.youtube.com/watch?v=1_pkEXI7ZSI

Si cela peut aider ;)

bubar26, 2023-04-10:
Je vais essayer ça merci. Apparemment il y a des soucis avec les conbee et z2mqtt. J'ai commandé une sonoff. J'attends de la recevoir.

bubar26, 2023-04-13:
Hello, pour un premier retour:
Avec la clé Sonoff Z2mqtt est plus stable, l'interface reagi correctement.
Je n'ai pas pu ré appairer mes plug tel quel, il a fallu que je les rapproche du controlleur, les debranché rebrancher plusieurs fois avant de pouvoir les retrouver.

Pour certaine prise je n'y arrive toujours pas: lors de l'appairage z2mqtt me met le popup rouge avec l'ID expliquant que le device a bien été retiré (???)
Si vous avez une idée je suis preneur.

Enfin l'objet essentiel de cette migration, le lixee n'est toujours pas détecté et lui je ne peux pas le rapprocher. 
Même avec ZHA et Conbee c’était fastidieux, je vais d’abord trouvé un plug de plus pour densifié le maillage a proximité du lixee avant toute chose.

[33 more replies]