fairecasoimeme/Zlinky_TIC #79: Probleme d'appairage

squady, 2022-07-30:
Bonjour,
Je sais pas si je suis au bon endroit pour solliciter de l'aider, mais je tente quand même :/
Je suis sous home assistant et ZHA.
J'ai acheté un ZLinky et une Zigate Passerelle USB TTL Version 2 et il m'est impossible de pairer la ZLinky.
J'ai branché le ZLinky, la led s'allume et reste fixe, d’après la doc cela signifie que tout va bien.
Pour pairer je passe ZHA en mode inclusion, puis, sur le ZLinky, j’appuie sur le bouton pendant 10s, la led s’éteint puis se rallume fixe (normal).
J'ai beau attendre, recommencer la manipulation, le ZLinky n'est jamais detecté par ZHA.
Le module ZLinky se trouve à moins de 5m de la clé zigate.
J'ai tenté de pairer d'autres périphériques pour m'assurer que la clé zigate n’était pas en cause, et tout fonctionne bien, seul le zlinky reste invisible.
Que puis-je faire de plus ?

capi92, 2022-09-02:
pour suivre, J’ai le même pb avec zigbee2mqtt en passant pas une clé sonoff.

pvince83, 2022-09-07:
Meme chose chez moi!

Sn4kY, 2022-09-07:
Bonjour
exactement le même problème. je ne parviens pas à associer le Zlinky avec la clé USB ZiGate+

capi92, 2022-09-09:
J’ai un répéteur à moins de 1m et ça ne change rien…

fairecasoimeme, 2022-09-09:
Existe t-il des logs ? ou il n'y a absolument rien ? Pouvez-vous partager des photos sur l'installation ?

JulienFauconnier, 2022-09-09:
Même problème, mais aucun logs de disponibles car il n'y a tout simplement rien de vu dans Home Assistant (ZHA)

fairecasoimeme, 2022-09-12:
> Même problème, mais aucun logs de disponibles car il n'y a tout simplement rien de vu dans Home Assistant (ZHA)

A quelle distance ?

Selon l'environnement de chacun, il est possible aussi de tester la version avec  antenne externe permettant de limiter les interférences et augmenter la portée.

pvince83, 2022-09-12:
Bonne nouvelle, j'ai réussi a associer mon ZLinky !!!

Le problème était du au firmware de mon dongle SONOFF (j'avais mis un firmware custom pour avoir 20db en mode routeur) en remettant celui fourni par z2m j'ai réussi a l'associer mais j'ai du le placer a 10 cm.

Un fois associer j'ai pu remettre le dongle a l'intérieure de la maison (a environ a 4m du Zlinky)

Par contre malgré un LQI a plus de 100 il ne se rafraichit pas toutes les 7 secondes mais entre 10s et 5 min, pas terrible pour la conso instantanée mais ça permet d'alimenter correctement la vue énergie de home assistant.

Une chance de résoudre ce problème avec l'antenne?

plouf34, 2022-10-01:
Bonjour
Mon zlinky tout fraiChement installé (en extérieur près du linky), je me heurte à sa son connection à son router. 
Comment appliquer un firmware non route sur HA et zigbee2mqtt?

RBXIII, 2022-10-10:
Meme problème pour moi (Zigbee2Mqtt + SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P), j'ai finalement pu l'associer à mon coordinateur Zigbee en plaçant ce dernier à tout près du module lixee. Cela ne semble pas fonctioner via un des routeurs zigbee..