fairecasoimeme/Zlinky_TIC #81: Problème de portée très courte <5mètres

Kuinox, 2022-08-10:
Bonjour,

J'ai un problème de portée avec mon ZLinky.

Mon linky est au fond du jardin, mais ayant un abri de jardin avec un relais sonoff "BASICZBR3", situé a 10-15 mètres, je me suis dit qu'il n'y aurait pas de problème de portée.
À la première installation, l'appairage ne se faisant pas, j'ai tiré une rallonge pour approcher ma Raspberry Pi (qui a le dongle zigbee). L'appairage n'a pu se faire qu'une fois que la raspberry pi à 2 mètres du caisson.

Ce soir, j'ai voulu tester si le ZLinky pouvait communiquer par rebond, avec une rallonge j'ai ramené une multiprise zigbee, et j'ai bien pu confirmer que le ZLinky pouvait communiquer via un autre appareil.
Via l'interface de visualisation d'home assistant, j'ai bien pus voir que le ZLinky communiquait bien avec la multiprise.
Je sais que cette multiprise zigbee a une excellente portée, vu qu'elle arrive à communiquer directement avec la raspberry pi situé a à 25 mètres, dans la maison.

J'ai donc essayé d'estimer la portée du signal, en appuyant sur le bouton "identify" dans home assistant, et en éloignant la multiprise jusqu'à que cela ne marche plus.

Sur cette image, on a dépassé la portée du signal et appuyer sur le bouton identify ne fonctionnait plus :
[image]

On est a moins de 5 mètres, le caisson est ouvert, avec ligne de vue (moins quelques plantes) sur le ZLinky.  

La multiprise était toujours pilotable via home assistant.
Le dernier LQI lus était de 75 avec la multiprise.

Est ce normal que la portée soit aussi courte ?

fairecasoimeme, 2022-08-12:
Bonjour,
Non ça ne me semble pas normal <5m pourtant un LQI de 75 est "acceptable". Bizarre que cela tombe aussi rapidement.
Pouvez-vous donner plus d'information sur le dongle zigbee ?
Contactez moi par mail pour faire un échange ou tester avec la version antenne externe

Kuinox, 2022-08-12:
> Pouvez-vous donner plus d'information sur le dongle zigbee ?

C'est celui ci: https://www.domadoo.fr/fr/interface-domotique/5431-popp-dongle-usb-zigbee-zb-stick-chipset-efr32mg13-4251295701554.html

> Pouvez-vous donner le lqi quand la multiprise est collée au ZLinky ?

Collé: 203

Après plus de test, j'ai remarqué que la puissance du signal est très directionnelle: 
En face, je peux facilement dépasser les 10 mètres.  
Sur le coté (comme sur la photo plus haut) le signal baisse très rapidement.  
Et mon relais, situé a 15 mètres est dans cette angle mort.

fredleger, 2022-08-14:
Je pense avoir le même souci j'ai un relay zigbee via un Philips Hue Discover white and color ambiance flood light à moins de 10m et un LQI de 93 et pourtant je n'arrive pas à faire d'appairage avec le lixee qui se trouve sur le côté. Je vais faire plus de tests avec une multiprise ou autre c'est une bonne idée

Kuinox, 2022-08-14:
J'ai testé en mettant du papier aluminium pour refléter les ondes, j'ai gagné quelques mètres mais ce n'est pas suffisant.

closed 2022-08-30 by Kuinox

fairecasoimeme, 2022-08-30:
Parfait. 
Merci pour le retour d'expérience. Si ça pouvait être toujours le cas de doubler avec l'antenne ...
Mais malheureusement, chaque environnement est spécifique.