fairecasoimeme/Zlinky_TIC #83: Attribute 0xff66/0x0200 value seems strange

pipiche38, 2022-08-20:
The data received from the ZLinky on attribute 0xff66/0x200 is 10 bytes long with  not really ascii characters . Is that expected ?

ReadCluster ff66 - 5f21/01 Attribute: 0200 Type: 42 Size: 0010 Data: 0201002000010002020042000d

fairecasoimeme, 2022-08-30:
it's weird : 
it's seems to be ok at home : 
Type: 42 Size: 10 Data:  20 20 48 65 75 72 65 20 70 6C 65 69 6E 65 00 20 -->   Heure pleine

pipiche38, 2022-09-02:
Your are in the case where you are in Standard mode, but if you are in Historique mode,

fairecasoimeme, 2022-09-02:
with the "historique mode" :
10:21:10.069 <- 01 80 02 00 26 D9 00 01 04 FF 66 01 01 02 75 E5 02 00 00 18 08 01 00 02 00 42 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 30 03
Type: 42 Size: 10 Data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

pipiche38, 2022-09-02:
Je comprends pas l'approche consistant à mettre des NULL alors que c'est un type 42 donc la longueur devrait etre simplement de 0

fairecasoimeme, 2022-09-05:
L'approche est due au fait que dans le protocole ENEDIS en mode standard, l'étiquette LTARF contient 16 caractères de manière fixe. Donc j'ai créé une variable à taille fixe. (Sachant qu'en mode standard, l'étiquette LTARF est obligatoire)
Si tu es donc en mode historique, l'étiquette n'est du coup jamais analysé et reste null.
Mais on s'éloigne de ton problème. Dans quel contexte tu obtiens des caractères non null dans les données ?

closed 2022-09-12 by pipiche38